ros2/ros2 #886: Eloquent Elusor Patch Release 1 improperly released

bverhagen, 2020-04-07:
## Bug report

**Required Info:**

- Operating System:
  - any
- Installation type:
  - From source
- Version or commit hash:
  -  Eloquent Elusor Patch Release 1 (`release-eloquent-20200124` tag)
- DDS implementation:
  - any
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
When building the `release-eloquent-20200124` tag from source, it has the exact same result as building `master` - including build failures and changes.

#### Expected behavior
Eloquent Elusor Patch Release 1 builds the same software and features as it did on the day it was released.

#### Actual behavior
Eloquent Elustor Patch Release 1 builds the latest master. This is because https://github.com/ros2/ros2/blob/release-eloquent-20200124/ros2.repos is still listing `master` branches for every dependency, rather than the proper release version.

#### Additional information
I am not sure anything can be fixed for the Eloquent Elusor Patch Release 1, as changing a tag in a popular public repository is frowned upon, but at least it is documented here and some awareness is created to doublecheck this for Eloquent Elusor Patch Release 2 and further.

A workaround is to use Eloquent Elusor Patch Release 0, whose versioning is ok.

nuclearsandwich, 2020-04-07:
Moving tags is indeed frowned upon because any existing clone of the repository will need to explicitly remove and re-fetch the updated tag.

One minimal course of action could be to:
- [ ] Update the GitHub release description to explain the situation.
- [ ] Post to the release announcement on discourse advertising this issue.
- [ ] Wait for patch release 2 to come and make sure it's tagged appropriately and recommend people jump to that.

I would prefer adding a new tag that documents where patch release 1 actually was. In an offline discussion with @mjcarroll it seems that the binary archives uploaded to GitHub for that release were built from the release branch rather than the tag, which means they should be accurate.

Since the bionic packages are built from the distribution.yaml they are also not affected by this. 

I'm not sure if there's a way we can move the archives to a separate tag without re-uploading them, maybe we can just link back to the binaries on the bad tag but that could confuse people as we link them out and then back to get binary archives.

As an extreme option we could basically re-issue the release, with a new tag and new binaries built fresh but that's a lot of work with patch 2 right around the corner.

mjcarroll, 2020-04-07:
Yes, in this case it should only be the tag itself that is incorrect.  The tag should point to https://github.com/ros2/ros2/commit/b6f1c8610e69ba72e89e916e528f4fc7f8d32030 rather than where it currently is.

The binary packages for the release were built of a set of package versions that matches what is in that commit, so they should be correct.

bverhagen, 2020-04-08:
I agree that 'deprecating' the current tag (even if only in the description on Gitlab) and creating a new tag pointing to [b8f1c86](https://github.com/ros2/ros2/commit/b6f1c8610e69ba72e89e916e528f4fc7f8d32030) is a sufficient workaround until Path Release 2 is released.

mjcarroll, 2020-04-08:
I have added a new tag to point to the correct commit for patch release 1 versions: https://github.com/ros2/ros2/tree/release-eloquent-20201024-1

bverhagen, 2020-04-10:
One small remark though: I think you made a typo in your tag. Shouldn't it be released on `2020-01-24` rather than `2020-10-24`?

mjcarroll, 2020-04-10:
Wow, you are correct, sorry about that: https://github.com/ros2/ros2/tree/release-eloquent-20200124-1

mjcarroll, 2020-04-10:
I have also made sure that the release notes point to the correctly location.

bverhagen, 2020-04-11:
Thank you very much!

closed 2020-04-11 by bverhagen