ros2/ros2 #895: Ros2 build fails on mojave 10.14.5

hwmaybe, 2020-04-13:
## Bug report

**Required Info:**

- Operating System:
  - mojave (macos 10.14.5)
- Installation type:
  - build from source
- Version or commit hash:
  - latest repos
- DDS implementation:
  - default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
colcon build --symlink-install --packages-select qt_gui_cpp

```

#### Expected behavior
successful build
#### Actual behavior
```
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                             
In file included from siplibqt_gui_cpp_sipQList0101qt_gui_cppPluginDescriptor.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
In file included from siplibqt_gui_cpp_sipqt_gui_cppRosPluginlibPluginProvider_ForPlugins.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
In file included from siplibqt_gui_cpp_sipqt_gui_cppRecursivePluginProvider.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
In file included from siplibqt_gui_cpp_sipQList0101qt_gui_cppPluginProvider.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
In file included from siplibqt_gui_cpp_sipqt_gui_cppRosPluginlibPluginProvider_ForPluginProviders.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
1 error generated.
In file included from siplibqt_gui_cpp_sipQMap0100QString0100QString.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
make[3]: *** [siplibqt_gui_cpp_sipQList0101qt_gui_cppPluginDescriptor.o] Error 1
make[3]: *** Waiting for unfinished jobs....
1 error generated.
1 error generated.
In file included from siplibqt_gui_cpp_sipcmodule.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
In file included from siplibqt_gui_cpp_sipqt_gui_cppSettings.cpp:7:
./sipAPIlibqt_gui_cpp_sip.h:12:10: fatal error: 'QMetaType' file not found
#include <QMetaType>
         ^~~~~~~~~~~
make[3]: *** [siplibqt_gui_cpp_sipQMap0100QString0100QString.o] Error 1
make[3]: *** [siplibqt_gui_cpp_sipQList0101qt_gui_cppPluginProvider.o] Error 1
1 error generated.
1 error generated.
make[3]: *** [siplibqt_gui_cpp_sipqt_gui_cppSettings.o] Error 1
make[3]: *** [siplibqt_gui_cpp_sipcmodule.o] Error 1
1 error generated.
1 error generated.
1 error generated.
make[3]: *** [siplibqt_gui_cpp_sipqt_gui_cppRosPluginlibPluginProvider_ForPluginProviders.o] Error 1
make[3]: *** [siplibqt_gui_cpp_sipqt_gui_cppRosPluginlibPluginProvider_ForPlugins.o] Error 1
make[3]: *** [siplibqt_gui_cpp_sipqt_gui_cppRecursivePluginProvider.o] Error 1
make[2]: *** [src/qt_gui_cpp_sip/libqt_gui_cpp_sip.dylib] Error 2
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp	[ Exited with code 2 ]

Summary: 0 packages finished [1.90s]
  1 package failed: qt_gui_cpp
  1 package had stderr output: qt_gui_cpp
```
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hwmaybe, 2020-04-13:
It looks like that complier can't find the header file <QMetaType>. But I do this following the instructions of tutorials. I search all days for solving the problems, but fails. The feeling is bad. 
Who can help me?

nuclearsandwich, 2020-04-16:
Can you please link the tutorials that you're following?

It looks like there may be a discrepancy between our instructions for building on macos and for installing the binaries. The build instructions are missing info for RQt https://index.ros.org/doc/ros2/Installation/Foxy/macOS-Install-Binary/#system-requirements

zmk5, 2020-04-22:
I have eloquent installed on Catalina from source. One thing I noticed is that you have to follow the instructions from homebrew about linking packages you installed. There should be a set up instructions after the package is installed. You can use `brew doc` I believe to check if things are linked properly.

nuclearsandwich, 2020-04-23:
> I have eloquent installed on Catalina from source

@zmk5 if there are specific packages that required linking would you please open a PR against our installation docs (found here: https://github.com/ros2/ros2_documentation/tree/master/source/Installation/Foxy) However I should also notify you that Catalina is not a supported platform by Eloquent or any other ROS 2 distribution so changes that don't apply to Mojave as well as Catalina may not be merged.

zmk5, 2020-04-30:
@nuclearsandwich Hey, turns out I did do a pull request for linking the proper Qt5 library to PyQt5.

@hwmaybe I did the following to get it to work. I first made sure that `sip` and `pyqt5` were installed using homebrew.

``` bash
~$ brew install sip pyqt5
```

then, I linked `sip` Qt5 to the PyQt5 folder using this command:

``ln -s /usr/local/share/sip/Qt5 /usr/local/share/sip/PyQt5``

NOTE: If you had previously done the above and you still run into trouble, then you may need to reinstall `sip` and PyQt5. Once uninstalled delete the  `/user/local/share/sip/PyQt5` directory. Now reinstall `sip` and `pyqt5` and then link them with the above command.

hwmaybe, 2020-10-27:
Thank you for your help @zmk5 !

closed 2020-10-27 by hwmaybe