ros2/ros2 #898: Track released version of ros2_tracing

nuclearsandwich, 2020-04-15:
In an upcoming PR I'll pin a commit of ros2_tracing which fixes a deprecation warning currently affecting CI buids. This issue tracks lifting that pin and returning to the official release track of ros2_tracing.

rotu, 2020-04-16:
As mentioned in #899, if we're going to include it in ros2, we may as well use master. Despite @dirk-thomas, just because it lives outside of github doesn't mean the `master` branch is less stable or suitable than that of other projects.

dirk-thomas, 2020-04-16:
@rotu Please don't ignore feedback you get on other tickets - I commented twice on the referenced ticket.

There is a rational *why* we chose to use a tag in the past. Just because you feel differently isn't sufficient to change the decision. You need to bring up good enough arguments why the rational should be changed. And also consider what the impact of the proposed change is (e.g. if something break we don't have access to that repo is a pretty big deal). Changes from this third-party repo are gated on a pull request which explicitly levels the version and is usually with a set of CI builds showing that it is regression free.

rotu, 2020-04-16:
I did read your feedback and I discovered this issue, which is, I think a better place to present the feedback. You presented two rationales:

* "changes may regress the build" and "you don't have write access to the repo"
    *If something breaks, you can hit up the maintainers to fix it swiftly or pin back the repos file to a working version.
* "This repo doesn't have CI"
    * Adding CI to this repo is something that can be done. https://docs.gitlab.com/ee/integration/jenkins.html Or it can be run on each MR manually like the current process.

Here are the rationales for designating and using a stable branch instead:

* Upgrade path is automated and more obvious. It's silly to have to check out a new ros2.repos just because other ROS2 projects introduce compatibility breaks and/or deprecations or because this project makes improvements. By using a branch, `git pull` or `vcs pull` does that automatically
* Following a branch shows trust in the developer. ROS2 encourages widespread contribution and participation. By not allowing projects we use to specify their own trajectory, you're saying "we don't trust you to release stable, high quality code". It betrays the community principles.

nuclearsandwich, 2020-04-17:
<details><summary>Protocol sidebar</summary>

> I did read your feedback and I discovered this issue, which is, I think a better place to present the feedback.

That's a reasonable conclusion and trying to start conversations in the most appropriate place is really valuable but I think a better default for in-progress conversations is to "play them where they land" and keep a conversation going in the same thread unless explicitly asked by a maintainer or other project member to relocate the discussion. It's much easier to reference a complete discussion than to try and follow one between multiple threads.

How that we're here though. Let's continue.
</details>

In https://github.com/ros2/ros2/pull/748 which introduced tracetools to the repos file, this discussion was had and between the ROS 2 team and the tracetools contributors the conclusion at the time was that tagged releases were the best compromise. I think reviewing that discussion is going to better prepare anyone participating in this one but I'll try for a brief summary:

* In general we're trying to keep dependencies tight and reliable. (Recently we even removed poco, an example in the earlier discussion of a widely used library, in favor of a more tightly scoped internal implementation of the needed features.)
* A major concern when reviewing a branch based inclusion was the fact that their change control process is not *exactly our process, that is, they don't run our official CI for each change. The problem is not resolved by independent CI.
* The above concern makes it possible for a change to make it through an external review and automated testing process but still break our builds. And that compounds with the fact that as none of us are maintainers of the external project, we can't change the upstream in order to resolve the issue. Our only option for resolution is re-targeting either to an unreleased commit as in #899 or pinning back to a previously known good commit.

---

> You presented two rationales:
> 
>     * "changes may regress the build" and "you don't have write access to the repo"
>      ...
>     * "This repo doesn't have CI"

I think these positions are well covered by my summary above. What I had to think about is how and _why_ our position here is different than it is for many other aspects of our CI configuration. Tracking volatile sources is definitely our default. All our python packages from pypi are tracking volatile release tracks rather than frozen versions with explicit upgrades. We're pinned to a stable release channel for our host operating system but we don't locally cache apt repositories and only upgrade after tests, so why are we **reactive** in those cases but preferring a **proactive** approach here? The strongest rationale that I see is because of all of the above, only things built inside the repos file are likely to actually cause _build_ failures. rather than

dirk-thomas, 2020-04-17:
> If something breaks, you can hit up the maintainers to fix it swiftly or pin back the repos file to a working version.

The point is that we don't want such an expected build failure to happen in the first place. In the duration until we are able to pin and/or get upstream to address the problem a lot of developers are negatively impacted in their workflow. We have had that in the past a few times (not specifically on this repo but other similar ones) and that drove the decision to use a pinned version for third-party repos.

> Adding CI to this repo is something that can be done. https://docs.gitlab.com/ee/integration/jenkins.html Or it can be run on each MR manually like the current process.

It could be but currently that simply is not the case (and we have to deal with the current state and not some desired but fictional case).

> Following a branch shows trust in the developer. ROS2 encourages widespread contribution and participation. By not allowing projects we use to specify their own trajectory, you're saying "we don't trust you to release stable, high quality code". It betrays the community principles.

The past decision to use a tagged version was made in consensus with the upstream maintainer. And whenever the maintainers where ready they quickly proposed a bump which was merged after passing CI to avoid surprises.

So I am not sure where you take the idea from that we are "betraying the community principles". Anyway I find this phrase pretty misplaced and violating the [ROS communication guidelines](https://discourse.ros.org/faq) but that might be my subjective opinion. I will just refrain from this topic fro now on.

rotu, 2020-04-17:
Thank you, @nuclearsandwich. I had no idea about the context and, at the very least, it's good that this was a considered decision.

> All our python packages from pypi are tracking volatile release tracks rather than frozen versions with explicit upgrades. We're pinned to a stable release channel for our host operating system but we don't locally cache apt repositories and only upgrade after tests, so why are we **reactive** in those cases but preferring a **proactive** approach here? The strongest rationale that I see is because of all of the above, only things built inside the repos file are likely to actually cause _build_ failures rather than test failures

I didn't realize the subtlety between "this project has a high risk of breaking the build" versus "this project has a high risk of causing test failures" was a consideration here. Having "green CI" seemed like an all or nothing thing. 

> This is not an interpretation of the existing policy that gains purchase with me at all. But I and the other ROS team members at Open Robotics are working from a place of power relative to other contributors to the project and personally I would be very interested in learning if other contributors are put off by this policy.

Maybe I'm an outlier? I certainly know that the walled-garden nature has been a big putoff for me and I've always felt like a ROS outsider from day 1.

Figuring out whether to look at ci.ros.org or build.ros.org for a build and learning how to investigate failures (which often involves looking at consoleText) instead of a dashboard, or waiting for PRs to get reviewed, or waiting on green for an approved PR to get merged and not being able to do anything about it have all been big points of frustration. I think it's really important to explain these tools and the expectations around them, and provide opportunities to learn about them through experimentation.

Like vcs, colcon, ament, tf2, bloom, PR synchronization, CI it seems like something that everybody expects you to know but nobody puts into a friendly, accessible explanation. At this point, I still feel like a total noob but I probably could contribute to the developers' guide to demystify CI for others (with the help of [Cunningham's Law](https://en.wikipedia.org/wiki/Ward_Cunningham#Cunningham's_Law)).

> Speaking for myself this point seems particularly disingenuous coming from you, as you and I along with several others are in the process of updating branches (from pinned commits for released distros) for repositories you're contributing to in #894 and #891.

I'm very much still figuring this out! I understand code, but (1) I'm a beginner to open source development (2) I'm a newcomer to ROS with little institutional knowledge and understanding of what balance ROS strikes between stability and innovation.

> The decision to target tagged releases for ros2_tracing was reached in mutual agreement with the contributors maintaining that repository.

I can see that now. I had assumed this was imposed on that project by fiat and that nobody cared enough to fix the state of things.

rotu, 2020-04-17:
> We have had that in the past a few times (not specifically on this repo but other similar ones) and that drove the decision to use a pinned version for third-party repos.
> It could be but currently that simply is not the case (and we have to deal with the current state and not some desired but fictional case).

This sounds like the exact attitude of lack of trust that drives contributors away. Teach people what our integration standards are make the tools to meet those standards are. I assume bad integration experiences were more from unclear and over-complex processes, rather than from carelessness.

> The past decision to use a tagged version was made in consensus with the upstream maintainer.

Yep I totally missed this bit and that's how I misunderstood your intention with keeping it at a tag/hash.

> So I am not sure where you take the idea from that we are "betraying the community principles". Anyway I find this phrase pretty misplaced and violating the [ROS communication [...]

This certainly *sounds* like a lack of trust in that team's ability to commit high-quality code: "Before we used a tag because we *don't* want to track master and be exposed to any changes landing there and potentially regressing the build. That rational hasn't changed."

Here's what the developer guide says, and it's the [community principle](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#general-principles) that I thought you were balking at:

> **Shared ownership:** Everybody working on ROS 2 should feel ownership over all parts of the system. The original author of a chunk of code does not have any special permission or obligation to control or maintain that chunk of code. Everyone is free to propose changes anywhere, to handle any type of ticket, and to review any pull request.

I did not mean insult your honor or your intentions, and if I was out of line in what I said, I'm sorry.

christophebedard, 2020-04-24:
I'd like to get two small changes in (not really feature-related) before tagging a release for Foxy, but unfortunately I have been unable to find reviewers. I'd appreciate if someone could look at these:

* https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/merge_requests/164
* https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/merge_requests/167

PR #909 (2020-04-26): Update ros2_tracing to 1.0.0
Closes #898 

Released: https://github.com/ros/rosdistro/pull/24573
closed 2020-04-26 by jacobperron