ros2/ros2 #900: mark flaky / broken tests in a way to allow skipping them

dirk-thomas, 2020-04-16:
`pytest` already defines a marker for exactly that purpose: [pytest.mark.xfail](http://doc.pytest.org/en/latest/reference.html#pytest-mark-xfail-ref). I would suggest to use that for Python tests rather than coming up with a custom name. In CTest we can use a label with the same name (`xfail`).

The triggered CI / nightly jobs should be updated to skip tests with that marker / label. It probably makes sense to have a set of (nightly) jobs which only run those tests to keep track of the number of tests marker as such (as well as their status).

Beyond this technical approach there needs to be a "rule / policy" that tests marked like this (and therefore effectively being disabled) to spend resources on these tests to be able to reenable them.

dirk-thomas, 2020-04-16:
@ablasdel @dejanpan @gbiggs FYI

mjcarroll, 2020-04-16:
I like this idea for keeping CI and nightly results more readable.

If possible I think it would make sense to extend it to include a reason (which the Python API seems to have).  This way if we have tests that are known to be flaky on one operating system or RMW vendor configuration, we can still detect if there are regressions in other platforms/configurations.  This wouldn't have to be anything complicated, just appending platform or RMW configuration name would be sufficient in most cases, I believe.

dirk-thomas, 2020-04-16:
> This way if we have tests that are known to be flaky on one operating system or RMW vendor configuration, we can still detect if there are regressions in other platforms/configurations. This [...]

I would argue that platform / RMW specific failures are different. I would consider the `xfail` marker only for tests which fail due to how they are written / what they are testing. If a tests is "perfect" on one platform but flaky / broken on another I would argue it isn't the tests fault. The test could be marked "skipped" for a specific platform / RMW which would be orthogonal.

mjcarroll, 2020-04-16:
>  This is going to make such test failures “more acceptable” and likely go unfixed longer 

I think that tracking these separately makes a more explicit "scoreboard" of flaky tests.  This makes it more explicit and visible and presumably, more likely to get fixed.  I think that it's reasonable that this would be incorporated into the rule/policy that @dirk-thomas mentioned above.

dirk-thomas, 2020-04-16:
> does it make sense to have to check this into the test repo rather than a supplementary resource outside the test repo (a la ros2.repos)?

I don't understand what "check this into the test repo" means?

> We already ignore pytest warnings and it may make sense for some tests to downgrade the severity.

This is a completely orthogonal topic. Please keep it separate - there is already a ticket about that: ros2/ci#418.

> I don’t like that this reduces the impetus to fix broken tests.

This ticket is about the technical aproach to mark such tests. As I already explicitly stated in the original description: "Beyond this technical approach there needs to be a "rule / policy" that tests marked like this (and therefore effectively being disabled) to spend resources on these tests to be able to reenable them."

ablasdel, 2020-04-16:
I'm all for using existing names/info so I like "xfail" as expected to fail.

Identifying them like this and only running them when we want to run flakey tests is the best of all possible worlds in the "disabling tests" discussion. For metrics sake we should probably only track "non-xfail" tests in our code coverage metrics.

rotu, 2020-04-16:
I also worry that test suppression is going to bury the underlying issue. For instance, I found a big stability bug https://github.com/ros2/rosbag2/issues/329 which rosbag2 did a workaround for. I tried escalating to rclcpp but the issue seems unfixed and unacknowledged. I can’t tell whether test suppression will make the issue of “fix the test, but leave the bug” better or worse.

clalancette, 2020-04-16:
> I tried escalating to rclcpp but the issue seems unfixed and unacknowledged.

For what its worth, we do agree that there is a bug and it needs to be fixed.  We are pretty slammed trying to get Foxy API freeze done, so we haven't really had time to look into it.  As usual, any help there would be appreciated.

dirk-thomas, 2020-04-16:
Python tests marked with `xfail` are run (if not excluded explicitly as in ros2/ci#434) but marked with `x` if they fail as expected (`X` if they unexpectedly pass). Either way they don't show up as a failed test in the generated xml. When invoking `pytest` with `--runxfail` even tests with that marker are being run and reported as usually (pretending the `xfail` marker doesn't exist).

While CTest has the concept of labels it doesn't have a feature like `xfail` in `pytest`. So when not passing custom arguments tests with that label will run and be reported as any other test. The label only allows to skip the test when passing `-LE xfail` which is done in ros2/ci#434 but wouldn't be the case in a normal local run.

gbiggs, 2020-04-20:
This may be picking weeds (or it may not? the correct metaphor escapes me right now), but I'm a little concerned about the description given for `xfail`: "expected to fail". I don't think our tests should be _expected_ to fail. They're expected to work but they're broken. That's different from "I changed the API so this will fail for a while until I fix the tests".

Still, it's not a strong argument against using the existing flag since there's also infrastructure support for it.

rotu, 2020-04-20:
I think xfail *does* mean expected to fail. An xfail test is for an unfixed bug or an unimplemented feature - if a change causes a test to start failing, marking it as xfail is not the proper resolution.

IMO a test result should always tell you something. If you don't expect the test to either succeed or fail, there's not really a reason to be running the test in the first place.

rotu, 2020-04-20:
Also, I think an `xfail` test should be logging a low priority but visible warning if it *does* pass so that you can enable the test or figure out why it passed spuriously. If the test result is ignorable, then use `skip` instead.

gbiggs, 2020-04-20:
> An xfail test is for an unfixed bug or an unimplemented feature

OK, perhaps I should have said our case is different from "I implemented the tests for this feature but the feature itself isn't done yet". Nevertheless my point stands. Our tests are _not_ expected to fail. We're talking about [broken (previously called "flaky") tests](https://discourse.ros.org/t/tackling-of-flaky-tests/13686), and hiding them without at least some kind of deadline or process is important.

Fortunately @dirk-thomas already agrees on the need for a process so this is really about semantics and whether we want to use the tag for its intended purpose separately from our broken tests that should not be broken.

rotu, 2020-04-20:
[Agreed.](https://github.com/ros2/ros2/issues/900#issuecomment-614764177) This doesn't make test failures go away, and so IMO it provides an inadequate solution to the real problem of broken tests:

> 3\. I don’t like that this reduces the impetus to fix broken tests. E.g. timer tests tend to fail, more under some systems than others. But afaict, even on the “non-flaky” systems, they start failing when the CPU is under load. This is going to make such test failures “more acceptable” and likely go unfixed longer (not that they’re being fixed quickly today...)

It would be really nice to categorize failures (including issues *not related to a test case*) by whether their triage/fix state. I don't know how to do this and was hoping to elicit some guidance from @dejanpan on discourse:
https://discourse.ros.org/t/tackling-of-flaky-tests/13686/2?u=rotu

dirk-thomas, 2020-04-20:
The current state is that we have a marker / label (`xfail`) which can be used to not process a test when running CI on ci.ros2.org (and soon also build.ros2.org) and that was the requested ability in the TSC meeting. There are separate nightly jobs on ci.ros2.org (ending in `xfail`) which run those job so their state as well as their histooy can be easily checked.

If anyone would like to propose a different name than `xfail` please feel free to do so. The changes in the referenced PRs should be trivial to update to a new name. If we don't want to use `xfail` be aware that it means the Python tests will run by default when testing locally (which is fine with me since the same is already the case for CTest).

gbiggs, 2020-04-21:
`xbroken` is a better name to me, but I don't have a plan for how to implement more than just having a tag. We need the functionality of not interrupting CI, making sure these things don't get forgotten, etc., and a plan for how to get all the existing tests to not be broken any more.

rotu, 2020-04-21:
I guess, but `xbroken` is no better than `skip`.

I would consider this in light of the broader problem. Rather than finding a way to express it in every testing framework, curate a list of testing exceptions in an external metadata file.

Such a file would have predicates like “skip the test named X on windows” or “skip tests in package Z when using RMW W”. It seems to me like (1) this could be extended to non-test failures (2) this allows filtering by things the test should not have to be aware of (like which OS or which CI runner). (3) putting some metadata on each rule could allow CI to know where the issue is documented, and potentially to automatically re-enable a test when an issue is closed.

gbiggs, 2020-04-21:
That approach would work, but I feel that the approach taken by Apex (quarantining tests using a macro and a timeout, and tracking those in a test tracking tool) is easier to use.

rotu, 2020-04-21:
Apex uses TestRail to track test history. This tool seems to already implement the features I’m describing: filter broken tests and annotate test results with a link to the issue.

The reason I disfavor macros is that they have to be in place *beforehand*. So if you get a bunch of failures, you can’t classify them after the fact - you have to rerun CI instead.

dirk-thomas, 2020-04-27:
@dejanpan @ablasdel @chapulina @gbiggs In the TSC meeting this topic sounded very urgent so I spent the time to provide a way to mark and avoid running flaky tests. (Please feel free to propose changes to rename the marker / label.)

Yet until now there has been no action to mark any tests as such. So I am not sure if people are actually interested in this. Without any further input I am inclined to close this ticket since the described goal is now technically possible.

hidmic, 2020-05-14:
> Without any further input I am inclined to close this ticket since the described goal is now technically possible.

@dirk-thomas do you think this can be closed now then?

closed 2020-05-14 by dirk-thomas