ros2/ros2 #904: enable modern CMake usage

dirk-thomas, 2020-04-21:
This is a meta ticket to track progress toward modern CMake (exporting targets and using interface targets) instead of classic CMake (CMake variables ending in `_INCLUDE_DIRS` and `_LIBRARIES`).

dirk-thomas, 2020-04-23:
With the above referenced changes it is possible to build e.g. `demo_nodes_cpp` without using classic CMake variables within the ROS packages (external system dependencies might not provide modern CMake targets but only offer classic CMake variables).

To ensure that this is actually the case I made the following `ament_cmake` functions no-ops:
* `ament_export_definitions`
* `ament_export_include_directories`
* `ament_export_libraries`
* `ament_export_link_flags`

dirk-thomas, 2020-05-01:
See ros2/ros2#915 for related fixes.

dirk-thomas, 2020-05-04:
With the PR for the Fox migration notes ready and the `rosbag2` follow up ticketed I will go ahead and close this.

closed 2020-05-04 by dirk-thomas

RFRIEDM-Trimble, 2022-07-14:
Big win, ROSIDL just merged this PR adding alias targets, and it's being backported to Humble. Most ROS2 packages should now be able to call `target_link_libraries` with ALIAS targets for all their dependencies instead of `ament_target_dependencies`.
https://github.com/ros2/rosidl/pull/605

Ryanf55, 2023-07-27:
Found another blocker to modern CMake - the location that ROS 2 installs CMake executables (nodes) is nonstandard and contradicts the GNUInstallDirs recommendations: 
https://github.com/ros2/ros2cli/issues/845
The attached installation code there is the latest demo of how I use CMake+ament (very little `ament`)