ros2/ros2 #908: ros2-dashing  RMW_IMPLEMENTATION error

aurora12344, 2020-04-24:
Hi :
When I run  RMW_IMPLEMENTATION=rmw_connext_cpp ros2 run demo_nodes_cpp talker  , Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connext_cpp'), exiting with 1.  there is a cyclonedds file in ros2_dashing/install.  Why do have an execution error?

Thanks.
Yan

hidmic, 2020-04-24:
Hmm `rmw_connext_cpp != rmw_cyclonedds_cpp`.

@aurora12344 this sort of questions are better suited for [`answers.ros.org`](https://answers.ros.org/). If you're still facing trouble after sorting out what RMW implementation you actually intend to use, consider posting a question there. I'll close this for now.

closed 2020-04-24 by hidmic

fujitatomoya, 2020-04-24:
@aurora12344 

i guess you could check https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Binary/#installing-the-missing-dependencies.