ros2/ros2 #914: Source Build still makes packages within install directory

zmk5, 2020-04-29:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10, macOS Mojave, macOS Catalina 
- Installation type:
  - From Source
- Version or commit hash:
  - Eloquent, Foxy
- DDS implementation:
  - N/A
- Client library (if applicable):
  - `rttest`, `tlsf`, `tlsf_cpp`, `pendulum_control`

#### Steps to reproduce issue

Issue is reproduced by building a ros2 release from sourc and attempting to source the installation with `. install/setup.bash`

#### Expected behavior

The expected behavior is for the bash line to go next after loading the ros2 packages.

#### Actual behavior

I get 4 packages saying that their respective `local_setup.bash` files are `not found`

```bash
~$ . install/setup.bash
not found: "~/ros2_foxy/install/rttest/share/rttest/local_setup.bash"
not found: "~/ros2_foxy/install/tlsf/share/tlsf/local_setup.bash"
not found: "~/ros2_foxy/install/tlsf_cpp/share/tlsf_cpp/local_setup.bash"
not found: "~/ros2_foxy/install/pendulum_control/share/pendulum_control/local_setup.bash"
~$
```

#### Additional information

While checking the `CMakeLists.txt` files for the above packages, I noticed that these don't build in macOS, Windows, and/or Android. However, `colcon` still creates package directories within the `install` directory. This just seems like a minor bug/inconvenience for installation, but this could through off people who attempt to install ros2 on a non-linux machine.
----
## Feature request

#### Feature description

Something added to the installation documentation to show a command/fix that prevents package directories creation in `install` during building from source if the package isn't supported on the platform.

#### Implementation considerations

N/A

hidmic, 2020-05-11:
Thanks for reporting ! At the moment, it is unclear how to avoid this situation as `colcon` cannot tell what any given package did or did not do at build time. Suggestions on how to mitigate this situation are most welcomed.

dirk-thomas, 2020-09-22:
This has been addressed in the latest release of `colcon-ros` 0.3.20.

closed 2020-09-22 by dirk-thomas

zmk5, 2020-09-24:
Thanks @dirk-thomas!