ros2/ros2 #915: Overlays fail to build against nightly Linux packaging job 1844

clalancette, 2020-04-29:
## Bug report

I downloaded the binary packaging job from https://ci.ros2.org/view/packaging/job/packaging_linux/1844/ .  I then tried to build an overlay workspace on top of it, just rebuilding https://github.com/ros2/geometry2 on top (hash https://github.com/ros2/geometry2/commit/b8b9e7e1ca3b0fedf0da9f43e7ea684777131674 with no changes).  When trying to build in that scenario, I get a build error while building tf2_msgs:

```
make[2]: *** No rule to make target '/home/jenkins-agent/workspace/packaging_linux/ws/install/lib/libstd_msgs__rosidl_typesupport_introspection_cpp.so', needed by 'libtf2_msgs__rosidl_typesupport_fastrtps_cpp.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
```

**Required Info:**

- Operating System:
  - Ubuntu Linux (tested on both 18.04 and 20.04)
- Installation type:
  - Binary nightly packaging job https://ci.ros2.org/view/packaging/job/packaging_linux/1844/
- Version or commit hash:
  - N/A
- DDS implementation:
  - All of Fast-RTPS, Connext, and CycloneDDS are enabled
- Client library (if applicable):
  - rosidl_generator

#### Steps to reproduce issue
Download and unpack the latest nightly build:
```
wget https://ci.ros2.org/view/packaging/job/packaging_linux/1844/artifact/ws/ros2-package-linux-x86_64.tar.bz2
tar -xvf ros2-package-linux-x86_64.tar.bz2
```

Make an overlay with geometry2 in it:
```
mkdir -p custom_tf/src
cd custom_tf/src
git clone https://github.com/ros2/geometry2
cd ..
```

Build the overlay:
```
colcon build --event-handlers console_direct+ --executor sequential
```

#### Expected behavior

Successful build of the overlay.

#### Actual behavior

While building tf2_msgs, I see messages like the following:

```
Starting >>> tf2_msgs
-- The C compiler identification is GNU 9.3.0 
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done       
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done     
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.1 (/home/ubuntu/ros2-linux/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Found rosidl_default_generators: 0.9.0 (/home/ubuntu/ros2-linux/share/rosidl_default_generators/cmake)
-- Found RTI Connext: /opt/rti.com/rti_connext_dds-5.3.1
-- Found rosidl_adapter: 0.9.0 (/home/ubuntu/ros2-linux/share/rosidl_adapter/cmake)
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.1.1f")  
-- Found FastRTPS: /home/ubuntu/ros2-linux/include  
-- Found builtin_interfaces: 0.9.0 (/home/ubuntu/ros2-linux/share/builtin_interfaces/cmake)
-- Found action_msgs: 0.9.0 (/home/ubuntu/ros2-linux/share/action_msgs/cmake)
-- Found std_msgs: 0.9.0 (/home/ubuntu/ros2-linux/share/std_msgs/cmake)
-- Found geometry_msgs: 0.9.0 (/home/ubuntu/ros2-linux/share/geometry_msgs/cmake)
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.2") 
-- Using all available rosidl_typesupport_c: rosidl_typesupport_connext_c;rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Found ament_cmake_cppcheck: 0.9.1 (/home/ubuntu/ros2-linux/share/ament_cmake_cppcheck/cmake)
-- Found ament_cmake_cpplint: 0.9.1 (/home/ubuntu/ros2-linux/share/ament_cmake_cpplint/cmake)
-- Found ament_cmake_uncrustify: 0.9.1 (/home/ubuntu/ros2-linux/share/ament_cmake_uncrustify/cmake)
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_connext_cpp;rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found ament_cmake_ros: 0.9.0 (/home/ubuntu/ros2-linux/share/ament_cmake_ros/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.2", minimum required is "3.5") 
-- Found python_cmake_module: 0.8.0 (/home/ubuntu/ros2-linux/share/python_cmake_module/cmake)
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable version "3.8.2", minimum required is "3.5") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.8
-- Using PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- Found PythonExtra: .so  
-- Found ament_cmake_flake8: 0.9.1 (/home/ubuntu/ros2-linux/share/ament_cmake_flake8/cmake)
-- Found ament_cmake_pep257: 0.9.1 (/home/ubuntu/ros2-linux/share/ament_cmake_pep257/cmake)
-- Configuring done                            
-- Generating done                             
-- Build files have been written to: /home/ubuntu/custom_tf/build/tf2_msgs
Scanning dependencies of target tf2_msgs__cpp  
[  1%] Generating C code for ROS interfaces
Scanning dependencies of target tf2_msgs__dds_connext_idl
[  2%] Generating C++ code for ROS interfaces
[  3%] Generating DDS interfaces
[  3%] Built target tf2_msgs__dds_connext_idl     
[  3%] Built target tf2_msgs__cpp                 
Scanning dependencies of target tf2_msgs__rosidl_generator_c
[  4%] Generating C++ type support dispatch for ROS interfaces
[  5%] Generating C++ type support for eProsima Fast-RTPS
[  6%] Generating C++ introspection for ROS interfaces
[  7%] Generating C++ type support for RTI Connext (using '/opt/rti.com/rti_connext_dds-5.3.1/bin/rtiddsgen_server')
[  8%] Building C object CMakeFiles/tf2_msgs__rosidl_generator_c.dir/rosidl_generator_c/tf2_msgs/msg/detail/tf_message__functions.c.o
[  9%] Building C object CMakeFiles/tf2_msgs__rosidl_generator_c.dir/rosidl_generator_c/tf2_msgs/srv/detail/frame_graph__functions.c.o
[ 10%] Building C object CMakeFiles/tf2_msgs__rosidl_generator_c.dir/rosidl_generator_c/tf2_msgs/msg/detail/tf2_error__functions.c.o
[ 11%] Building C object CMakeFiles/tf2_msgs__rosidl_generator_c.dir/rosidl_generator_c/tf2_msgs/action/detail/lookup_transform__functions.c.o
[ 12%] Linking C shared library libtf2_msgs__rosidl_generator_c.so
[ 12%] Built target tf2_msgs__rosidl_generator_c   
[ 13%] Generating C type support dispatch for ROS interfaces
[ 14%] Generating C introspection for ROS interfaces
Scanning dependencies of target tf2_msgs__rosidl_typesupport_cpp
[ 15%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_cpp.dir/rosidl_typesupport_cpp/tf2_msgs/msg/tf2_error__type_support.cpp.o
[ 17%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_cpp.dir/rosidl_typesupport_cpp/tf2_msgs/srv/frame_graph__type_support.cpp.o
[ 17%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_cpp.dir/rosidl_typesupport_cpp/tf2_msgs/action/lookup_transform__type_support.cpp.o
[ 18%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_cpp.dir/rosidl_typesupport_cpp/tf2_msgs/msg/tf_message__type_support.cpp.o
Scanning dependencies of target tf2_msgs__rosidl_typesupport_introspection_cpp
[ 19%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/tf2_msgs/msg/detail/tf2_error__type_support.cpp.o
[ 20%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/tf2_msgs/msg/detail/tf_message__type_support.cpp.o
[ 21%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/tf2_msgs/srv/detail/frame_graph__type_support.cpp.o
[ 22%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/tf2_msgs/action/detail/lookup_transform__type_support.cpp.o
Scanning dependencies of target tf2_msgs__rosidl_typesupport_c
[ 23%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_c.dir/rosidl_typesupport_c/tf2_msgs/msg/tf2_error__type_support.cpp.o
[ 24%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_c.dir/rosidl_typesupport_c/tf2_msgs/msg/tf_message__type_support.cpp.o
[ 25%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_c.dir/rosidl_typesupport_c/tf2_msgs/srv/frame_graph__type_support.cpp.o
Scanning dependencies of target tf2_msgs__rosidl_typesupport_fastrtps_cpp
[ 26%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_c.dir/rosidl_typesupport_c/tf2_msgs/action/lookup_transform__type_support.cpp.o
[ 27%] Building CXX object CMakeFiles/tf2_msgs__rosidl_typesupport_fastrtps_cpp.dir/rosidl_typesupport_fastrtps_cpp/tf2_msgs/msg/dds_fastrtps/tf2_error__type_support.cpp.o
make[2]: *** No rule to make target '/home/jenkins-agent/workspace/packaging_linux/ws/install/lib/libstd_msgs__rosidl_typesupport_introspection_cpp.so', needed by 'libtf2_msgs__rosidl_typesupport_fastrtps_cpp.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
```

dirk-thomas, 2020-04-29:
I can reproduce the problem. The reason is the absolute library path `/home/jenkins-agent/workspace/packaging_linux/ws/install/lib/libstd_msgs__rosidl_typesupport_introspection_cpp.so` from the build machine.

dirk-thomas, 2020-04-29:
I can also reproduce it with a local underlay workspace if I relocate the install space of the underlay before sourcing it and using it for the overlay (which is what is happening with the archive from the packaging job if you don't extract it into the same absolute path it was located in on Jenkins).

dirk-thomas, 2020-04-29:
Locally ament/ament_cmake#251 resolves the problem for me since it makes sure that the exported targets only have interface target names rather than absolute include directories.

dirk-thomas, 2020-04-30:
The `rosidl_typesupport_introspection_c|cpp` libraries are still referenced by an absolute path.

dirk-thomas, 2020-04-30:
With the referenced PR merged the number of absolute paths embedded in the archive has been reduced significantly. The following ones are remaining:

<details>

<summary>All absolute paths in `.cmake` files</summary>

```
lib/urdfdom/cmake/urdfdom-config.cmake:
   6: set(urdfdom_INCLUDE_DIRS "/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/include" "/usr/include")
  12:     PATHS "/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib"
  15:     PATHS "/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib"

lib/urdfdom_headers/cmake/urdfdom_headers-config.cmake:
  6: set(urdfdom_headers_INCLUDE_DIRS "/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/include")

share/kdl_parser/cmake/kdl_parserExport.cmake:
  57:   INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/include;/usr/include;/usr/include/eigen3"

share/laser_geometry/cmake/laser_geometryExport.cmake:
  57:   INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/usr/include/eigen3;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/include;/usr/include;${_IMPORT_PREFIX}/include"
  58:   INTERFACE_LINK_LIBRARIES "rclcpp::rclcpp;sensor_msgs::sensor_msgs__rosidl_generator_cpp;sensor_msgs::sensor_msgs__rosidl_generator_c;sensor_msgs::sensor_msgs__rosidl_typesupport_introspection_cpp;sensor_msgs::sensor_msgs__rosidl_typesupport_introspection_c;sensor_msgs::sensor_msgs__rosidl_typesupport_cpp;sensor_msgs::sensor_msgs__rosidl_typesupport_c;console_bridge::console_bridge;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libgeometry_msgs__rosidl_generator_c.so;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libgeometry_msgs__rosidl_typesupport_introspection_cpp.so;rosidl_typesupport_introspection_cpp::rosidl_typesupport_introspection_cpp;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libgeometry_msgs__rosidl_typesupport_introspection_c.so;geometry_msgs::geometry_msgs__rosidl_generator_c;rosidl_typesupport_introspection_c::rosidl_typesupport_introspection_c;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libgeometry_msgs__rosidl_typesupport_cpp.so;rosidl_runtime_cpp::rosidl_runtime_cpp;rosidl_typesupport_cpp::rosidl_typesupport_cpp;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libgeometry_msgs__rosidl_typesupport_c.so;rosidl_runtime_c::rosidl_runtime_c;rosidl_typesupport_c::rosidl_typesupport_c;rosidl_typesupport_interface::rosidl_typesupport_interface;std_msgs::std_msgs__rosidl_generator_cpp;std_msgs::std_msgs__rosidl_generator_c;std_msgs::std_msgs__rosidl_typesupport_introspection_cpp;std_msgs::std_msgs__rosidl_typesupport_introspection_c;std_msgs::std_msgs__rosidl_typesupport_cpp;std_msgs::std_msgs__rosidl_typesupport_c;builtin_interfaces::builtin_interfaces__rosidl_generator_cpp;builtin_interfaces::builtin_interfaces__rosidl_generator_c;builtin_interfaces::builtin_interfaces__rosidl_typesupport_introspection_cpp;builtin_interfaces::builtin_interfaces__rosidl_typesupport_introspection_c;builtin_interfaces::builtin_interfaces__rosidl_typesupport_cpp;builtin_interfaces::builtin_interfaces__rosidl_typesupport_c;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librcutils.so;dl;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libtf2.so"

share/rcl_yaml_param_parser/cmake/rcl_yaml_param_parserExport.cmake:
  58:   INTERFACE_LINK_LIBRARIES "rcutils::rcutils;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libyaml.so;yaml"

share/rosbag2_transport/cmake/export_rosbag2_transportExport.cmake:
  57:   INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/include;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/opt/yaml_cpp_vendor/lib/cmake/yaml-cpp/../../../include;/usr/include"
  58:   INTERFACE_LINK_LIBRARIES "rcl::rcl;rclcpp::rclcpp;rcutils::rcutils;rmw::rmw;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librosbag2_compression_zstd.so;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librosbag2_compression.so;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libzstd.so;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librosbag2_cpp.so;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librosbag2_storage.so;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/libclass_loader.so;console_bridge::console_bridge;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librcutils.so;dl;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librcpputils.so;/usr/lib/x86_64-linux-gnu/libtinyxml2.so;ament_index_cpp::ament_index_cpp;class_loader::class_loader;rcutils::rcutils;rcpputils::rcpputils;stdc++fs;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librosidl_runtime_c.so;rosidl_typesupport_interface::rosidl_typesupport_interface;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/librosidl_typesupport_introspection_cpp.so;rosidl_runtime_cpp::rosidl_runtime_cpp;rosidl_typesupport_introspection_c::rosidl_typesupport_introspection_c;yaml-cpp"

share/rviz_common/cmake/rviz_commonExport.cmake:
  58:   INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/include;/usr/include;/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/include;/usr/include;/home/jenk [... cut ...] jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/liburdfdom_world.so>;\$<\$<CONFIG:Debug>:/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/lib/liburdfdom_world.so>;console_bridge::console_bridge;/usr/lib/x86_64-linux-gnu/libtinyxml.so"
```

</details>

The packages with issues:

* `kdl_parser`
* `laser_geometry`
* `rcl_yaml_param_parser` / `yaml`
* `rosbag2_transport`
* `rviz_common`
* `rviz_rendering`
* `rviz_visual_testing_framework`
* `urdf`
* `urdfdom`
* `urdfdom_headers`

rotu, 2020-05-01:
It seems all the related PRs were merged without running CI! Has there been a change to the PR policy?

dirk-thomas, 2020-05-04:
The packages part of ROS core / base have been updated. There are two areas pending:

* `rosbag2` et al:

@Karsten1987 @ros2/tooling_working_group Can you please follow up on making these relocatable.

* `rviz` et al:

@wjwwood Can you please follow up on making these relocatable.

https://ci.ros2.org/job/ci_packaging_linux/347/ is an isolated packaging build which makes it easier to find usage of `/jenkins-agent/` in any `*.cmake` files. (Usage of absolute paths in `.sh` files is not an issue and not addressable.)

dirk-thomas, 2020-05-04:
I created tickets in the `rosbag2` and `rviz` repos as referenced above. With this I am going to close this ticket since the packaging jobs work and can be used in underlays for the majority of cases.

closed 2020-05-04 by dirk-thomas