ros2/ros2 #917: [Meta-Ticket] Topic Statistics Improvements

dabonnie, 2020-05-01:
Follow up issues from the Topic Statistics addition in Foxy (#901).

## 'rclcpp' Improvements

 - [ ] Use ROS Time instead of `std::chrono` in `Subscription::handle_message`
 - [ ] Use a single publisher for a node's Topic Statistics, instead of a publisher per subscriber
 - [ ] Implement `publisher` Topic Statistics, e.g., add a publish period statistic 

## General Improvements

- [ ] Implement topic statistics in `rcl`

fujitatomoya, 2020-06-03:
@dabonnie 

> Implement topic statistics in rcl

+1

Barry-Xu-2018, 2020-06-03:
@dabonnie  

Do you start to Implement topic statistics in rcl ？Or do you have time schedule on moving to rcl ?

dabonnie, 2020-06-03:
> @dabonnie
> 
> Do you start to Implement topic statistics in rcl ？Or do you have time schedule on moving to rcl ?

Hi @Barry-Xu-2018, we haven't started this implementation (or have it on our [roadmap](https://github.com/ros-tooling/aws-roadmap)) yet.

fujitatomoya, 2020-06-04:
@dabonnie  

would you share your thoughts about the followings?

- statistics for each publication.
  I think that this is useful to check which topic and publication is something wrong precisely.

- agnostic from user implementation
  typically user wants to do just switching on/off this feature, not implementing anything in user code.

Barry-Xu-2018, 2020-06-04:
@dabonnie 

> Hi @Barry-Xu-2018, we haven't started this implementation (or have it on our [roadmap](https://github.com/ros-tooling/aws-roadmap)) yet.

Thanks for sharing current status.

dabonnie, 2020-06-12:
@fujitatomoya 

> statistics for each publication.
> I think that this is useful to check which topic and publication is something wrong precisely.

I'm not sure what you mean if something is wrong with the topic and publication: log a warning if the publishing bounds are exceeded?

> * agnostic from user implementation
>   typically user wants to do just switching on/off this feature, not implementing anything in user code.

Yes, I agree. I think the "Add Topic Statistics Options to be configured via launch file parameters" task captures this, so a user doesn't have to modify the subscription options in code. What do you think?

fujitatomoya, 2020-06-13:
@dabonnie 

> I'm not sure what you mean if something is wrong with the topic and publication

If publisher A and B are publishing data in topic, statistics for each publication would be nice. (can we do this in current implementation? as far as i see, there is no such identification via message.) based on the statistics for each publication we can definitely know which one is publishing data with unexpected frequency or something.

> "Add Topic Statistics Options to be configured via launch file parameters" task captures this, so a user doesn't have to modify the subscription options in code. What do you think?

launch file parameter and arguments would be nice, but as user aspect, i think that user wants to do the following.

- enable giant switch on/off to see the statistics entire whole system. (ros2 system wide switch)
  i think that this is 1st step that user may have. to check system statistics. but user does not want to modify each single launch file or arguments to enable statistics.
- enable topic statistics. (not nodes or endpoint but topic perspective)
  user wants to see the topic statistics. but not want to set the statistics mode for each nodes or endpoint. maybe user just wants to say "Hey I want to see the statistics for specific topic!" then internally related publication and subscription which are related to the topic, then user can see the statistic for the topic.

above are our use cases, I'd really like to hear from the others about the use case.

dabonnie, 2020-06-16:
> @dabonnie
>
>
> If publisher A and B are publishing data in topic, statistics for each publication would be nice. (can we do this in current implementation? as far as i see, there is no such [...]

Thanks for explaining in detail, as it now makes sense to me. I agree that your proposals are worthwhile use cases and would like to hear other opinions as well.

InitialDuan, 2020-11-06:
hi @dabonnie, I was wondering do you have started to implement [Add Topic Statistics Options to be configured via Parameters](https://github.com/ros-tooling/roadmap/issues/1#issue-670105501)？Or do you have a time schedule on it?

dabonnie, 2020-11-06:
Hi @InitialDuan, we currently don't have plans to start work on that feature. Help from the community would be most welcome and we'd be happy to help review any changes.

InitialDuan, 2020-11-06:
@dabonnie Thanks for info. So currently to enable Topic Statistics feature, a user have to modify the subscription options in each subscriber's code, right?

I am new to community and have no experince on conribution, but would love to learn about how it works. Can you estimate the difficulty and workload of its implemention? Is that suited for such a newbie like me? As far as I know, configuring Topic Statistics Options via Parameters is implemented in ROS1. Would it simplify the implemenation in ROS2?

dabonnie, 2020-11-09:
@InitialDuan I don't think it would be too difficult since you can copy other parameter behaviors. This would greatly simplify the ability for someone to use the feature because you are correct that a user needs to modify their subscription options ([see this tutorial](https://index.ros.org/doc/ros2/Tutorials/Topics/Topic-Statistics-Tutorial/).