ros2/ros2 #918: Fails to build qt_gui_cpp with 'Cannot specify include directories for imported target "tinyxml2_vendor"'

fujitatomoya, 2020-05-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/732abee4bee90d352088c4b1a29af68eb2897ad3
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

follow the instruction [here](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/) and then

```
colcon build --symlink-install
```

#### Expected behavior

Build successful.

#### Actual behavior

Fails to build qt_gui_cpp with 'Cannot specify include directories for imported target "tinyxml2_vendor"'

#### Additional information

```
--- stderr: qt_gui_cpp
CMake Error at /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlib-extras.cmake:49 (target_include_directories):
  Cannot specify include directories for imported target "tinyxml2_vendor".
Call Stack (most recent call first):
  /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlibConfig.cmake:41 (include)
  CMakeLists.txt:17 (find_package)

CMake Error at /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlib-extras.cmake:51 (target_link_libraries):
  Cannot specify link libraries for target "tinyxml2_vendor" which is not
  built by this project.
Call Stack (most recent call first):
  /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlibConfig.cmake:41 (include)
  CMakeLists.txt:17 (find_package)
---

```

Anyone facing something similar? Thanks in advance!

fujitatomoya, 2020-05-12:
rosbag2_storage is also failing.

```
--- stderr: rosbag2_storage                                                                     
CMake Error at /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlib-extras.cmake:49 (target_include_directories):
  Cannot specify include directories for imported target "tinyxml2_vendor".
Call Stack (most recent call first):
  /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlibConfig.cmake:41 (include)
  CMakeLists.txt:24 (find_package)

CMake Error at /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlib-extras.cmake:51 (target_link_libraries):
  Cannot specify link libraries for target "tinyxml2_vendor" which is not
  built by this project.
Call Stack (most recent call first):
  /root/docker_ws/ros2_colcon/install/pluginlib/share/pluginlib/cmake/pluginlibConfig.cmake:41 (include)
  CMakeLists.txt:24 (find_package)

---
```

MiguelCompany, 2020-05-12:
I think ros2/tinyxml2_vendor#8 may solve this.

MiguelCompany, 2020-05-12:
And ros2/rosbag2#402 seems also worth to look at.

fujitatomoya, 2020-05-12:
@MiguelCompany 

importing https://github.com/ros2/tinyxml2_vendor/pull/8/files, still does not work. failed to build qt_gui_cpp.

xGonZh10n, 2020-05-13:
@fujitatomoya 
You may upgrade your cmake to v3.11 or later. (see: [https://cmake.org/cmake/help/v3.11/release/3.11.html#commands](https://cmake.org/cmake/help/v3.11/release/3.11.html#commands))

clalancette, 2020-05-13:
> You may upgrade your cmake to v3.11 or later. (see: https://cmake.org/cmake/help/v3.11/release/3.11.html#commands)

We have a series of PRs (in the `export_targets` branches) that we are working on to make things work on CMake 3.10.  But they aren't quite ready yet.

fujitatomoya, 2020-05-14:
@xGonZh10n @clalancette 

appreciate for the info. installing specific version of cmake is not ordinary procedure to build ros2 from source, right? no pressure, just a question. so i guess i will wait for the cmake 3.10 support.

clalancette, 2020-05-14:
> appreciate for the info. installing specific version of cmake is not ordinary procedure to build ros2 from source, right?

It shouldn't generally be necessary.  Foxy is only officially supported on Ubuntu Focal, which means CMake 3.16.3, but we are doing our best to make sure it can still be built from source on Bionic (CMake 3.10).

fujitatomoya, 2020-05-15:
@clalancette 

nice catch, i will try Focal and close this issue.

xGonZh10n, 2020-05-15:
@clalancette @fujitatomoya 
Thanks for your replies. this is just a little hint.
Everyone’s choice is different. If you are a maintainer, you should modify the code to solve this problem, but if you are a user, in order not to affect your work progress, you can try more methods before the problem is solved.

fujitatomoya, 2020-05-15:
Using Ubuntu20.04, still meeting with following error.

```
--- stderr: orocos_kdl
In file included from /root/ros2_ws/colcon_ws/src/ros2/orocos_kinematics_dynamics/orocos_kdl/src/articulatedbodyinertia.cpp:22:
/root/ros2_ws/colcon_ws/src/ros2/orocos_kinematics_dynamics/orocos_kdl/src/articulatedbodyinertia.hpp:30:10: fatal error: Eigen/Core: No such file or directory
   30 | #include <Eigen/Core>
      |          ^~~~~~~~~~~~
compilation terminated.
```

described https://index.ros.org/doc/ros2/Troubleshooting/#cmake-packages-unable-to-find-asio-tinyxml2-tinyxml-eigen-or-log4cxx, (libeigen3-dev is installed)

I just followed the procedure https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/, the end up meeting this error.

if this is already know issue, and duplicated issue. i will close this.

clalancette, 2020-05-15:
> I just followed the procedure https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/, the end up meeting this error.
> 
> if this is already know issue, and duplicated issue. i will close this.

Hm, no, we aren't seeing that here.  CI is passing: https://ci.ros2.org/view/nightly/job/nightly_linux_debug/1554/ , and it builds fine locally for me as well.

This almost seems like you are missing the `eigen3_cmake_module` package somehow.  Can you make sure that `eigen3_cmake_module` package is in your workspace, and it is on tag 0.1.1?

fujitatomoya, 2020-05-15:
@clalancette 

> Can you make sure that eigen3_cmake_module package is in your workspace, and it is on tag 0.1.1?

yes, under colcol_workspace/src/ros2/eigen3_cmake_module

```
commit 526c6c745da360fe78fb727aa8a1f6daa8199abd (HEAD -> master, tag: 0.1.1, origin/master, origin/eloquent, origin/HEAD)
Author: Shane Loretz <sloretz@osrfoundation.org>
Date:   Thu Aug 8 16:45:40 2019 -0700

    Changelog + 0.1.1

    Signed-off-by: Shane Loretz <sloretz@osrfoundation.org>
```

BTW, I am using docker ubuntu:20.04 container.

fujitatomoya, 2020-05-16:
@clalancette @xGonZh10n @MiguelCompany 

i just started from the top with clean container image, and i got everything built successfully. I am not sure what went wrong, but it must be my mistake. sorry for making noise, and appreciate for your support. closing this.

closed 2020-05-16 by fujitatomoya