ros2/ros2 #919: [Meta-ticket]: Remove use of pkg_resources in python code

clalancette, 2020-05-12:
This meta-ticket tracks the removal of pkg_resources through the ROS 2 core.  There are a lot of individual pieces that need to be done to accomplish this, so this is a central place to track all of that.

# Motivation
pkg_resources is quite a heavy dependency.  Running on various machines, just importing `pkg_resources` takes something like 200 milliseconds:

```
$ time python3 -c 'import pkg_resources'

real	0m0.207s
user	0m0.162s
sys	0m0.022s
```

This hurts in two separate places:
* When running the command-line tools.  A delay of 200+ ms just to start up makes them feel "sluggish".
* In CI, especially during the `ros2cli` tests

Additionally, there are other modules that replicate the behavior of pkg_resources with much less overhead:

* https://importlib-metadata.readthedocs.io/en/latest
* https://importlib-resources.readthedocs.io/en/latest/
* https://github.com/pipxproject/pipx/pull/344#issuecomment-579927360
* https://docs.python.org/3/library/importlib.html#module-importlib.resources

# Strategy
The idea here is to replace the use of pkg_resources with `importlib` wherever possible.  Repeating the same test above with importlib:

```
$ time python3 -c 'import importlib.metadata'

real	0m0.054s
user	0m0.049s
sys	0m0.005s

$ time python3 -c 'import importlib.resources'

real	0m0.051s
user	0m0.039s
sys	0m0.011s
```

Even if we have to import both of them (which we typically don't), we'll still save ~100ms per invocation (on my computer, obviously; YMMV).

There are 2 major things we need to do in order to completely remove pkg_resources from the codebase:

1.  Remove our direct use of the pkg_resources APIs.  The good news is that we can incrementally remove the use of pkg_resources in favor of either `importlib.metadata` or `importlib.resources` without disrupting everything.  Here's the basic list of packages that would need this treatment:

* [ ] https://github.com/ros2/ros2cli
* [ ] https://github.com/ros2/launch
* [ ] https://github.com/ament/ament_lint
* [ ] https://github.com/ament/ament_package
* [ ] https://github.com/ros2/sros2
* [ ] https://github.com/osrf/osrf_pycommon

I have a prototype that does this part in the `remove-pkg-resources` branch of the above repositories.

2.  Remove the use of pkg_resources from the generated "binaries" from setuptools.  By default when you use setuptools to generate a `console_script` (like the `ros2` binary), it hard-codes the use of `pkg_resources`.  It is possible to override this with something like: https://github.com/ninjaaron/fast-entry_points .  So the second part here would be figuring out how to do that nicely.  Here are the list of repos that would need this treatment:

* [ ] https://github.com/ros2/geometry2
* [ ] https://github.com/ros2/examples
* [ ] https://github.com/ros2/demos
* [ ] https://github.com/ament/ament_index
* [ ] https://github.com/ament/ament_lint
* [ ] https://github.com/ros2/launch
* [ ] https://github.com/ros2/ros2cli

I have a prototype that does this part in the `add-fast-entry-points` branch of the above repositories.

sloretz, 2020-05-12:
> Remove the use of pkg_resources from the generated "binaries" from setuptools. By default when you use setuptools to generate a console_script (like the ros2 binary), it hard-codes the use of [...]

A while back I got curious about why the CLI tools take so long to start up and looked at this part. One of the ways to get rid of the `pkg_resource` import is to build using `wheel`s instead of `egg`s. That makes the installed script contain a path to the module to load rather than searching for it using `pkg_resources`. https://github.com/sloretz/colcon-core/commit/f9ecd4ae0f702eabab14b178201a3adc56b13472 . IIRC it works, but it creates some folders in the source space. I didn't bother to investigate more.

crystaldust, 2020-05-23:
I've searched all the git repos under ros2.repos file and found 34 lines of code that use `pkg_resources`:
```python
./ros2cli/ros2cli/ros2cli/entry_points.py:19:from pkg_resources import iter_entry_points
./ros2cli/ros2cli/ros2cli/entry_points.py:20:from pkg_resources import WorkingSet
./ros2cli/ros2cli/ros2cli/plugin_system.py:19:from pkg_resources import parse_version
./ros2cli/ros2pkg/ros2pkg/api/create.py:20:import pkg_resources
./ros2cli/ros2pkg/ros2pkg/api/create.py:67:    template_path = pkg_resources.resource_filename('ros2pkg', 'resource/' + template_file_name)
./ros2cli/ros2doctor/ros2doctor/api/__init__.py:19:from pkg_resources import iter_entry_points
./ros2cli/ros2doctor/ros2doctor/api/__init__.py:20:from pkg_resources import UnknownExtra
./rqt_plot/src/rqt_plot/data_plot/pyqtgraph_data_plot.py:39:        from pkg_resources import parse_version
./rqt_plot/src/rqt_plot/data_plot/pyqtgraph_data_plot.py:52:        import pkg_resources
./rqt_plot/src/rqt_plot/data_plot/pyqtgraph_data_plot.py:53:        pyqtgraph_version = pkg_resources.get_distribution("pyqtgraph").version
./rqt_plot/src/rqt_plot/data_plot/mat_data_plot.py:36:    from pkg_resources import parse_version
./ament_package/ament_package/__init__.py:19:    import pkg_resources
./ament_package/ament_package/__init__.py:21:        __version__ = pkg_resources.require('ament_package')[0].version
./ament_package/ament_package/__init__.py:22:    except pkg_resources.DistributionNotFound:
./ament_package/ament_package/__init__.py:25:        del pkg_resources
./ament_package/ament_package/templates.py:18:import pkg_resources
./ament_package/ament_package/templates.py:24:    return pkg_resources.resource_filename('ament_package', 'template/environment_hook/' + name)
./ament_package/ament_package/templates.py:40:    return pkg_resources.resource_filename('ament_package', 'template/package_level/' + name)
./ament_package/ament_package/templates.py:59:    return pkg_resources.resource_filename('ament_package', 'template/prefix_level/' + name)
./ament_package/ament_package/templates.py:78:    return pkg_resources.resource_filename(
./launch/launch/launch/frontend/parser.py:27:from pkg_resources import iter_entry_points
./ament_lint/ament_copyright/ament_copyright/__init__.py:15:import pkg_resources
./ament_lint/ament_copyright/ament_copyright/__init__.py:38:    for entry_point in pkg_resources.iter_entry_points(
./ament_lint/ament_copyright/ament_copyright/__init__.py:49:    for entry_point in pkg_resources.iter_entry_points(
./osrf_pycommon/osrf_pycommon/cli_utils/verb_pattern.py:20:import pkg_resources
./osrf_pycommon/osrf_pycommon/cli_utils/verb_pattern.py:152:    for entry_point in pkg_resources.iter_entry_points(group=group):
./osrf_pycommon/osrf_pycommon/cli_utils/verb_pattern.py:165:    for entry_point in pkg_resources.iter_entry_points(group=group):
./sros2/sros2/sros2/policy/__init__.py:19:import pkg_resources
./sros2/sros2/sros2/policy/__init__.py:25:    return pkg_resources.resource_filename(
./sros2/sros2/sros2/policy/__init__.py:31:    return pkg_resources.resource_filename(
./sros2/sros2/sros2/policy/__init__.py:37:    return pkg_resources.resource_filename(
./sros2/sros2/sros2/policy/__init__.py:43:    return pkg_resources.resource_filename(
./sros2/sros2/sros2/policy/__init__.py:49:    return pkg_resources.resource_filename(
./sros2/sros2/sros2/policy/__init__.py:55:    return pkg_resources.resource_filename(
clalancette
```

@clalancette  Let's keep a record here, how is everything being done? If possible I would like to fix some of the codes

clalancette, 2020-05-26:
I have proof-of-concept branches of all of the above in my forked repositories.  It all works, but note that just removing it from our code is not enough; we also have to figure out a solution for the console_entry_points, since by default those use pkg_resources as well.  One solution is what @sloretz mentioned above, which is to use wheels.  Another solution is to use https://github.com/ninjaaron/fast-entry_points .  But we don't have to solve this all at once; we can incrementally remove the use of pkg_resources, and then figure out the console_entry_points later.

I haven't pursued this further right now because of lack of time, and the release of Foxy.  Once Foxy is out the door, we can continue on with this problem.

Just for reference, here are my linked repositories.  In all repositories, look for the `remove-pkg-resources` and/or `add-fast-entry-points` branch:

* https://github.com/clalancette/ros2cli
* https://github.com/clalancette/launch
* https://github.com/clalancette/ament_lint
* https://github.com/clalancette/ament_package
* https://github.com/clalancette/sros2
* https://github.com/clalancette/osrf_pycommon

crystaldust, 2020-05-27:
I did exactly the same thing to remove the pkg_resources in the code these days and made some tests, it saves about 25%~30% time to launch `ros2`, this is how I do this:
```bash
## The original version with pkg_resources
$ python3 -c 'import ros2cli     
              print(ros2cli.__path__)'
['/opt/ros/eloquent/lib/python3.6/site-packages/ros2cli']
$ which ros2           
/opt/ros/eloquent/bin/ros2
$ time ros2 --help          
# ... help info
0.80user 0.10system 0:00.91elapsed 100%CPU (0avgtext+0avgdata 61248maxresident)k
0inputs+0outputs (0major+23165minor)pagefaults 0swaps

## Use the improved version
$ source install/setup.bash      
$ python3 -c 'import ros2cli       
              print(ros2cli.__path__)'
['/ros/ros2cli/install/ros2cli/lib/python3.6/site-packages/ros2cli']
$ which ros2
/ros/ros2cli/install/ros2cli/bin/ros2
$ time ros2 --help
# ... help info
0.59user 0.04system 0:00.63elapsed 99%CPU (0avgtext+0avgdata 36112maxresident)k
0inputs+0outputs (0major+6204minor)pagefaults 0swaps
```
That's an exciting result! I'm considering removing all the pkg_resources in my code as much as possible, cheers~

crystaldust, 2020-05-30:
I made a research on this and did exactly the same thing to remove the pkg_resources in the code these days until I see @clalancette's `add-fast-entry-points` branch. Made some tests on [ros2cli repo](https://github.com/ros2/ros2cli) based on [clalancetee's code](https://github.com/clalancette/ros2cli/tree/add-fast-entry-points) with a few lines of scripts [here](https://github.com/clalancette/ros2cli/tree/add-fast-entry-points). The launch time decreases with the code removing pkg_resources being added, finally it saves almost 75% time(from 0.39s to 0.11s) to launch ros2 cmd line tool:

```bash
# test with command "time ros2 --help"

# the base point
75e0ea1 (HEAD) [ros2param] Convert test_verb_dump into launch test (#485)
real	0m0.395s
user	0m0.348s
sys	0m0.047s

e57be12 (HEAD, chris/master) Make sure to add ros2doctor verbs to the extension points. (#495)
real	0m0.394s
user	0m0.385s
sys	0m0.008s

cff3c03 (HEAD) Remove use of pkg_resources from ros2cli.
real	0m0.214s
user	0m0.182s
sys	0m0.031s

343e837 (HEAD) Switch ros2pkg to using importlib.
real	0m0.213s
user	0m0.189s
sys	0m0.024s

658bcb4 (HEAD, chris/add-fast-entry-points) Add in fastentrypoints.
real	0m0.111s
user	0m0.105s
sys	0m0.006s

```
That's an exciting result! Hope this can be merged ASAP, cheers~

clalancette, 2020-06-01:
@crystaldust Great, thanks for doing the legwork on a performance test.  That makes it easier to justify getting this in.  Now the hard part is actually doing that work to get it landed :).

clalancette, 2020-06-22:
Here's preliminary CI on all of the changes:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2020-06-25:
> 2.  Remove the use of pkg_resources from the generated "binaries" from setuptools.  By default when you use setuptools to generate a `console_script` (like the `ros2` binary), it hard-codes the use [...]

As of `setuptools` 47.3.0 / 47.2.0 (https://github.com/pypa/setuptools/blob/master/CHANGES.rst#v4730) the generated entry points use `importlib_metadata` when available.

clalancette, 2020-06-26:
OK, so with most of the hand-coded uses of pkg_resources now removed from the code, we can turn to look at the entry points.

So far, I have seen 3 different solutions:

1.  Use https://github.com/ninjaaron/fast-entry_points .

Pros: Pretty simple to understand.  Doesn't require any additional dependencies outside itself.  Doesn't require any changes to infrastructure (buildfarm, CI jobs, etc).
Cons: Would need to make a package that all in-tree pure-Python packages would depend on so setup.py could import.  Monkey-patches setup.py (so will probably break at some point).

2.  Build wheels instead of eggs via colcon, ala https://github.com/sloretz/colcon-core/commit/f9ecd4ae0f702eabab14b178201a3adc56b13472

Pros: Works with existing versions of Python packages in Ubuntu Focal.  Eggs are "deprecated", and have been for a long time, so this would get rid of that.
Cons: Would also need some change on the buildfarm for building debians, as those don't use colcon.  Wheel building always leaves behind some files in the source repository.

3.  Use setuptools 47.3.0 or later.

Pros: Will work in H-Turtle (Ubuntu 22.04) out-of-the-box.  If developers forget to upgrade it on e.g. Ubuntu 20.04, everything will still work (just a bit slower).  No changes to colcon required.
Cons: Ubuntu 20.04 ships with setuptools 45.2.0, so we would have to pip install to override it; we'd have to do this both in the source setup instructions and on the buildfarm to generate debians.

I discussed this a bit offline with @shane yesterday.  We agreed that the first solution isn't really desirable.  I've tested both solutions 2 and 3 locally, and they both seem to work (in that they no longer use pkg_resources and start up faster).  I'm slightly leaning towards solution 3, as it has the least amount of downside, but I'm interested to hear feedback from others.

dirk-thomas, 2020-06-26:
> We agreed that the first solution isn't really desirable.

:+1: Too much effort for a temporary workaround which we would want to revert afterwards.

Just to clarify without doing anything:
* it already works (in terms of "being faster") on macOS and Windows (assuming the instructions are followed which install the latest version of `setuptools` via `pip` and
* it already works on Linux for from-source builds (either by users or for the fat archive we built) - again assuming the instructions are being followed.

The only scenario where it uses an older version of `setuptools` are the Debian packages since they use the Ubuntu package `python3-setuptools`. That being said option 2. would change anything for that use case. The only option to make that use case "faster" is by patching the pipeline building the Debian packages to install a newer version of `setuptools` via `pip`.

clalancette, 2020-06-26:
> Just to clarify - without doing anything:
> 
>   it already works (in terms of "being faster") on macOS and Windows (assuming the instructions are followed which install the latest version of `setuptools` via `pip`) and

Yes, that's true.

> 
>  it already works on Linux for from-source builds (either by users or for the fat archive we built) - again assuming the instructions are being followed.

You are right for the packaging builds; we already install setuptools via pip there.

But as far as I know, it isn't the case for source builds.  We don't have the users install `setuptools` via pip in any of the Linux from-source instructions ([Rolling](https://index.ros.org/doc/ros2/Installation/Rolling/Linux-Development-Setup/), for instance), so they will still be using the system version.

But you are right; the speedup is essentially restricted to Linux builds, from-source and on build.ros2.org.  So with that in mind, we can easily get the speedup by updating the from-source instructions on index.ros.org to recommend setuptools from pip, and also making a change to the buildfarm scripts to use an updated setuptools.

dirk-thomas, 2020-06-26:
> But as far as I know, it isn't the case for source builds.  We don't have the users install `setuptools` via pip in any of the Linux from-source instructions [...]

You are right. We removed the `pip` installation in #590. Maybe we should reintroduce it for `setuptools` - for now for `rolling` and if we backport the improvement then also for `foxy`.

clalancette, 2020-06-29:
I've opened up https://github.com/ros2/ros2_documentation/pull/799, which should take care of the from-source case.

For http://build.ros2.org , I had an offline conversation with @nuclearsandwich about it this morning.  The basic conclusion of the conversation is that it is probably not a good idea to inject newer setuptools dependencies on the buildfarm, as they can keep changing.  Even if we pin it, it's hard to guarantee that using setuptools 47.3.1 in our packages won't cause problems with the rest of the packages in the system, leading to hard-to-debug problems on the buildfarm.

We discussed 3 other options for the buildfarm:
1.  Use wheels.
1.  Package setuptools 47.3.0 and host it ourselves.
1.  Wait until Ubuntu 22.04
1.  Add a debbuild post script that rewrites entry-points by hand for the packages that we want.

Option 1 doesn't actually solve anything in the buildfarm case, since wheels aren't allowed in Debian packages (see https://www.debian.org/doc/packaging-manuals/python-policy/module_packages.html).  Option 2 isn't desirable, because we don't know what side-effects it will have on other python packages in the system.  Option 3 isn't great, since we are still 2 years out from Ubuntu 22.04.

That leaves us with option 4.  This would work by patching a debbuild post-script into the release repositories for the packages we generate binaries for.  We would only need to patch a handful of repositories; here's the full list of Python executables that we currently generate:

* ament_flake8
* ament_xmllint
* ament_lint_cmake
* launch_test
* ament_copyright
* ros2
* _ros2_daemon
* ament_cppcheck
* ament_pep257
* ament_index
* ament_cpplint
* ament_uncrustify

The good news is that `importlib.metadata` is installed by default as part of `libpython3.8-minimal`, so we don't need to inject any additional dependencies.  It should be relatively straightforward to write the post script to do this; basically look to see if a package has generated any Python executables with the `EASY-INSTALL-ENTRY-SCRIPT` string, and then rewrite those binaries to use `importlib.metadata`.

So I'm thinking about going ahead with option 4 here.  I'd like to hear opinions one way or another about this direction before I start work on it.

dirk-thomas, 2020-06-29:
Before ruling out option 2 I would suggest to compare the artifacts generated by the upstream version of setuptools with a version built by setuptools `47.3.1`. If the only difference for the enumerated packages is the generated executables then I would think it is much less work than option 4.

clalancette, 2020-06-29:
Here's the full diff between 45.2.0 (~ what is in 20.04) and 47.2.0 (the earliest version that uses importlib_metadata): https://gist.github.com/clalancette/ae5b4eb810a44445198426680e697860 .

Some of it is definitely benign; we can ignore the changes to the tests, and the CI configuration stuff, etc.  There are also some internal renames that wouldn't matter.  But there are enough other changes in there around pkg_resources and setuptools core that I don't think I would feel like I have the expertise to say what they would affect.

I will say that I think we'll only have to patch 3 release repositories to do the debbuild post thing; ament_lint, launch, and ros2cli.  The only thing that will be somewhat difficult is making the post script work to begin with.

dirk-thomas, 2020-06-29:
> Here's the full diff

I was referring to the diff in generated artifacts - so the diff in your packages - cause by the difference in `setuptools`.

clalancette, 2020-06-29:
Oh, that diff is this:

```
--- /tmp/ros2	2020-06-29 16:31:07.952526148 +0000
+++ install/ros2cli/bin/ros2	2020-06-29 16:31:35.177428290 +0000
@@ -1,12 +1,33 @@
 #!/usr/bin/python3
 # EASY-INSTALL-ENTRY-SCRIPT: 'ros2cli==0.9.5','console_scripts','ros2'
-__requires__ = 'ros2cli==0.9.5'
 import re
 import sys
-from pkg_resources import load_entry_point
+
+# for compatibility with easy_install; see #2198
+__requires__ = 'ros2cli==0.9.5'
+
+try:
+    from importlib.metadata import distribution
+except ImportError:
+    try:
+        from importlib_metadata import distribution
+    except ImportError:
+        from pkg_resources import load_entry_point
+
+
+def importlib_load_entry_point(spec, group, name):
+    dist_name, _, _ = spec.partition('==')
+    matches = (
+        entry_point
+        for entry_point in distribution(dist_name).entry_points
+        if entry_point.group == group and entry_point.name == name
+    )
+    return next(matches).load()
+
+
+globals().setdefault('load_entry_point', importlib_load_entry_point)
+
 
 if __name__ == '__main__':
     sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
-    sys.exit(
-        load_entry_point('ros2cli==0.9.5', 'console_scripts', 'ros2')()
-    )
+    sys.exit(load_entry_point('ros2cli==0.9.5', 'console_scripts', 'ros2')())
```

And that's essentially what I was going to patch in using debbuild post scripts.

dirk-thomas, 2020-06-29:
If the delta is exact what we want (nothing more and nothing less) what would be the advantage of doing it manually as in option 4? Option 2 could easily pin a specific `setuptools` version (so the diff wouldn't change) and could be limited to the package we need it for.

clalancette, 2020-06-29:
> If the delta is exact what we want (nothing more and nothing less) what would be the advantage of doing it manually as in option 4? Option 2 could easily pin a specific `setuptools` version (so the diff wouldn't change) and could be limited to the package we need it for.

I don't see how we would do it with upgrading the package.  Or, at least, I don't see how to do it in a consistent way.  We would either need to package and import newer setuptools *only* on the buildfarm, or we would package and import newer setuptools and make it available to all downstream users.  The former would mean we would run on end-user systems with a different setuptools than we used for building/packaging.  The latter would mean we would effectively upgrade all end-user systems to have a newer setuptools, and all of the possible chaos that would provide.  Neither seems like a good option to me.

dirk-thomas, 2020-06-29:
Maybe I am missing something. I thought the newer `setuptools` needs to be only present at build time of a package since it will generate different code for the entry points. At runtime of the package the updates version would be sufficient.

If that is the case installing a new version via `pip` when creating the Debian package should be sufficient. That would neither require packaging `setuptools` nor affect a users system in any way.

clalancette, 2022-03-28:
This has now been done, thanks to @wjwwood and @nuclearsandwich .  Closing this out.

closed 2022-03-28 by clalancette