ros2/ros2 #922: Unreliable service performance

hidmic, 2020-05-20:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10, Ubuntu 20.04, OSX
- Installation type:
  - Binaries and form source
- Version or commit hash:
  - [Foxy](https://github.com/ros2/ros2/blob/4772a2a9b926126567008a0d83c5c7323d5c684b/ros2.repos)
- DDS implementation:
  - Fast-RTPS most often, but also with CycloneDDS
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue

Any service / client pair will do, e.g.:
 
```sh
ros2 run demo_nodes_cpp add_two_ints_server
```

and

```sh
ros2 run demo_nodes_cpp add_two_ints_client
```

#### Expected behavior
Client sends request, servers gets request then sends response, and client gets response.

#### Actual behavior

The client will hang some times and not get any result. How often that happens seems to vary on a platform and RMW implementation. 

#### Additional information

Issues with services performance has been observed before, as early as https://github.com/ros2/demos/issues/304. More recent occurrences may be found in https://github.com/ros2/rmw_cyclonedds/issues/185,  https://github.com/ros2/system_tests/issues/420 and https://github.com/ros2/rmw_cyclonedds/issues/74. Nondeterministic as it is, it has seemed resolved in the past, suggesting race conditions as root cause.

Parameter servers as well as action servers and clients relying on services for their implementation are also negatively affected by these issues.

hidmic, 2020-05-20:
CC @jacobperron

hidmic, 2020-05-20:
@MiguelCompany @richiprosima it'd be great if a fix for `rmw_fastrtps_cpp` landed before Foxy gets released.

MiguelCompany, 2020-05-24:
@hidmic I've just created ros2/rmw_fastrtps#390, which should fix the issue of sending a response when the reader has not yet been matched.

sloretz, 2020-06-18:
Closing in favor of https://github.com/ros2/rmw_fastrtps/issues/392 and https://github.com/ros2/rmw_cyclonedds/issues/191 since these appear to be duplicates and the other two are on more specific repos.

closed 2020-06-18 by sloretz