ros2/ros2 #924: [Meta-Ticket]: Track adding the REP-2006 language to all of the Tier{1/2} repositories

clalancette, 2020-05-22:
As specified in [REP-2004](https://www.ros.org/reps/rep-2004.html), all Quality Level 1 and 2 packages *must* have a Vulnerability Policy Disclosure policy.  This is currently in progress as REP-2006 at https://github.com/ros-infrastructure/rep/pull/262 .  Once that is finalized, we will merge it and it will be a "live" REP.

In the meantime, we will open up draft PRs that add a pointer to the eventual location of REP-2006 to all of the packages that are aiming for Quality Level 1 and 2.  This ticket is a meta-ticket to track those open PRs.  Note that there is no checklist here; just the fact of the PR being open and mentioning this issue is how we'll track them.  Once REP-2006 is in, and all of the linked tickets are in, we can close this tracking issue.

christophebedard, 2020-05-23:
https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/merge_requests/180

clalancette, 2020-05-28:
All right, I merged REP-2006, so upgrading all of the PRs from drafts to "real" ones.

dirk-thomas, 2020-05-28:
Nitpick: please update the title. There is no such term as "tier 1/2 packages. "Tier levels" are only defined for platforms.

clalancette, 2020-06-04:
All right, all of these have been merged now, so closing this ticket.

closed 2020-06-04 by clalancette