ros2/ros2 #929: Sourcing setup.bash on ROS2 takes considerably long time

fujitatomoya, 2020-06-02:
## Bug report

- Operating System:
  - Ubuntu 18.04 Bionic / Raspi4
- Installation type:
  - Binary
- Version or commit hash:
  - Eloquent (ros:eloquent docker images 2b65b6987efa)
- DDS implementation:
  - N.A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
# time source /opt/ros/eloquent/setup.bash

real    0m1.409s
user    0m0.745s
sys     0m0.697s
```

almost 1.5 sec every single time to source.

#### Expected behavior

1st time is necessary since colcon build _cached_ordered_commands, but 2nd time or later can be faster and optimized if environment is not modified. hopefully 100-200msec.

#### Actual behavior

almost 1.5 sec every single time to source.

fujitatomoya, 2020-06-02:
originally discussed here https://github.com/ros2/ros2/issues/573

fujitatomoya, 2020-06-03:
@richichiare @clalancette @hdino @EricCousineau-TRI 

This is just proposal and experimental, but we can make it faster with file based _cached_ordered_commands, result and environment are followings.

### Environment

Raspi4, ros:eloquent hash is 2b65b6987efa(arm64) via `docker pull ros:eloquent`

### Improvement Result

- Before

```
time source /opt/ros/eloquent/setup.bash

real    0m1.517s
user    0m0.772s
sys     0m0.808s
```

- After

```
time source /opt/ros/eloquent/setup.bash
Use cached environmnet !
If you want to update cached environment, please remove "/root/.ros/env.cache" !
And re-execute ". setup.bash" again.

real    0m0.221s
user    0m0.132s
sys     0m0.098s
```

### source scripts for this trial

https://github.com/fujitatomoya/ros2_test_prover/tree/master/script/ros2_issues_929

replace them with the following,

```
/opt/ros/eloquent/setup.bash
/opt/ros/eloquent/local_setup.sh
```

1st `source` will create cache file, but from 2nd time or later, it will use cached file instead making commands.

### Check Check Check

**Do not use `docker run` since it will source the ros2 environment, this will cause incomplete cached file. Please use `docker exec` instead.**

### Note

- `ros2 run demo_nodes_cpp talker/listener` can be running w/o error.
- at production phase, we do not expect runtime package and environment changes a lot. (even if so, we can re-create the cached file before release.)

@Barry-Xu-2018

can we introduce the colcon template script change?

Barry-Xu-2018, 2020-06-03:
Now ros2 environment support different shell (such as bash, powershell).  
We only modify script for bash as example to show how to modify.  

Modification :  
- At first time of execute `. setup.bash` to save ordered command to cache file.  
- From second time of execution, directly use cached environment (Avoid parsing dsv file)

If you install environment by Debian package, the script for configuring environment is from https://github.com/ament/ament_package.  

The change is https://github.com/Barry-Xu-2018/ament_package/commit/baf342c0a22fec60c64ab1fcc78b1cb8a194a404

if you build ros environment,  the script for configuring environment is from https://github.com/colcon/colcon-bash

The change is https://github.com/Barry-Xu-2018/colcon-bash/commit/01cc7142394707807315c379bd0c82a023c8c520

fujitatomoya, 2020-06-03:
@wjwwood @dirk-thomas 

friendly ping, and really appreciate if you could share your thoughts.

dirk-thomas, 2020-06-03:
I don't think this kind of hack (creating a temporary cache which the user has to manually invalidate) is a viable workaround.

fujitatomoya, 2020-06-04:
> creating a temporary cache which the user has to manually invalidate

at this point, i do agree. thanks

dirk-thomas, 2020-06-04:
Can each of you please share the *two* times you see for these different `source` invocations:
* `time . /opt/ros/<distro>/setup.sh`
* `time . /opt/ros/<distro>/local_setup.sh`

Please run each command in a separate "clean" terminal.

Barry-Xu-2018, 2020-06-04:
@dirk-thomas 

> Can each of you please share the two times you see for these different source invocations:

Sorry.  Currently I only modify code for bash.  And entry is setup.bash.  If run from that scripts, cache functionality doesn't work.  
Could you tell me the purpose of executing these commands ?

dirk-thomas, 2020-06-04:
I am referring of the time it takes to source without your modifications. And purpose is to see any potential difference in timing between them.

nuclearsandwich, 2020-06-04:
Using a 20.04 LXC container. With no ROS packages installed but ros-workspace:
```
root@ros2-focal:~# time bash -c '. /opt/ros/foxy/setup.bash'
real	0m0.079s
user	0m0.078s
sys	0m0.006s

root@ros2-focal:~# time bash -c '. /opt/ros/foxy/local_setup.bash'
real	0m0.071s
user	0m0.067s
sys	0m0.007s
```

With ros-foxy-desktop installed:
```
root@ros2-focal:~# time bash -c '. /opt/ros/foxy/setup.bash'
real	0m0.511s
user	0m0.397s
sys	0m0.143s

root@ros2-focal:~# time bash -c '. /opt/ros/foxy/local_setup.bash'

real	0m0.167s
user	0m0.135s
sys	0m0.037s
```

clalancette, 2020-06-04:
Similar setup here; 20.04 LXC container, ros-foxy-desktop installed:

```
ubuntu@ros2-focal:~$ time . /opt/ros/foxy/setup.sh 

real	0m0.791s
user	0m0.332s
sys	0m0.321s
```

A new terminal:

```
ubuntu@ros2-focal:~$ time . /opt/ros/foxy/local_setup.sh 

real	0m0.155s
user	0m0.133s
sys	0m0.025s
```

audrow, 2020-06-04:
With a 20.04 LXC container, ros-foxy-desktop installed:

Terminal 1:
```
root@dceb574c5a84:/# time . /opt/ros/foxy/setup.sh 

real	0m0.673s
user	0m0.451s
sys	0m0.257s
```
Terminal 2:
```
root@dceb574c5a84:/# time . /opt/ros/foxy/local_setup.sh 

real	0m0.102s
user	0m0.080s
sys	0m0.024s
```

dirk-thomas, 2020-06-04:
Thanks for the stats.

This confirms that the problem is in the `setup.sh` file and the `local_setup.sh` file is reasonably fast (for what it is doing).

I am fairly certain that the problem is not present when building from source in which case `colcon` generates those setup files.

I updated the ticket title accordingly.

I guess https://github.com/ament/ament_package/blob/82a6d0ef05818268f06eb83866a76da26fff1b2e/ament_package/template/prefix_level/setup.sh.in will need to make use of the optimizations added for Eloquent. I will take a closer look...

dirk-thomas, 2020-06-04:
Using the two referenced PRs the `setup.sh` file will be changed to the following (generated using the logic described in https://github.com/ament/ament_package/pull/115#issuecomment-639138324):

<details>

<summary>setup.py</summary>

```
# generated from ament_package/template/prefix_level/setup.sh.in

# since this file is sourced use either the provided AMENT_CURRENT_PREFIX
# or fall back to the destination set at configure time
: ${AMENT_CURRENT_PREFIX:=/opt/ros/foxy}

# set type of shell if not already set
: ${AMENT_SHELL:=sh}

# function to append non-duplicate values to environment variables
# using colons as separators and avoiding leading separators
ament_append_unique_value() {
  # arguments
  _listname=$1
  _value=$2
  #echo "listname $_listname"
  #eval echo "list value \$$_listname"
  #echo "value $_value"

  # check if the list contains the value
  eval _values=\$$_listname
  _duplicate=
  _ament_append_unique_value_IFS=$IFS
  IFS=":"
  if [ "$AMENT_SHELL" = "zsh" ]; then
    ament_zsh_to_array _values
  fi
  for _item in $_values; do
    # ignore empty strings
    if [ -z "$_item" ]; then
      continue
    fi
    if [ $_item = $_value ]; then
      _duplicate=1
    fi
  done
  unset _item

  # append only non-duplicates
  if [ -z "$_duplicate" ]; then
    # avoid leading separator
    if [ -z "$_values" ]; then
      eval $_listname=\"$_value\"
      #eval echo "set list \$$_listname"
    else
      # field separator must not be a colon
      unset IFS
      eval $_listname=\"\$$_listname:$_value\"
      #eval echo "append list \$$_listname"
    fi
  fi
  IFS=$_ament_append_unique_value_IFS
  unset _ament_append_unique_value_IFS
  unset _duplicate
  unset _values

  unset _value
  unset _listname
}

# iterate over all parent_prefix_path files
_prefix_setup_IFS=$IFS
IFS="
"
# this variable contains the concatenated prefix paths in reverse order
_UNIQUE_PREFIX_PATH=""

# this check is used to skip parent prefix path in the Debian package
if [ -z "SKIP_PARENT_PREFIX_PATH" ]; then
  # find parent prefix path files for all packages under the current prefix
  _RESOURCES="$(\find "$AMENT_CURRENT_PREFIX/share/ament_index/resource_index/parent_prefix_path" -mindepth 1 -maxdepth 1 2> /dev/null | \sort)"

  if [ "$AMENT_SHELL" = "zsh" ]; then
    ament_zsh_to_array _RESOURCES
  fi
  for _resource in $_RESOURCES; do
    # read the content of the parent_prefix_path file
    _PARENT_PREFIX_PATH="$(\cat "$_resource")"
    # reverse the list
    _REVERSED_PARENT_PREFIX_PATH=""
    IFS=":"
    if [ "$AMENT_SHELL" = "zsh" ]; then
      ament_zsh_to_array _PARENT_PREFIX_PATH
    fi
    for _path in $_PARENT_PREFIX_PATH; do
      # replace placeholder of current prefix
      if [ "$_path" = "{prefix}" ]; then
        _path="$AMENT_CURRENT_PREFIX"
      fi
      # avoid leading separator
      if [ -z "$_REVERSED_PARENT_PREFIX_PATH" ]; then
        _REVERSED_PARENT_PREFIX_PATH=$_path
      else
        _REVERSED_PARENT_PREFIX_PATH=$_path:$_REVERSED_PARENT_PREFIX_PATH
      fi
    done
    unset _PARENT_PREFIX_PATH
    # collect all unique parent prefix path
    if [ "$AMENT_SHELL" = "zsh" ]; then
      ament_zsh_to_array _REVERSED_PARENT_PREFIX_PATH
    fi
    for _path in $_REVERSED_PARENT_PREFIX_PATH; do
      ament_append_unique_value _UNIQUE_PREFIX_PATH "$_path"
    done
    unset _REVERSED_PARENT_PREFIX_PATH
  done
  unset _resource
  unset _RESOURCES
fi

# append this directory to the prefix path
ament_append_unique_value _UNIQUE_PREFIX_PATH "$AMENT_CURRENT_PREFIX"
unset AMENT_CURRENT_PREFIX

# store AMENT_SHELL to restore it after each prefix
_prefix_setup_AMENT_SHELL=$AMENT_SHELL
# source local_setup.EXT or local_setup.sh file for each prefix path
IFS=":"
if [ "$AMENT_SHELL" = "zsh" ]; then
  ament_zsh_to_array _UNIQUE_PREFIX_PATH
fi
for _path in $_UNIQUE_PREFIX_PATH; do
  # trace output
  if [ -n "$AMENT_TRACE_SETUP_FILES" ]; then
    echo ". \"$_path/local_setup.$AMENT_SHELL\""
  fi
  if [ -f "$_path/local_setup.$AMENT_SHELL" ]; then
    if [ "$AMENT_SHELL" = "sh" ]; then
      # provide AMENT_CURRENT_PREFIX to .sh files
      AMENT_CURRENT_PREFIX=$_path
    fi
    # restore IFS before sourcing other files
    IFS=$_prefix_setup_IFS
    . "$_path/local_setup.$AMENT_SHELL"
    # restore AMENT_SHELL after each prefix-level local_setup file
    AMENT_SHELL=$_prefix_setup_AMENT_SHELL
  fi
done
unset _path
IFS=$_prefix_setup_IFS
unset _prefix_setup_IFS
unset _prefix_setup_AMENT_SHELL
unset _UNIQUE_PREFIX_PATH
unset AMENT_SHELL
```

</details>

@nuclearsandwich @clalancette @audrow If either of you could run the `time` command again with this `setup.sh` file and share their result. Thanks.

Barry-Xu-2018, 2020-06-05:
@dirk-thomas

> I am referring of the time it takes to source without your modifications. And purpose is to see any potential difference in timing between them.

Thanks. Got it.

I also tested on my environment (docker ros:eloquent. Image ID: c75f02804aa2)

- Original script
    - setup.sh
         ```shell
         root@d8c6e6d725ea:/# time . /opt/ros/eloquent/setup.sh 

         real	0m0.291s
         user	0m0.251s
         sys	0m0.060s
        ```
    - local_setup.sh
        ```shell
        root@d8c6e6d725ea:/# time . /opt/ros/eloquent/local_setup.sh 

        real	0m0.097s
        user	0m0.073s
        sys	0m0.027s
        ```

- Update script
    - setup.sh (You provide it)
        ```shell
        root@d8c6e6d725ea:/# time . /opt/ros/eloquent/setup.sh

        real	0m0.101s
        user	0m0.096s
        sys	0m0.008s
        ```
The consumed time of executing `setup.sh` is almost the same as local_setup.sh.
So

closed 2020-06-05 by dirk-thomas

fujitatomoya, 2020-06-08:
@dirk-thomas and everyone,

sorry to be late to catch up. appreciate for the optimization!