ros2/ros2 #930: Network Error (subscription error) of  ROS2 Dashing  and  Eloquent  installed in Jetson AGX Xavier

ivis-Matsumoto-Shigeki, 2020-06-03:
Hi, I’m faced with a problem, 

About ROS2 Dashing Diademata and  Eloquent Elusor installed in Jetson AGX Xavier (Ubuntu18.04, ARM), Timestamps of large data(sensor_msgs/Image,  480 x 640, Mono8, 10Hz), that  subscribed  at  Xavier, is mixed up.

A publisher  node is in another machine (Windows 10, ROS2 Bouncy Bolson).

For Bouncy Bolson and  for Crystal Clemmys, there are no problems above.
For  Dashing Diademata installed in normal PC(not ARM) , there are no problems, too.

 thanks,

fujitatomoya, 2020-06-04:
can you add more information, i do not understand what the problem is.

> Timestamps ... is mixed up.

How mixed up? can you share your log including timestamp on both publisher and subscriber.

sloretz, 2020-06-11:
Hi @ivis-Matsumoto-Shigeki ,

I'm nut sure I understand. Would you mind posting a minimal reproducible example? Is it *only* on the Jetson board, or does it happen on your desktop too?

sloretz, 2020-06-18:
@ivis-Matsumoto-Shigeki friendly ping :)