ros2/ros2 #931: Fast-RPTS 2.0.x might introduce service performance issue

daisukes, 2020-06-03:
## Bug report

**Required Info:**
This is related to these issues
https://github.com/ros-planning/navigation2/issues/1788
https://github.com/ros2/rclcpp/issues/1152
https://github.com/ros2/ros2/issues/922

- Operating System:
  - Ubuntu20.04
- Installation type:
  - Source build
- Version or commit hash:
  - Foxy master
- DDS implementation:
  - Fast-RTPS 2.0.x -> v1.10.0
- Client library (if applicable):
  - Navigation2

#### Steps to reproduce issue
```
# Launch tb3 simulation and provide an initial location to start lifecycle
$ ros2 launch nav2_bringup tb3_simulation_launch.py

# Then call service like
$ ros2 service call /global_costmap/clear_entirely_global_costmap nav2_msgs/srv/ClearEntireCostmap "{}"
```

Service call often failed with 2.0.x, but works well with v1.10.0

#### Additional information
I posted this issue also here, a suggestion by @SteveMacenski
I only tested with a specific service of navigation2 costmap that I want to use.
I'm not sure the disadvantage of downgrading the default Fast-RTPS version, but at least I experienced flaky service behavior with 2.0.x and it is improved with v1.10.0

SteveMacenski, 2020-06-03:
and to be explicit:

> This problem does not appear if I use Fast-RTPS v1.10.0 instead of 2.0.x.

So it looks like something may have made the default settings cause services to essentially stop working on all but the most powerful CPUs from his analysis in the navigation2 ticket.

dirk-thomas, 2020-06-03:
What commit hash of Fast-RTPS are you using?

ivanpauno, 2020-06-03:
As discussed in https://github.com/ros2/rmw_fastrtps/pull/390 and described in https://github.com/ros2/rmw_fastrtps/issues/392, there's a long term race condition in service discovery. Any unrelated change in fastrtps might have made the race more/less likely to happen, but it's probably not an issue in `fastrtps` itself.

This is also an issue in cyclonedds, as discussed in https://github.com/ros2/rmw_cyclonedds/pull/187.

It doesn't seem to be an issue in Connext, as we're using an implementation of RPC over DDS provided by them (which I assume it solves the race condition discovery problem).

daisukes, 2020-06-03:
> What commit hash of Fast-RTPS are you using?

v1.10.0 release
https://github.com/eProsima/Fast-DDS/commit/a87af8e6772aa06790bc334996f035adb1fde00c

dirk-thomas, 2020-06-03:
> v1.10.0 release

I was referring to the 2.0.x commit which has a problem for you.

daisukes, 2020-06-03:
The docker image was overwritten, so I can only guess, but I got this problem last Friday and all images have been build from the source this week so, it should be the latest commit of 2.0.x
https://github.com/eProsima/Fast-DDS/commit/b710b1f53a4ecf6c92f87661347a93c46e5f4854

dkuenster, 2020-06-05:
https://github.com/ros-planning/navigation2/issues/1772 can also be fixed by either switching to Cyclone DDS or going back to v1.10.0 of Fast-RTPS.

daisukes, 2020-06-05:
I tested some commits of Fast-RTPS.

It works until the commit https://github.com/eProsima/Fast-DDS/commit/1157990b9bc04daf822f3d6d5a5349e86a2cd9a4
Diable SHM reduces chance about 30% https://github.com/eProsima/Fast-DDS/commit/872df3e32df5e2a1eca8f2a0e7b0992f79dc6ef5
Same symptoms also with https://github.com/eProsima/Fast-DDS/commit/d5c9d6bcd4fdfe7edadb137c6203a2db8d01154f
The next commit, changes of StatefulWriter reproduce the problem https://github.com/eProsima/Fast-DDS/commit/a9bd1a9003adb7ca80c0f6854de58e181059de94

sloretz, 2020-06-18:
Closing in favor of https://github.com/ros2/rmw_fastrtps/issues/392 since the tickets are duplicates and `rmw_fastrtps` is a more specific repo.

closed 2020-06-18 by sloretz