ros2/ros2 #942: cppcheck 2.0 nightly failures on Windows

nuclearsandwich, 2020-06-09:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - source build
- Version or commit hash:
  - N/A
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Install cppcheck 2.0 via Chocolatey and build almost any ROS 2 package that uses cppcheck. https://ci.ros2.org/view/nightly/job/nightly_win_rel/1576/testReport/ has many examples.

#### Expected behavior

cppcheck tests pass

#### Actual behavior

Thousands of lines of output in failed cppcheck tests similar to these:

```
39: [src\rclpy\_rclpy.c:77]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If RCUTILS_STRINGIFY is a macro then please configure it.
39: [src\rclpy\_rclpy.c:77]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If RCUTILS_STRINGIFY is a macro then please configure it.
39: [src\rclpy\_rclpy.c:77]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If RCUTILS_STRINGIFY is a macro then please configure it.
4: [test\test_component_manager.cpp:68]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If ; is a macro then please configure it.
4: [test\test_component_manager.cpp:68]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If ; is a macro then please configure it.
4: [test\test_component_manager.cpp:68]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If ; is a macro then please configure it.
1: [src\rviz_default_plugins\robot\robot.cpp:252]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If ; is a macro then please configure it.
```

#### Additional information

cppcheck 2.0 was [published via chocolatey](https://chocolatey.org/packages/cppcheck#versionhistory) on 2020-06-08 after being [released](https://github.com/danmar/cppcheck/releases/tag/2.0) on 2020-05-20. cppcheck 2.0 is also available on macOS via homebrew but we have not yet updated the macOS hosts.

claireyywang, 2020-06-10:
I was able to reproduce the cppcheck error on osx 
```
[src/rclpy/_rclpy.c:78]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If RCUTILS_STRINGIFY is a macro then please configure it.
1 errors
```
and to confirm `RCUTILS_STRINGIFY`is the one triggering the check error, but I can't figure out why. This macro is used across ros2 codebase and doesn't seem to need `include` directory to use. This is as much as I got so far.

jacobperron, 2021-01-22:
FWIW, with https://github.com/ament/ament_lint/pull/268 merged, we shouldn't see "unknown macro" errors anymore. Maybe it's possible to unpin cppcheck now?