ros2/ros2 #944: Foxy build failing on Debian 10 (Buster) due to linking failure to pthread

rasmusan, 2020-06-11:
## Bug report

**Required Info:**

- Operating System:
  - Debian 10.4 (Buster)
- Installation type:
  - source, following these steps: https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/
- Version or commit hash:
  - [foxy-release](https://raw.githubusercontent.com/ros2/ros2/foxy/ros2.repos)
- DDS implementation:
  - Default (Fast-RTPS)
- Client library (if applicable):
  - tf2_ros, dummy_map_server (probably also others)

#### Steps to reproduce issue

On a clean Debian 10.4, the steps from ros.org: https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/.

#### Expected behavior

A succesfull build of ROS2 Foxy

#### Actual behavior

1. Everything until _colcon build_ succeeds
2. _colcon build --symlink-install --merge-install_ succeeds in building 171/302 packages
3. The failed package is tf2_ros. It fails with an apparent linking issue:

> --- stderr: tf2_ros                                                                                                                     
/usr/bin/ld: CMakeFiles/tf2_monitor.dir/src/tf2_monitor.cpp.o: undefined reference to symbol 'pthread_rwlock_unlock@@GLIBC_2.2.5'
/usr/bin/ld: //lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/tf2_monitor.dir/build.make:157: tf2_monitor] Error 1
make[1]: *** [CMakeFiles/Makefile2:403: CMakeFiles/tf2_monitor.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

#### Additional information

tf2_ros can be build with the following additions to it's CMakeLists:
`find_package(Threads REQUIRED)`
`...`
`target_link_libraries(${PROJECT_NAME} Threads::Threads)`

The build failure is not unique to tf2_ros. The package ros2/demos/dummy_robot/dummy_map_server also fails, with the same error message:
> --- stderr: dummy_map_server                                   
/usr/bin/ld: CMakeFiles/dummy_map_server.dir/src/dummy_map_server.cpp.o: undefined reference to symbol 'pthread_rwlock_unlock@@GLIBC_2.2.5'
/usr/bin/ld: //lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dummy_map_server.dir/build.make:141: dummy_map_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/dummy_map_server.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

The build succeeds with ROS2 Eloquent. I have searched on related issues and found this from 2015: https://github.com/ros2/demos/issues/11. I have not been able to identify if the cause is related. I have not installed RTI Connext at the moment (as it requires a license).

emersonknapp, 2020-06-15:
Interesting - we're also seeing this when building with Clang+Libcxx, see https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/451/console

This was also passing in Eloquent. Eloquent uses Debian Jessie, right?

I'm guessing that the solution will be the same for both

clalancette, 2020-06-16:
> [](https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/451/console)
> 
> This was also passing in Eloquent. Eloquent uses Debian Jessie, right?

FYI, here is the table for Eloquent: https://www.ros.org/reps/rep-2000.html#eloquent-elusor-november-2019-november-2020 .  It looks like it targeted Debian Buster as Tier 3.

emersonknapp, 2020-06-16:
Ok, should have double checked before commenting :) edited the original so there's no misinformation. Regardless - it seems likely that the solution to this is related to the solution to the clang+libcxx issue.

rasmusan, 2020-06-16:
An addition; I did manage to build foxy by adding linking to Threads in:
- tf2_ros
- realtime_support/tlsf_cpp
- rviz2/test/tools

... and ignoring the packages:
- micro-ROS
- ros/ros_tutorials
- ros2/demos
- ros2/examples
- ros2/system_tests

clalancette, 2020-06-16:
> An addition; I did manage to build foxy by adding linking to Threads in:

We would definitely be willing to review PRs to those projects if you'd like to contribute.

rasmusan, 2020-06-17:
> We would definitely be willing to review PRs to those projects if you'd like to contribute.

@clalancette I did not consider making a PR with this solution, as it seems to me to be a workaround more than a solution to the root cause. None of the packages for which I added linking to Threads had this (explicit) linking in Eloquent. However; I can make a PR, if you think this solution can be acceptable (at least temporarily)?

clalancette, 2020-06-17:
> @clalancette I did not consider making a PR with this solution, as it seems to me to be a workaround more than a solution to the root cause. None of the packages for which I added linking to Threads [...]

Yes, I'm not quite sure why it is needed now either.  But there has been a lot of change since Eloquent, so maybe we have new dependencies.  It would be worthwhile doing a little more investigation to find out why.

That being said, adding in the threads dependency is a thing we have done for some other packages for cross-compiling, so I think it would be OK for some more.

dirk-thomas, 2020-06-17:
Addressing the symptom of missing pthread in a few packages doesn't sound like the right approach to me. Instead the actual source of the problem (a package which needs it but doesn't export that information) should be identified and fixed. If the symptoms have been patches as proposed that will make it even more difficult imo to find the actual problem.

dirk-thomas, 2020-07-01:
@rasmusan Please try to apply the fix from ros2/rclcpp#1210 and comment if that resolves the problem for you.

rasmusan, 2020-07-02:
> @rasmusan Please try to apply the fix from [ros2/rclcpp#1210](https://github.com/ros2/rclcpp/pull/1210) and comment if that resolves the problem for you.

Yes, this fixed the problem. Thanks!
I verified that the solution works by undoing all of my modifications, pulling the latest commit of rclcpp's foxy branch (3a3ba55f), and cherrypicking the commit with the fix (from the master branch). After this, everything compiles as expected.

closed 2020-07-06 by nuclearsandwich