ros2/ros2 #946: Large messages are being dropped sometimes

marcbone, 2020-06-15:
## Bug report

**Required Info:**

- Operating System:
- Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  -  Fast-RTPS, RTI Connext, CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

I am trying to measure latencies in ROS2 local communication. However, if I am trying to send a large message (10 MB in this case) some messages are not arriving, or are taking very long to arrive. To illustrate this issue I made a minimal example based on the ros2 tutorial:

```c++
#include <string>
#include "rclcpp/rclcpp.hpp"
#include <functional>
#include "std_msgs/msg/string.hpp"
using namespace std::chrono_literals;

class MinimalPublisher : public rclcpp::Node
{
public:
    MinimalPublisher()
            : Node("minimal_publisher")
    {
        data=random_string(10000000); // generate a ~10MB string
        publisher_ = this->create_publisher<std_msgs::msg::String>("topic", 1);
        timer_ = this->create_wall_timer(
                100ms, std::bind(&MinimalPublisher::timer_callback, this));
    }

private:
    void timer_callback()
    {
        auto message = std_msgs::msg::String();
        message.data = data;
        publisher_->publish(message);
    }
    std::string random_string( size_t length )
    {
        auto randchar = []() -> char
        {
            const char charset[] =
                    "0123456789"
                    "ABCDEFGHIJKLMNOPQRSTUVWXYZ"
                    "abcdefghijklmnopqrstuvwxyz";
            const size_t max_index = (sizeof(charset) - 1);
            return charset[ rand() % max_index ];
        };
        std::string str(length,0);
        std::generate_n( str.begin(), length, randchar );
        return str;
    }
    rclcpp::TimerBase::SharedPtr timer_;
    rclcpp::Publisher<std_msgs::msg::String>::SharedPtr publisher_;
    std::string data;
};

int main(int argc, char * argv[])
{
    rclcpp::init(argc, argv);
    rclcpp::spin(std::make_shared<MinimalPublisher>());
    rclcpp::shutdown();
    return 0;
}
```

```c++
#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"
using std::placeholders::_1;

class MinimalSubscriber : public rclcpp::Node
{
public:
    MinimalSubscriber()
            : Node("minimal_subscriber")
    {
        subscription_ = this->create_subscription<std_msgs::msg::String>(
                "topic", 1, std::bind(&MinimalSubscriber::topic_callback, this, _1));
        last_time =  rclcpp::Node::now();
    }

private:
    void topic_callback(const std_msgs::msg::String::SharedPtr msg)
    {
        auto time_now = rclcpp::Node::now();
        auto time_passed = time_now - last_time;
        RCLCPP_INFO(this->get_logger(), "%f ms passed since last message",
                    time_passed.seconds() * 1000);
        last_time = time_now;
    }
    rclcpp::Subscription<std_msgs::msg::String>::SharedPtr subscription_;
    rclcpp::Time last_time;
};

int main(int argc, char * argv[])
{
    rclcpp::init(argc, argv);
    rclcpp::spin(std::make_shared<MinimalSubscriber>());
    rclcpp::shutdown();
    return 0;
}
```
What it does is to generate a large string and publishing it. The subscriber then measures the passed time between the arrival of this message and the last message.
#### Expected behavior
Since messages are sent every 100ms I would expect about this much time between the arrival of the messages
#### Actual behavior
Most of the times it does work. However  sometimes messages arrive 200 ms or even 300 ms apart.
```
[INFO] [1592217390.549296582] [minimal_subscriber]: 97.179156 ms passed since last message
[INFO] [1592217390.659681077] [minimal_subscriber]: 110.383583 ms passed since last message
[INFO] [1592217390.757322661] [minimal_subscriber]: 97.642025 ms passed since last message
[INFO] [1592217390.950667916] [minimal_subscriber]: 193.345806 ms passed since last message
[INFO] [1592217391.049816656] [minimal_subscriber]: 99.147818 ms passed since last message
[INFO] [1592217391.149279756] [minimal_subscriber]: 99.463591 ms passed since last message
[INFO] [1592217391.249153896] [minimal_subscriber]: 99.875162 ms passed since last message
[INFO] [1592217391.551635286] [minimal_subscriber]: 302.481089 ms passed since last message
[INFO] [1592217391.653201138] [minimal_subscriber]: 101.565111 ms passed since last message
[INFO] [1592217391.749771664] [minimal_subscriber]: 96.570076 ms passed since last message
[INFO] [1592217391.849963019] [minimal_subscriber]: 100.191705 ms passed since last message
[INFO] [1592217391.949536646] [minimal_subscriber]: 99.573136 ms passed since last message
[INFO] [1592217392.049755252] [minimal_subscriber]: 100.219387 ms passed since last message
[INFO] [1592217392.248516973] [minimal_subscriber]: 198.761311 ms passed since last message
[INFO] [1592217392.351339091] [minimal_subscriber]: 102.822849 ms passed since last message
```
#### Additional information
In the above example I used FastRTPS. The result gets worse with Cyclone DDS and with RTI Connext I get a message about once every 6 seconds (I thought it uses shared memory for local communication?!?)

clalancette, 2020-06-15:
Unfortunately, there is no one-size-fits-all solution to this problem.

For Cyclone, we have been doing some debugging with the developers of Cyclone to improve the situation with large messages.  You can see the results of that debugging at https://github.com/eclipse-cyclonedds/cyclonedds/issues/484 .

For Connext, we recently published a piece of documentation detailing how to tune it for large messages: https://index.ros.org/doc/ros2/Troubleshooting/DDS-tuning/

There is also some general troubleshooting information on that page.  I know none of this is an immediate solution to your problem, but any additional information (like wireshark captures, etc) you can provide is of course helpful.

marcbone, 2020-06-15:
Thank you for your fast response. Seems like the DDS tuning guide is new (could not find it two weeks ago)

I managed to get it to work under cyclone, but not on the others. But it is still happening very very rarely. Yet you have the complete problem again when you set the message size to 50MB. Increasing the "MinimumSocketReceiveBufferSize" did not solve the probelm and I could not increase `rmem_max` as it apparently only accepts specific values and I could not find out how this works (and I guess it probably affects my other network traffic negatively?).

I am wondering why this is even needed. ROS1 also uses the loopback device for communication and there it works without problems and is about twice as fast for 40 Mb messages (50ms vs 100ms). So what is makes it so different for DDS?

I could provide wireshark captures if needed but I guess this is not a problem speciffic to my machine and regarding the message size these dumps would get huge.

clalancette, 2020-06-15:
> I managed to get it to work under cyclone, but not on the others. But it is still happening very very rarely. Yet you have the complete problem again when you set the message size to 50MB. [...]

Hm, that's weird.  `rmem_max` should accept arbitrary values up to 2 billion.  It won't accept larger than that.  What's the error you got?

Also, increasing `rmem_max` shouldn't have an affect on your other network traffic, assuming that it is well behaved.  That is, `rmem_max` is limiting the largest receive buffer that any single process can request.  Increasing it allows processes to request much larger receive buffers.  So the main downside is that if you have a poorly-behaved process, it can request a really large buffer and use up way more memory than it should.  Otherwise, there should be no negative effects from tuning that.

As I said above, we are working with Cyclone to improve this, so it works better out-of-the-box.

marcbone, 2020-06-15:
> Hm, that's weird. rmem_max should accept arbitrary values up to 2 billion.
Oh so I guess 2147483647 is already the largest value.

> As I said above, we are working with Cyclone to improve this, so it works better out-of-the-box.
I get that cyclone is very new. But as the other DDS implemenations have the same problem I thought that there is maybe something in the protocol itself which generates this overhead or prevents it from being as fast as ROS1

eboasson, 2020-06-15:
@marcbone, I’m the primary Cyclone guy around — and rather grateful for the help @clalancette has been giving me, trying things for me. Last round I broke something quite badly, it seems, but that’ll get resolved. Part of the problem is that, try as I might, on my MBP I just seem to be unable to reproduce the problems he sees.

To your question:

> But as the other DDS implementations have the same problem I thought that there is maybe something in the protocol itself which generates this overhead or prevents it from being as fast as ROS1

Yes, there is something in the protocol that makes it exceedingly hard to do it well. The support for large samples was grafted onto the DDSI-RTPS protocol rather late in the game and the nastiest consequence is that one only ever acknowledges full samples. Because of that, tricks like doing flow-control based on acknowledgements and retransmit requests, which works well enough for small samples, falls apart completely. So you have to find a way of adding some non-specified behaviours, that still interoperate well enough with other implementations with slightly different implementation choices.

Be a little too aggressive in retransmitting and you generate a lot of useless traffic; be a little too reticent and you don’t come close to the hardware throughput. (Screw up and it stalls completely 😖) When everything goes smoothly you do end up saturating 10GbE networks, so it is not like it is impossible to get good performance, it is just difficult to get good performance under all kinds of real-world behaviours.

Another problem that implementations of the protocol face is that they operate at application level, where there just is no detailed information available about the amount of available buffer space in the kernel. Furthermore, you don’t have a receive buffer per-peer (like in TCP), but rather have all peers write to the same receive buffer. So 1-to-1 flows have different characteristics than N-to-1 flows. This means there is just no way to do transmit/receive window management like TCP can do it, and so you can never guarantee that a the receiver has enough buffer space to store a burst of packets.

(Most implementations do provide a way of using TCP instead of UDP, that takes away some of these problems while bringing some other complications. One of them is that there is no interoperability yet using TCP.)

fujitatomoya, 2020-06-16:
CC: @MiguelCompany

fujitatomoya, 2020-06-16:
I created colcon project for the reproduction based on bug report.

the procedure is the following,

```
# cd <colcon_workspace>/src
# git clone git@github.com:fujitatomoya/ros2_test_prover.git
# cd <colcon_workspace>
# colcon build --symlink-install
# source install/local_setup.bash
# ros2 run ros2_test_prover ros2_946_pub
# ros2 run ros2_test_prover ros2_946_sub
```

- eProsima fast-dds result on AMD Ryzen 9 3900X/Ubuntu20.04/source build

```
[INFO] [1592270707.026378292] [minimal_subscriber]: 97.890522 ms passed since last message
[INFO] [1592270707.133408466] [minimal_subscriber]: 107.025764 ms passed since last message
[INFO] [1592270707.233236110] [minimal_subscriber]: 99.827634 ms passed since last message
[INFO] [1592270707.325747761] [minimal_subscriber]: 92.514991 ms passed since last message
[INFO] [1592270707.433194253] [minimal_subscriber]: 107.443422 ms passed since last message
[INFO] [1592270707.525831659] [minimal_subscriber]: 92.640866 ms passed since last message
[INFO] [1592270707.632868714] [minimal_subscriber]: 107.033695 ms passed since last message
[INFO] [1592270707.725995604] [minimal_subscriber]: 93.130460 ms passed since last message
[INFO] [1592270707.833281173] [minimal_subscriber]: 107.281809 ms passed since last message
```

- ADLINK cyclonedds result on AMD Ryzen 9 3900X/Ubuntu20.04/source build

```
[INFO] [1592271144.902212180] [minimal_subscriber]: 96.797391 ms passed since last message
[INFO] [1592271144.999012880] [minimal_subscriber]: 96.803670 ms passed since last message
[INFO] [1592271145.099128598] [minimal_subscriber]: 100.115658 ms passed since last message
[INFO] [1592271145.199684262] [minimal_subscriber]: 100.555824 ms passed since last message
[INFO] [1592271145.299835596] [minimal_subscriber]: 100.151324 ms passed since last message
[INFO] [1592271145.400012758] [minimal_subscriber]: 100.176852 ms passed since last message
[INFO] [1592271145.499601076] [minimal_subscriber]: 99.588378 ms passed since last message
[INFO] [1592271145.599914100] [minimal_subscriber]: 100.313084 ms passed since last message
[INFO] [1592271145.699549426] [minimal_subscriber]: 99.635526 ms passed since last message
```

fujitatomoya, 2020-06-16:
@marcbone 

could you describe your platform a little bit more if possible?

marcbone, 2020-06-16:
@fujitatomoya 
I tried your project and for fastrtps my results differ from yours (I get the 200 or 300 ms delay). On cyclone I get the same results as you (apart from very rarely where I still get 200ms sometimes). but as soon as you increase the string size to 50MB you can see the same problem.

I have to admit that my hardware is very new and has some other problems. I am using Ubuntu 20.04 with the ros2-testing repo on an Asus Zephyrus G14 laptop with an ryzen 4900HS cpu. I also have a more solid Thinkpad (18.04 - eloquent), where I tried your repo but I get some cmake error. I will investigate and look if I have the same problem on the thinkpad.

fujitatomoya, 2020-06-16:
@marcbone 

thanks for the information. i see that, the bigger files the more probability for delay or message drops. i am really interested in your criteria and use case, could you share your thoughts if possible?

marcbone, 2020-06-16:
I do not have a real use case. I try to compare the communication performance between ROS1 and ROS2. But a real world scenario could be transferring high res images without nodelets.  For example the Basler daA3840-45uc color camera has a resolution of 3840 x 2160. With 3 Bytes per pixel that would be around 25 MB per image. In ROS1 that would be no issue apart from a bit more latency (around 30-40 ms). However, in ROS2 the latency is much higher and the dropped messages could be a problem that forces you to choose a different framework.

eboasson, 2020-06-17:
@marcbone would you be willing to give https://github.com/eboasson/cyclonedds/tree/rexmit-improvements a try? It contains the changes I made for the PR that @clalancette mentioned (https://github.com/eclipse-cyclonedds/cyclonedds/issues/484) and so far it appears to really improve things quite a bit (at least when the socket buffers are small, i.e., out-of-the-box behaviour).

I have the kernel of a tweak that I think will bring it closer to TCP performance, but I think I'll give priority to merging the improvements so far into Cyclone master. It sounds like your scenario is a good one to check out before doing that merge.

marcbone, 2020-06-17:
@eboasson I tested your branch. 
The 10 MB message without any settings like rmem or minimum buffer size feels like what I got with these settings with the default cyclone. With the rmem settings it gets even better but there are still some droppings. with the minimum receive buffer it is even better and droppings happen only about once per minute.

For 50 MB messages the droppings occur more often and there is often at least 500ms between the messages or more (up to 2 seconds)

marcbone, 2020-06-17:
So this time with the right branch. First I want to say that this branch brings down the latency for 4 MB messages from 8 ms to about 5 ms. I would say that it is now almost as fast as fastrtps (about 4ms in best effort mode) but without all the dropped messages.

With 50 MB and no special settings I dont get any message droppings but there is now a lot more spread between the messages:

```
[INFO] [1592430834.162961796] [minimal_subscriber]: 103.122466 ms passed since last message
[INFO] [1592430834.259356535] [minimal_subscriber]: 96.394117 ms passed since last message
[INFO] [1592430834.389298428] [minimal_subscriber]: 129.942094 ms passed since last message
[INFO] [1592430834.473835525] [minimal_subscriber]: 84.536816 ms passed since last message
[INFO] [1592430834.557189503] [minimal_subscriber]: 83.354820 ms passed since last message
[INFO] [1592430834.644789637] [minimal_subscriber]: 87.598481 ms passed since last message
[INFO] [1592430834.776496965] [minimal_subscriber]: 131.708380 ms passed since last message
[INFO] [1592430834.840916423] [minimal_subscriber]: 64.419948 ms passed since last message
[INFO] [1592430834.960471700] [minimal_subscriber]: 119.554886 ms passed since last message
[INFO] [1592430835.055590527] [minimal_subscriber]: 95.108629 ms passed since last message
[INFO] [1592430835.156698894] [minimal_subscriber]: 101.118896 ms passed since last message

```

for the 10 MB messages I get a few message droppings once in a while but not with 50 MB messages :shrug: . 

With rmem and minimum buffersize there are no message droppings for 10 MB anymore and the spread for 50 Mb messages is normal again.

So apart from the spread for 50 MB messages this is a great improvement :+1:

eboasson, 2020-06-18:
@marcbone thanks! Even if it isn't perfect yet, at least it gives some more confidence that I didn't break anything.

The increased variances I suspect to be caused by my now using smaller datagrams when retransmitting than when transmitting the message for the first time. Generally single-packet datagrams have the best change of making it through a bad network, and retransmitting with a bit more overhead but a higher likelihood of success makes sense to me.

Regarding the dropped messages: it is only acceptable for messages to be dropped if
1. the subscription is best-effort;
2. if there are temporary disconnections;
3. the subscription or publisher has a keep-last history.

I'm pretty sure the default QoS is reliable in ROS 2, so that rules out (1). It is not (2) either, I have seen it in the wild but it takes more craziness to cause those. But (3) is very likely, I believe the `1` in the calls to `create_subscriber` and `create_publisher` means there's only a single place buffer on both sides, so you can overwrite data on either side. You might want to play with that a bit.

Generally speaking, to get the best performance you need to make sure there is no packet loss whatsoever. Experience shows that having a socket receive buffer large enough to easily hold a full sample takes care of that. A smaller one should be enough because the reader does start draining the buffer as soon as packets start arriving, but burst rates on loopback are really high and scheduling latency is an issue. So my changes are aimed at getting acceptable out-of-the-box behaviour with no changes to any buffer settings.

(As an aside: another nice way to cause problems is to have a switch or WiFi access point along the path where the bandwidth drops considerably. Up to a certain size it all works fine, and then suddenly it falls apart. Add a bit of traffic shaping on the sender and it runs smoothly again. It is a tricky business and it'll be a while before it approaches perfection ...)

clalancette, 2020-06-18:
To summarize here:

- Since Connext is closed-source, there is not much we can do beyond tuning.  That is covered in https://index.ros.org/doc/ros2/Troubleshooting/DDS-tuning/
- Cyclone is being worked on under the auspices of https://github.com/eclipse-cyclonedds/cyclonedds/issues/484
- It's not clear to me what the Fast-RTPS situation is.  I think there is still a problem there, but I'm honestly not sure.  @marcbone If you are still having problems with Fast-RTPS, I encourage you to open a bug against https://github.com/eProsima/Fast-DDS/issues

Since this is being worked on under different bugs for the individual DDS vendors, I'm going to go ahead and close this issue.  Please feel free to keep commenting or reopen if you think this is in error.

closed 2020-06-18 by clalancette