ros2/ros2 #947: package not found

lh315936716, 2020-06-16:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10

- Installation type:
  - binaries
- Version or commit hash:
  -debug 

I installed the latest version of ROS2 as I found it is supported on Windows 10. When I run the command **ros2 run demo_nodes_cpp talker**, it said Package **'demo_nodes_cpp"** not found.
Then I try **ros2 pkg list**， nothing was shown on screen. What should I do in this situation?
Thanks a lot!

flynneva, 2020-06-16:
seems more like a question for ROS answers:  https://answers.ros.org/questions/

I would double check you are [sourcing the underlay](https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Install-Binary/#environment-setup) in **every terminal** you open. a good test to see if this worked is to simply run `ros2 -h`. you should see printed out advice on what you can do with the ros2 CLI

lh315936716, 2020-06-17:
> seems more like a question for ROS answers: https://answers.ros.org/questions/
>
> I would double check you are [sourcing the [...]

Hello flynneva, 
Thank you for your suggestions. It was the sourcing problem indeed. I called local_setup.bat in terminal following the instruction of [installation tutorial ](https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Install-Binary/#environment-setup). An warnning message which is "[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[C:\Program Files\rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available." was returned. 
Then I called /local_setup.ps1 and it worked. The problem I met was fixed then.

closed 2020-06-17 by dirk-thomas