ros2/ros2 #948: Mac OS Catalina fails - fatal error: 'Availability.h' file not found

raix, 2020-06-16:
## Bug report

**Required Info:**

- Operating System:
Mac OS Catalina 10.15.5
- Installation type:
Following the guide
- Version or commit hash:
 foxy
- DDS implementation:

- Client library (if applicable):

#### Steps to reproduce issue
https://index.ros.org/doc/ros2/Installation/Foxy/macOS-Development-Setup/

```
$ colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> fastcdr
Finished <<< gtest_vendor [0.99s]                                                                                                       
Starting >>> gmock_vendor
--- stderr: fastcdr                                                                                                                
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: for: /Users/raix/ros2_foxy/install/fastcdr/lib/libfastcdr.1.0.13.dylib (for architecture x86_64) option "-add_rpath /Users/raix/ros2_foxy/install/fastcdr/lib" would duplicate path, file already has LC_RPATH for: /Users/raix/ros2_foxy/install/fastcdr/lib
---
Finished <<< fastcdr [1.11s]
Starting >>> ament_cppcheck
Finished <<< ament_package [1.62s]                                                                                                
Finished <<< ament_lint [1.66s]
Starting >>> ament_cmake_core                                                            
Starting >>> ament_flake8
Finished <<< gmock_vendor [0.84s]                                                                                                        
Starting >>> cyclonedds
Finished <<< ament_cppcheck [1.29s]                                                                                                        
Starting >>> osrf_pycommon
Finished <<< ament_cmake_core [1.25s]                                                                                                       
Starting >>> ament_cmake_libraries
Finished <<< ament_flake8 [1.36s]                                                                                                             
Starting >>> ament_cmake_export_libraries
--- stderr: cyclonedds                                                                                                                                                 
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: no LC_RPATH load command with path: /Users/raix/ros2_foxy/build/cyclonedds/lib found in: /Users/raix/ros2_foxy/install/cyclonedds/bin/pubsub (for architecture x86_64), required for specified option "-delete_rpath /Users/raix/ros2_foxy/build/cyclonedds/lib"
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: for: /Users/raix/ros2_foxy/install/cyclonedds/bin/pubsub (for architecture x86_64) option "-add_rpath @loader_path/../lib" would duplicate path, file already has LC_RPATH for: @loader_path/../lib
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: no LC_RPATH load command with path: /Users/raix/ros2_foxy/build/cyclonedds/lib found in: /Users/raix/ros2_foxy/install/cyclonedds/bin/ddsls (for architecture x86_64), required for specified option "-delete_rpath /Users/raix/ros2_foxy/build/cyclonedds/lib"
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: for: /Users/raix/ros2_foxy/install/cyclonedds/bin/ddsls (for architecture x86_64) option "-add_rpath @loader_path/../lib" would duplicate path, file already has LC_RPATH for: @loader_path/../lib
---
Finished <<< cyclonedds [1.51s]
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_libraries [0.57s]                                                                                                                                                    
Starting >>> ament_cmake_python
Finished <<< ament_cmake_export_libraries [0.52s]                                                                                                                                   
Starting >>> ament_cmake_export_dependencies
Finished <<< osrf_pycommon [1.34s]                                                                                                                                                            
Starting >>> ament_cmake_export_definitions
Finished <<< ament_cmake_include_directories [0.51s]                                                                                                                       
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_python [0.60s]                                                                                                                                     
Finished <<< ament_cmake_export_dependencies [0.53s]                                                                                                                                      
Starting >>> ament_cmake_test
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_export_definitions [0.58s]                                                                                                                            
Starting >>> ament_cmake_version
Finished <<< ament_cmake_export_include_directories [0.64s]                                                                                                                                          
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_export_link_flags [0.62s]                                                                                                                                                    
Starting >>> ament_cmake_export_targets
Finished <<< ament_cmake_test [0.76s]                                                                                                                                                      
Starting >>> ament_cmake_target_dependencies                                                                                                               
[... 169 lines cut ...]
make[1]: *** [CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
---
Failed   <<< rviz_ogre_vendor [7.50s, exited with code 2]
Aborted  <<< rosidl_typesupport_fastrtps_cpp [0.63s]                                                                                                                         
Aborted  <<< rosidl_typesupport_connext_cpp [3.89s]                                                                                 
Aborted  <<< resource_retriever [10.6s]                                     

Summary: 93 packages finished [43.5s]
  1 package failed: rviz_ogre_vendor
  3 packages aborted: resource_retriever rosidl_typesupport_connext_cpp rosidl_typesupport_fastrtps_cpp
  7 packages had stderr output: cyclonedds fastcdr orocos_kdl rosidl_typesupport_connext_cpp rosidl_typesupport_introspection_cpp rviz_ogre_vendor urdfdom
  204 packages not processed
```

#### Expected behavior
Should install
#### Actual behavior
fatal error: 'Availability.h' file not found

#### Open questions
Not sure if ros2 supports Catalina or if it's still only Mojave - if so feel free to close this issue.

Note: I did not have have no intention of disabling SIP - that's the only point where I did not follow the guide. *(not sure if this that is a hard requirement / if it triggers this issue or if this issue is caused by an OS deprecation of library files)*

clalancette, 2020-06-16:
There are 3 things here:

1.  You need to install the full Xcode, not just Xcode Command-line tools.  The instructions say that now: https://index.ros.org/doc/ros2/Installation/Foxy/macOS-Development-Setup/
1.  We only support Mojave at the moment.
1.  You'll be able to build without disabling SIP, but you won't be able to execute pretty much anything.

Given all of the above, I'm going to close this out.  I'll suggest you open questions like this against https://answers.ros.org in the future, as it is a more central place for these kinds of questions.

closed 2020-06-16 by clalancette

wolfv, 2021-02-22:
hmm, unfortunately it doesn't seem that the situation has changed in the past 6 months, has it?

I was running into other issues in rosbag when compiling with 10.14 SDK (because rosbag2 is using the `std::filesystem` API that seems to be only available on 10.15... but now running into this for OGRE:

clalancette, 2021-02-22:
> I was running into other issues in rosbag when compiling with 10.14 SDK (because rosbag2 is using the `std::filesystem` API that seems to be only available on 10.15... but now running into this for OGRE:

rosbag2 does *not* use the `std::filesystem` API (currently, nothing in ROS 2 does).  It uses `rcpputils::filesystem`, which is a more-or-less drop-in replacement for `std::filesystem`.

For what it is worth, I have been able to successfully compile all of ROS 2 on macOS Catalina with Xcode 12.3 (I haven't gotten around to upgrading to 12.4).  It still isn't officially supported, but as far as I can tell it should work.

It's been a while, but I believe that problem with `Availability.h` comes from having either Xcode or Xcode command-line tools installed improperly.  I would try reinstalling them, as it should work.

clalancette, 2021-02-22:
Oh, one note: to compile on Catalina, I did have to disable the `-Werror` flag on rosbag2.  Otherwise, it should build out-of-the-box.

wolfv, 2021-02-22:
std::filesystem is used here: https://github.com/ros/pluginlib/blob/9fecba30c6e388024bc14137b53694b4e555696f/pluginlib/include/pluginlib/class_loader_imp.hpp#L539

and then from here:

https://github.com/ros/pluginlib/blob/9fecba30c6e388024bc14137b53694b4e555696f/pluginlib/include/pluginlib/impl/filesystem_helper.hpp#L47

I think with proper OSX deployment target detection this would work with an older deployment target than 10.15 as well, but in it's current state it definitely requires 10.15.  This is the run where rosbag2 was failing: https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=1468&view=logs&j=d8aa8ae6-4ee0-509c-2416-7578dbf33661&t=92bab187-11ce-5b74-46b1-29d1132b53a5&l=1462

And here is the run where we got much further, but then OGRE is failing (on the way to RViz 2): https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=1477&view=logs&j=e51fa4e6-06f0-5082-594c-751befdf766b&t=cb0520f8-7dc9-51e0-c0f9-886b99e2037d

On the positive note: we made really good progress on adding ROS 2 Foxy packages to the conda-package collection! :) RViz 2 for OS X is within reach (on Linux it's already working)! Now we just need to fix OGRE (we can de-vendor it as well, and use the package from conda-forge), and add Windows support and we have a solid story for Foxy. cc @nuclearsandwich @tfoote 

PS here is a related issue: https://github.com/ros-perception/image_common/issues/139

clalancette, 2021-02-22:
> https://github.com/ros/pluginlib/blob/9fecba30c6e388024bc14137b53694b4e555696f/pluginlib/include/pluginlib/impl/filesystem_helper.hpp#L47

But that is only if you build with `-std=c++17` or later.  In our default install, we use `-std=c++14` and we don't use `std::filesystem` at all.  We can't; our currently supported macOS client is Mojave, and Mojave doesn't have `std::filesystem`.

wolfv, 2021-02-22:
Thanks for the hint! We could also build with c++14.