ros2/ros2 #950: colcon build --symlink-install failing at cyclonedds when building from source in a docker container : Error : try_compile() works only for enabled languages.  Currently these are: C

akshaysarvesh25, 2020-06-17:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 

#### Steps to reproduce issue
The source installation of ROS2 is being performed as per the link : https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/
The following is the error message I am getting :
```
Finished <<< osrf_pycommon [2.06s]
Finished <<< ament_pycodestyle [2.07s]
Starting >>> ament_pyflakes
--- stderr: cyclonedds
CMake Error at /usr/share/cmake-3.10/Modules/CheckCXXSourceCompiles.cmake:97 (try_compile):
  Unknown extension ".cxx" for file

    /root/ros2_dashing/build/cyclonedds/CMakeFiles/CMakeTmp/src.cxx

  try_compile() works only for enabled languages.  Currently these are:

    C

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/CheckCXXCompilerFlag.cmake:49 (CHECK_CXX_SOURCE_COMPILES)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:215 (check_cxx_compiler_flag)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:374 (_test_compiler_hidden_visibility)
  src/security/builtin_plugins/access_control/CMakeLists.txt:23 (generate_export_header)

CMake Error at /usr/share/cmake-3.10/Modules/CheckCXXSourceCompiles.cmake:97 (try_compile):
  Unknown extension ".cxx" for file

    /root/ros2_dashing/build/cyclonedds/CMakeFiles/CMakeTmp/src.cxx

  try_compile() works only for enabled languages.  Currently these are:

    C

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/CheckCXXCompilerFlag.cmake:49 (CHECK_CXX_SOURCE_COMPILES)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:216 (check_cxx_compiler_flag)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:374 (_test_compiler_hidden_visibility)
  src/security/builtin_plugins/access_control/CMakeLists.txt:23 (generate_export_header)

CMake Error at /usr/share/cmake-3.10/Modules/CheckCXXSourceCompiles.cmake:97 (try_compile):
  Unknown extension ".cxx" for file

    /root/ros2_dashing/build/cyclonedds/CMakeFiles/CMakeTmp/src.cxx

  try_compile() works only for enabled languages.  Currently these are:

    C

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:191 (check_cxx_source_compiles)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:234 (_check_cxx_compiler_attribute)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:375 (_test_compiler_has_deprecated)
  src/security/builtin_plugins/access_control/CMakeLists.txt:23 (generate_export_header)

CMake Error at /usr/share/cmake-3.10/Modules/CheckCXXSourceCompiles.cmake:97 (try_compile):
  Unknown extension ".cxx" for file

    /root/ros2_dashing/build/cyclonedds/CMakeFiles/CMakeTmp/src.cxx

  try_compile() works only for enabled languages.  Currently these are:

    C

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:191 (check_cxx_source_compiles)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:240 (_check_cxx_compiler_attribute)
  /usr/share/cmake-3.10/Modules/GenerateExportHeader.cmake:375 (_test_compiler_has_deprecated)
  src/security/builtin_plugins/access_control/CMakeLists.txt:23 (generate_export_header)

You have called ADD_LIBRARY for library ddsc without any source files. This typically indicates a problem with your CMakeLists.txt file
---
Failed   <<< cyclonedds [2.48s, exited with code 1]
[5.253s] WARNING:asyncio:Unknown child process pid 936, will report returncode 255
Aborted  <<< ament_flake8 [0.80s]
Aborted  <<< fastcdr [2.84s]
Aborted  <<< urdfdom_headers [2.82s]
Aborted  <<< ament_pyflakes [0.75s]
Aborted  <<< ament_cmake_core [0.82s]
Aborted  <<< osrf_testing_tools_cpp [3.18s]

Summary: 7 packages finished [5.51s]
  1 package failed: cyclonedds
  6 packages aborted: ament_cmake_core ament_flake8 ament_pyflakes fastcdr osrf_testing_tools_cpp urdfdom_headers
  1 package had stderr output: cyclonedds
  289 packages not processed
The command '/bin/sh -c colcon build --symlink-install' returned a non-zero code: 1

```
The following is the code for the Dockerfile
```
FROM ubuntu:bionic
#FROM osrf/ros:dashing-desktop
FROM ros:melodic-ros-core-bionic

RUN apt-get update
RUN apt-get -y install locales lsb-release wget
RUN apt-get update && apt-get install -y gnupg2 cppcheck

RUN locale-gen en_US en_US.UTF-8
RUN update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
RUN export LANG=en_US.UTF-8

ENV DEBIAN_FRONTEND=noninteractive
ENV DEBCONF_NONINTERACTIVE_SEEN true

RUN sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'

RUN apt-get update &&  apt-get install -y build-essential cmake git python3-colcon-common-extensions python3-pip python-rosdep python3-vcstool wget cppcheck

RUN cmake --version

RUN python3 -m pip install -U argcomplete flake8 flake8-blind-except flake8-builtins flake8-class-newline flake8-comprehensions flake8-deprecated flake8-docstrings flake8-import-order flake8-quotes pytest-repeat pytest-rerunfailures pytest pytest-cov pytest-runner setuptools

RUN apt-get install --no-install-recommends -y libasio-dev libtinyxml2-dev

RUN apt-get install --no-install-recommends -y libcunit1-dev

RUN mkdir /root/ros2_dashing
RUN mkdir /root/ros2_dashing/src
WORKDIR /root/ros2_dashing
RUN wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
RUN vcs-import src < ros2.repos

RUN rosdep init
RUN rosdep update
RUN rosdep install --from-paths src --ignore-src --rosdistro dashing -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers"

RUN cd ~/ros2_dashing/
RUN colcon build --symlink-install
#RUN colcon build --symlink-install --packages-skip cyclonedds rmw_cyclonedds_cpp

CMD ["bash"]

```
The cmake version used is : 3.10.2

Any suggestions on how to fix this?

clalancette, 2020-06-17:
See https://github.com/eclipse-cyclonedds/cyclonedds/issues/546, with a fix in https://github.com/eclipse-cyclonedds/cyclonedds/pull/547 .

closed 2020-06-17 by clalancette

akshaysarvesh25, 2020-06-17:
Thanks for the suggestion, I couldn't fix the original Dockerfile with ubuntu:bionic and ros2:dashing. Instead, I changed the Dockerfile to use ubuntu:focal. And, as ros2:dashing is incompatible with ubuntu:focal, I had to switch to ros2:noetic.

Here is the new Dockerfile which works:
```
FROM ubuntu:focal
#FROM osrf/ros:dashing-desktop
FROM ros:noetic-ros-core-focal
RUN apt-get update
RUN apt-get -y install locales lsb-release wget
RUN apt-get update && apt-get install -y gnupg2

RUN locale-gen en_US en_US.UTF-8
RUN update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
RUN export LANG=en_US.UTF-8

ENV DEBIAN_FRONTEND=noninteractive
ENV DEBCONF_NONINTERACTIVE_SEEN true

RUN apt update && apt install curl
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -

RUN sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'

RUN  apt update &&  apt install -y build-essential cmake git libbullet-dev python3-colcon-common-extensions python3-flake8 python3-pip python3-pytest-cov python3-rosdep python3-setuptools python3-vcstool wget
#RUN sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'

#RUN apt-get update &&  apt-get install -y build-essential cmake git python3-colcon-common-extensions python3-pip python-rosdep python3-vcstool wget cppcheck

RUN cmake --version
RUN cmake --version

RUN python3 -m pip install -U argcomplete flake8 flake8-blind-except flake8-builtins flake8-class-newline flake8-comprehensions flake8-deprecated flake8-docstrings flake8-import-order flake8-quotes pytest-repeat pytest-rerunfailures pytest pytest-cov pytest-runner setuptools

RUN apt-get install --no-install-recommends -y libasio-dev libtinyxml2-dev

RUN apt-get install --no-install-recommends -y libcunit1-dev

RUN mkdir /root/ros2_foxy
RUN mkdir /root/ros2_foxy/src
WORKDIR /root/ros2_foxy
#RUN wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
RUN wget https://raw.githubusercontent.com/ros2/ros2/foxy/ros2.repos
RUN vcs-import src < ros2.repos

```

dirk-thomas, 2020-06-18:
@akshaysarvesh25 You are mixing ROS 1 and ROS 2 distros here. `Dashing` is a ROS 2 distro targeting Ubuntu 18.04. If you want to target ROS 2 on Ubuntu 20.04 you should use `Foxy` - `Noetic` is a ROS 1 distribution.

akshaysarvesh25, 2020-06-18:
@dirk-thomas , you are right. I want ROS (binary installed) as well as ROS2 (built from source) in a Ubuntu docker for my project. My intention was to use ROS2 ```dashing``` with ROS1 ```melodic``` on Ubuntu 18.04. However, the build for that is failing due to the error mentioned above. To circumvent that, I'm now using ROS1 ```Noetic``` on a Ubuntu 20.04 docker platform with ROS2 ```Foxy```. If you follow the above code in the Dockerfile towards the end, I'm installing ROS2 ```Foxy``` by building it from source.

Is there a better way to do this?

dirk-thomas, 2020-06-18:
I am sorry I missed the intention to do dual-ROS versions. Your approach is fine as is - if you really want / need to build ROS 2 from source.

There is a Docker image which already combines Noetic and Foxy: `osrf/ros:foxy-ros1-bridge`. Maybe using that as a starting point and only build the packages you need to customize might be faster.

akshaysarvesh25, 2020-06-18:
Ah, I see. I will check it out. Thank you @dirk-thomas .

k0ekk0ek, 2020-06-19:
For future reference: eclipse-cyclonedds/cyclonedds#550 will be used instead.