ros2/ros2 #951: [Meta-ticket]: Add pytest.ini to all packages which use pytest

clalancette, 2020-06-18:
In the current state, if you run any of the pytest tests locally, you get warning messages like:

```
/home/ubuntu/.local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  /home/ubuntu/.local/lib/python3.8/site-packages/_pytest/junitxml.py:417: PytestDeprecationWarning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.
    _issue_warning_captured(deprecated.JUNIT_XML_DEFAULT_FAMILY, config.hook, 2)
```

We currently workaround this in CI by always adding a pytest.ini before starting tests (see https://github.com/ros2/ci/pull/471), but it would be nice to get rid of that and just have local and CI work the same.

This is a tracking bug for all of the places we need to change it.  Once I've opened the relevant PRs, I'll run one CI job for all of them here.

clalancette, 2020-06-18:
We'll also likely have to backport these to Dashing and Eloquent to keep the warnings down there.

clalancette, 2020-06-18:
CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

clalancette, 2020-06-19:
The single failure in macOS and Windows is there in nightlies.  So I think these are good to start reviewing and merging.

clalancette, 2020-06-19:
All right, with everything merged, let's see how a CI job does with the pytest.ini workaround removed:

* Linux [image]

clalancette, 2020-06-19:
I totally missed the whole ament_lint repository, so the failure in the last job is because of that.  Here's another try, with that PR in place and the one to micro-ROS to see how we are doing:

* Linux [image]

clalancette, 2020-06-22:
I was missing ament_cmake_ros as well.  Another try:

* Linux [image]

clalancette, 2020-06-23:
Here's a regular CI, just to show that the linked PRs don't have regressions:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

clalancette, 2020-06-23:
All right, all failures in CI are also in nightlies.

clalancette, 2020-06-30:
All right, these are all in now.  In theory, we could remove the pytest.ini workaround from CI; however, if we do that, we'll break CI for Dashing, Eloquent, and Foxy.  The obvious solution is to backport all of these changes to all of those active distributions, which would finally allow us to close https://github.com/ros2/ci/pull/471 , but is also a lot of churn (27*3 == 81 PRs to do this).

dirk-thomas, 2020-06-30:
We could also just remove the CI workaround for Foxy and rolling to ensure we don't accidentally reintroduce a problem.

clalancette, 2020-08-05:
> We could also just remove the CI workaround for Foxy and rolling to ensure we don't accidentally reintroduce a problem.

All right, since I've now done this in https://github.com/ros2/ci/pull/497, I'm going to close this tracking ticket.  Thanks for the feedback and reviews here.

closed 2020-08-05 by clalancette