ros2/ros2 #953: Qt5 issues with ROS2 on MacOS

SebastianGrans, 2020-06-22:
## Bug report
I believe the issues I am describing are mostly caused by Homebrew and not so much by ROS. Nevertheless, I thought it could be a good idea to create an issue so that other people can easily find how to solve it.

**Required Info:**

- Operating System:
MacOS 10.15.5 Catalina 
- Installation type:
Foxy from binary. 

#### Steps to reproduce issue

I was following the [installation](https://index.ros.org/doc/ros2/Installation/) instructions, and then continued to follow the [tutorials](https://index.ros.org/doc/ros2/Tutorials/). 

Everything worked smoothly until I came to the [Creating a workspace](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#resolve-dependencies) tutorial where I needed to install dependencies for a package using `rosdep`.

First issue:
```
> rosdep install -i --from-path src --rosdistro foxy -y
executing command [brew install qt5]
Warning: qt 5.15.0 is already installed and up-to-date
To reinstall 5.15.0, run `brew reinstall qt`
ERROR: the following rosdeps failed to install
```

I was able to resolve this by running `sudo ln -s qt qt5` as suggested in issue #490 [link to comment](https://github.com/ros-infrastructure/rosdep/issues/490#issuecomment-334959426). 

The next error came when I tried building using colcon which gave me the following error: 
```
> colcon build
Starting >>> turtlesim
--- stderr: turtlesim
CMake Error at /usr/local/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:14 (message):
  The imported target "Qt5::Core" references the file

     "/usr/local/.//mkspecs/macx-clang"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake:55 (_qt5_Core_check_file_exists)
  /usr/local/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:235 (include)
  /usr/local/lib/cmake/Qt5Gui/Qt5GuiConfig.cmake:93 (find_package)
  /usr/local/lib/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake:93 (find_package)
  /usr/local/lib/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< turtlesim [0.88s, exited with code 1]

Summary: 0 packages finished [1.18s]
  1 package failed: turtlesim
  1 package had stderr output: turtlesim
```

I resolved by following [this](https://github.com/Homebrew/legacy-homebrew/issues/29938#issuecomment-104722532) post which involves symlinking some directories from `/usr/local/Cellar` to `/usr/local/`.

```
ln -s /usr/local/Cellar/qt5/<insert version>/mkspecs /usr/local/mkspecs
ln -s /usr/local/Cellar/qt5/<insert version>/plugins /usr/local/plugins
```

I hope this helps someone 🙂

ahcorde, 2020-07-10:
Hello @SebastianGrans,

I was not able to reproduce your error in my MAC configuration, anyhow I will be happy to review a PR adding a section in the [Troubleshooting page for Windows](https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#windows). The repository to add this contribution is [ros2/ros2_documentation](https://github.com/ros2/ros2_documentation/)

SebastianGrans, 2020-07-13:
Strange. 

I noticed that the latter problem was already [documented](https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#qt-build-error-mkspecs-macx-clang-but-this-file-does-not-exist) there. I'll add the first problem and make a PR. 🙂

closed 2020-07-17 by SebastianGrans