ros2/ros2 #954: ROS 2 Galactic Geochelone

cottsay, 2020-06-23:
This meta ticket aggregates the tasks targeted for the next ROS 2 release, Galactic Geochelone.

Release date scheduled for May of 2021. More details about the release schedule can be found on the [Galactic Release Page](https://index.ros.org/doc/ros2/Releases/Release-Galactic-Geochelone/).

Tasks will be updated as they progress.

----

### Planned for Inclusion

- TBD

See [the release page](https://index.ros.org/doc/ros2/Releases/Release-Galactic-Geochelone/) for other changes included in Galactic.

---

### Roadmap

We hope to add more items to above list while we make progress and have more bandwidth available. More ideas for items to work on can be found on the [roadmap](https://index.ros.org/doc/ros2/Roadmap/). If you can help out on any roadmap items, let us know!

SteveMacenski, 2020-06-24:
Samsung
- Working on implementing designed CostmapFilters capabilities aligned with semantic navigation standard in development
  - [ ] Speed zones
  - [ ] Keep out zones
  - [ ] non-directional lane navigation
- [ ] Mentor community members in the mobile robot ecosystem to increase individual contributor rates and continue to encourage more companies to get directly involved in the success of their own open-source software needs/uses.
- [ ]  Push Navigation2 test coverage towards 80% goal by end of 2020 (currently at 70% up from ~50% in Eloquent)
- [ ] Create integrations with visual SLAM
- [ ] Create integrations with 3D SLAM
- [ ] Create integrations with existing AI-community capabilities 
- [ ] Add at minimum 1 new path planner to navigation2 to allow realistic planning capabilities for non-circular or non-small robots.

cottsay, 2021-05-23:
ROS 2 Galactic has been released: https://github.com/ros2/ros2/releases/tag/release-galactic-20210523

closed 2021-05-23 by cottsay