ros2/ros2 #957: Track master branch of ros2_tracing instead of release tag

christophebedard, 2020-06-25:
This is kind of a follow-up to https://github.com/ros2/ros2/pull/955#issuecomment-648936872 and the discussion under #898, which has a bunch of reasons why tracking a tag has been the right choice for `ros2_tracing` so far. The first reason is that this is what we agreed on when we integrated & added `ros2_tracing` to the core repos last year. Now I'd like to ask to revisit that and switch to tracking a branch.

@jacobperron seems to think it should be fine, but I'll go over the couple points that @nuclearsandwich nicely summarized under https://github.com/ros2/ros2/issues/898#issuecomment-615017143 and propose solutions.

> A major concern when reviewing a branch based inclusion was the fact that their change control process is not *exactly our process, that is, they don't run our official CI for each change. The problem is not resolved by independent CI.

One solution would be to give me access to ci.ros2.org so that I could run CI on all supported platforms. I was planning to add arm64 and Windows to our CI (RPi or Docker+qemu and AppVeyor), but it's time-consuming to set up and maintain, and ci.ros2.org can already do that.

> The above concern makes it possible for a change to make it through an external review and automated testing process but still break our builds. And that compounds with the fact that as none of us are maintainers of the external project, we can't change the upstream in order to resolve the issue. Our only option for resolution is re-targeting either to an unreleased commit as in #899 or pinning back to a previously known good commit.

I've recently given "developer" access (GitLab permission) to some Open Robotics people. This allows those people to push to non-protected branches and review merge requests. By changing a setting it can also allow those people to merge MRs on/to protected branches, which would allow them to merge fixes if @iluetkeb or myself can't get to it quickly enough.

> In general we're trying to keep dependencies tight and reliable. (Recently we even removed poco, an example in the earlier discussion of a widely used library, in favor of a more tightly scoped internal implementation of the needed features.)

I guess this point is more about confidence/trust in the maintainers, how reliable they are -- or have been -- and a bit about the nature of the dependency.

Even if `ros2_tracing` is far from being feature-complete, I think it has matured & stabilized.

Let me know what you think!

jacobperron, 2020-06-25:
I reason *for* switching back to a branch is to give us confidence with new releases of `ros2_tracing`. For example, https://github.com/ros2/ros2/pull/955 is bumping the version for Foxy, and I would have more confidence in the changes if we were using the same version (or similar) for our nightly builds.

nuclearsandwich, 2020-07-14:
> One solution would be to give me access to ci.ros2.org so that I could run CI on all supported platforms. I was planning to add arm64 and Windows to our CI (RPi or Docker+qemu and AppVeyor), but it's time-consuming to set up and maintain, and ci.ros2.org can already do that.

I am overall in favor of tracking master. The track record of communication and collaboration between the ROS 2 core team and the tracing team has created a strong foundation for further, closer, collaboration. One concern of mine is CI resource availability. Tracing is a low-level component in ROS 2 builds which means that it has a large surface area for potential issues. But if I understand correctly, since tracing is not enabled during CI builds the only packages which need to be built and tested are those that directly include references to tracing. Depending on how big that package set is it may mean that granting a member of the tracing team CI access to test changes would be a mild or moderate impact rather than a significant one.

@christophebedard what do you think a reasonable default CI run is for the ros2_tracing?

christophebedard, 2020-07-15:
> One concern of mine is CI resource availability. Tracing is a low-level component in ROS 2 builds which means that it has a large surface area for potential issues. But if I understand correctly, [...]

That has been true so far. `tracetools`'s main (and only, really) API, the `TRACEPOINT()` macro, which is used in `rcl` & `rclcpp`) expands to nothing if LTTng isn't installed, or if it is compiled with `-DTRACETOOLS_DISABLED=ON`. Same for `tracetools_test`: no tracing tests are run without LTTng.

However, we've been planning to explicitly declare a dependency on LTTng for some time now (see [this issue](https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/issues/71)). It was brought up a couple times ([here](https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/merge_requests/139) and [here](https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/issues/94)). This would allow actually using the feature (i.e. tracing) out of the box, instead of having to re-compile the core, and would have basically no performance impact while still allowing users to completely remove any tracing stuff by re-compiling the core.

Just as a note, userspace tracing is very different compared to kernel tracing. The former is definitely intented to be used by users (and is used for running test in `tracetools_test`). The latter isn't needed for building or testing; it's simply supported by the processing/analysis tools, but those aren't really part of the core. Userspace tracing is easy to setup and use in CI, while kernel tracing definitely requires more effort/setup/permissions, since it needs to be installed on the host. We haven't had any issues with userspace tracing itself on our CI, but then again our setup is very different.

That being said, I'm not sure that we should really combine these two issues/requests together: access to CI to run on all supported platforms, and shipping with LTTng & running tracing tests by default. I simply wanted to mention our future plans since they are relevant here. We could consider & talk about the second one separately, after the first one. Let me know how you'd like to proceed.

> what do you think a reasonable default CI run is for the ros2_tracing?

The only packages that depend on `tracetools` are `rcl` and `rclcpp`. `tracetools_test` contains tests for `tracetools`, but they are only *currently* configured/enabled if LTTng is installed, as mentioned above. The other packages probably don't require any special care.

nuclearsandwich, 2020-07-24:
Thanks @christophebedard with all of the above in mind I'll propose the following:

* Let's onboard @christophebedard with permissions to build on ci.ros2.org
* With that we'll expect tracetools/ros2_tracing changes will be run through ci.ros2.org before being merged so that we can have as much confidence in them as we do with other changes in ROS 2 core.
* We approve #958 and merge it once we've got the CI setup covered.

@christophebedard I'll be in contact with you via email to get you onboarded to our CI.

christophebedard, 2020-07-27:
Thanks @jacobperron and @nuclearsandwich.

> * With that we'll expect tracetools/ros2_tracing changes will be run through ci.ros2.org before being merged so that we can have as much confidence in them as we do with other changes in ROS 2 core.

Will do!

> I've recently given "developer" access (GitLab permission) to some Open Robotics people. This allows those people to push to non-protected branches and review merge requests. By changing a setting [...]

Regarding this, did you want us to give write access to OR people, or are you simply leaving this decision up to us? I think it would be a no-brainer for us, with the "implicit" understanding that it is to be used for fixes when Ingo and I aren't available so that you don't have to fork the repo to get a fix in.

nuclearsandwich, 2020-07-27:
> Regarding this, did you want us to give write access to OR people, or are you simply leaving this decision up to us?

I don't feel like this is necessary. If we have any issues with mainline we can always pin back to a working commit. I think it's reasonable to treat contributors from Open Robotics as any other contributors, granting merge rights when it makes sense for the tracing project if a contributor has a history of contributions.

christophebedard, 2020-07-27:
That does make sense. Thanks for explaining.

nuclearsandwich, 2020-07-29:
Thanks for starting the discussion @christophebedard. With #958 merged I think this issue is resolved.

closed 2020-07-29 by nuclearsandwich