ros2/ros2 #959: When attempting to build from source I run into: [ModuleNotFoundError: No module named 'importlib.resources']

Jrokisky, 2020-06-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04.4 LTS
- Installation type:
  - Attempting to build from source using the instructions found [here](https://index.ros.org/doc/ros2/Installation/Rolling/Linux-Development-Setup/#linux-latest)
- Version or commit hash:
  - Current (June 28) master (0ee1ca223bcc7185743da1ecccc754b774cd0958)
- DDS implementation:
  - default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
1. Follow the [build from source instructions](https://index.ros.org/doc/ros2/Installation/Rolling/Linux-Development-Setup/#linux-latest) on an Ubuntu 18.04.4 machine

2. When it's time to build the source, run: `colcon build --symlink-install`

#### Expected behavior
ROS2 is built.

#### Actual behavior
```
--- stderr: ament_cmake_core                                       
Traceback (most recent call last):
  File "/home/justin/ros2_rolling/build/ament_package/ament_package/templates.py", line 19, in <module>
    import importlib.resources as importlib_resources
ModuleNotFoundError: No module named 'importlib.resources'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/justin/ros2_rolling/src/ament/ament_cmake/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
  File "/home/justin/ros2_rolling/build/ament_package/ament_package/templates.py", line 21, in <module>
    import importlib_resources
ModuleNotFoundError: No module named 'importlib_resources'
CMake Error at ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3
  /home/justin/ros2_rolling/src/ament/ament_cmake/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/justin/ros2_rolling/build/ament_cmake_core/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  CMakeLists.txt:19 (include)

---
Failed   <<< ament_cmake_core [0.41s, exited with code 1]
Aborted  <<< ament_flake8 [0.41s]
Aborted  <<< test_osrf_testing_tools_cpp [0.49s]
Aborted  <<< ament_pycodestyle [1.06s]                        
                                   
Summary: 10 packages finished [4.88s]
  1 package failed: ament_cmake_core
  3 packages aborted: ament_flake8 ament_pycodestyle test_osrf_testing_tools_cpp
  1 package had stderr output: ament_cmake_core
  291 packages not processed
```

Jrokisky, 2020-06-29:
Possibly linked to https://github.com/ros2/ros2/issues/919?

After looking at https://github.com/ros2/ci/pull/483/files, running `pip3 install -U importlib-resources` solved the issue.

Still new to ROS's dependency management system, does this package need to be added as a dependency to ament_cmake_core?

fujitatomoya, 2020-06-29:
i do not have any problems.

- Environment
  - ubuntu18.04 LTS
  - Source build (https://github.com/ros2/ros2/commit/0ee1ca223bcc7185743da1ecccc754b774cd0958)

- Build log

```
# pip3 list | grep import
flake8-import-order        0.18.1
importlib-metadata         1.5.0
# colcon build --symlink-install --packages-select ament_cmake_core
Starting >>> ament_cmake_core
Finished <<< ament_cmake_core [0.14s]

Summary: 1 package finished [0.71s]
```

clalancette, 2020-06-29:
Yes, as of last week, you do need to install importlib-resources on Ubuntu 18.04 when building master.

It hasn't been a huge rush right now, since we don't officially support Ubuntu 18.04 with ROS 2 anymore (see https://index.ros.org/doc/ros2/Releases/Release-Rolling-Ridley/ for supported platforms).  But since we are keeping it working best-effort, I will submit a PR to the documentation to add this.