ros2/ros2 #960: [meta] Mocking infrastructure for C APIs

hidmic, 2020-06-29:
## Description

To boost test coverage for C APIs, integrate [Mimick](https://github.com/Snaipe/Mimick) C mocking framework into our testing pipeline.

## Rationale 

According to the [ROS 2 Developer Guide](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#code-coverage), core ROS packages must reach 95% coverage. But there's so much black-box and white-box unit tests, along with integration and end-to-end tests, can cover without mocking.

Unfortunately, neither client library APIs nor `rcl`'s or `rmw`'s, to name a few, lend themselves to dependency injection. Many open-source C/C++ mocking frameworks are off the table. We must resort to less explicit techniques, such as linker and/or dynamic linker based mocking. But again, no perfect solution was found (open-source, cross-platform, C11 / C++14 mocking framework with little to no code generation).

In an attempt to cover as much ground as possible, we'll try [Mimick](https://github.com/Snaipe/Mimick): a C-only mocking framework that tinkers with dynamic linker structures in run-time.

## Tasks
- [ ] Verify [Mimick](https://github.com/Snaipe/Mimick) works on every Tier 1 platform per [REP-2000](https://www.ros.org/reps/rep-2000.html).
- [ ] Vendor [Mimick](https://github.com/Snaipe/Mimick) as an `ament_cmake` package.
- [ ] Use [Mimick](https://github.com/Snaipe/Mimick) in an `rcutils` gtest for a proof of concept.

hidmic, 2020-06-29:
I can confirm `Mimick` builds and tests pass on:

- Ubuntu Focal 20.04 `amd64` using `gcc 9.3.0
- Windows Server 2019 `amd64` using `msvc` 19.26
- MacOS Mojave 10.14.5 `amd64` using  `clang` 10.0

hidmic, 2020-06-30:
Well, `Mimick` builds but fails the tests on:

* Ubuntu Focal 20.04 `aarch64` using `gcc` 9.3.0

---

I've been digging to figure out why. Tests fail because `Mimick` fails to inject the mock in the right place. If we take a look at its [ELF support](https://github.com/Snaipe/Mimick/blob/7681d1ee8e0bffff806751092947648a6bebff24/src/plt-elf.c#L227), we'll see it first checks for the function symbol in the `.rela.dyn` section (i.e. GOT) and only goes look for it in the `.rela.plt` section (i.e. PLT's GOT) if it fails the first lookup. In `aarch64`, the function symbol shows up in both tables, even though trampolines in code only seem to be paying attention to the latter table (as I would expect).

Even though I failed to find a concise explanation out there, my recollection is that, in the most general case, a dynamic ELF executable needs both an `R_<ARCH>_JUMP_SLOT` relocation in the `.rela.plt` section for lazy binding of an external callable symbol and an `R_<ARCH>_GLOB_DAT` relocation in the `.rela.dyn` section when/if the address of an external callable symbol is requested (i.e. the function pointer). And while some linkers may perform an optimization and deduplicate this information, they are not bound to it. 

If this is true, the fix for `Mimick` is simple. But I'd like to confirm this is the case before changing anything. @clalancette @wjwwood perhaps?

wjwwood, 2020-06-30:
That sounds like a reasonable theory about the issue @hidmic but it’s been a long time since I have looked at the ELF layout so I don’t remember any particulars.

hidmic, 2020-07-01:
Alright, got `Mimick` working on `aarch64`. I found a few other bugs in ELF support and elsewhere too.
I opened up a PR upstream from [my fork](https://github.com/hidmic/Mimick). In the meantime, we can rely on this [ros2](https://github.com/hidmic/Mimick/tree/ros2) branch.

hidmic, 2020-07-07:
We have a problem. Turns out `Mimick` uses a bunch of C99 features in macros that never made it C++ (like compound literals or designated initializers). That means we can't use Mimick in a regular `gtest`. Not as-is.

hidmic, 2020-07-08:
Alright, @Blast545 and I finally managed to get `Mimick` working in `gtest` context. [This branch](https://github.com/hidmic/Mimick/tree/hidmic/cxx-compatibility), currently PR'd upstream, includes CMake bits and user facing API fixes to get it to work.

hidmic, 2020-09-03:
@chapulina I think we can call this done.

chapulina, 2020-09-03:
Yup, as far as I'm concerned, everything is already in place for mocking. :+1:

closed 2020-09-03 by hidmic