ros2/ros2 #963: [Meta-ticket]: Synchronize rosdistro source branches with gbp release devel branches

clalancette, 2020-07-08:
This is a meta-ticket to synchronize rosdistro source branches with the gbp release devel branches for particular repositories.

For instance, as of the time of this PR, the Dashing rosdistro source entry for `ament_index` shows "dashing" as the [source branch](https://github.com/ros/rosdistro/blob/8d9d7cfa6ef2fddac296be705e7ad61554fa41b0/dashing/distribution.yaml#L99).  However, the ros2-gbp repository for `ament_index` shows the [master branch](https://github.com/ros2-gbp/ament_index-release/blob/d9afc52deadacc5c4f97016f35c603b5edd7f76e/tracks.yaml#L88) as the devel_branch for the dashing release in the gbp release repository.

The discontinuity between the rosdistro source branch and gbp devel_branch is more than a cosmetic problem.  When going to do patch releases for packages where the gbp devel_branch is wrong, `bloom-release` will pick the wrong release to release into the distribution.  This requires the human to notice this and take appropriate action (usually modifying the gbp release repo) to rectify.

This problem comes about due to the way we do releases.  In particular, during the release candidate period of a release we release all packages to a ROS distribution from the default (`master`, `ros2`, etc) branch.  Right before release, however, we fork all of the source repositories so that they have a branch for that release.  This allows us to do patch releases in the future.

To rectify this problem, there is a new script: https://gist.github.com/clalancette/8bf71d186ac1463b5c35f73a45ceffde .  The script looks at the rosdistro `distributions.yaml` file for all core ros2 packages, and opens PRs against the corresponding gbp release repos to synchronize the source branch in the former with the devel_branch in the latter.  I'll eventually move that script to a more permanent location.

Since we don't usually receive notifications of PRs against gbp repositories, this meta-ticket is a landing spot for all of those PRs to reference.  That way we can have one place to look to merge them all.

clalancette, 2020-07-20:
All right, this is done for now, so I'll close this issue out.

closed 2020-07-20 by clalancette

dirk-thomas, 2020-07-20:
Shouldn't the script be added to a ROS repository (maybe `ros/rosdistro`) to be run again in the future?

clalancette, 2020-07-23:
See https://github.com/ros/rosdistro/pull/25908