ros2/ros2 #979: ros-foxy-ros-base package is missing for ARM architecture

Leonti, 2020-07-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - http://packages.ros.org/ros2/ubuntu/dists/focal/main/binary-armhf/Packages
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
sudo apt update && sudo apt install curl gnupg2 lsb-release
curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
sudo sh -c 'echo "deb [arch=$(dpkg --print-architecture)] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" > /etc/apt/sources.list.d/ros2-latest.list'
sudo apt update
sudo apt install ros-foxy-ros-base
```

#### Expected behaviour
Should be able to successfully install `ros-foxy-ros-base` package on ARM (Raspberry Pi)
#### Actual behavior
Receiving
```
E: Unable to locate package ros-foxy-ros-base
```
error when trying to install the package
#### Additional information

I was trying to install `ros-foxy-ros-base` on Raspberry Pi following the installation tutorial:
https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Debians/
but it seems like `ros-foxy-ros-base` package is missing from the repository for `armhf` architecture  
Checking it manually here: http://packages.ros.org/ros2/ubuntu/dists/focal/main/binary-armhf/Packages also confirms that it's not there.  
Am I not following the right steps? Is it a temporary issue?

gabrik, 2020-07-16:
Hi everyone, same here, I was trying to install ROS2 Foxy on a RaspberryPi3 with Ubuntu arm64 and the package is missing.

Here the result of `uname -a` 
```
Linux nibbler 4.15.0-1065-raspi2 #69-Ubuntu SMP PREEMPT Thu Jun 25 15:17:08 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
```

And of `cat /etc/os-release`

```
NAME="Ubuntu"
VERSION="18.04.4 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.4 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

Leonti, 2020-07-16:
@gabrik I think only `eloquent` is supported on Ubuntu 18.04:
http://packages.ros.org/ros2/ubuntu/dists/bionic/main/binary-armhf/Packages
You'll find `ros-eloquent-ros-base` there

Ubuntu 20.04 is supposed to have `ros-foxy-ros-base`.
It's there for other architectures, for example: http://packages.ros.org/ros2/ubuntu/dists/focal/main/binary-amd64/Packages, but not for `armhf` :(

gabrik, 2020-07-16:
@Leonti I see.

It's there also for `arm64`: http://packages.ros.org/ros2/ubuntu/dists/focal/main/binary-arm64/Packages
So I'm going to bake another SD for the RPi with Ubuntu 20.04.

I thought as 18.04 is an LTS that they were going to support it.
But it is not a big deal to bake a new SD card

gavanderhoorn, 2020-07-16:
@gabrik: in these cases REP 2000 (*ROS 2 Releases and Target Platforms*) should tell you whether binaries will be built for a specific OS. See [Platforms by Distribution: Foxy Fitzroy (May 2020 - May 2023)](https://www.ros.org/reps/rep-2000.html#foxy-fitzroy-may-2020-may-2023) for the Foxy section.

clalancette, 2020-07-16:
There are actually two separate issues being discussed here.

The original comment from @Leonti is trying to install Foxy on 20.04, but is trying to install on armhf.  As @gavanderhoorn pointed out, REP 2000 describes armhf for Foxy as Tier 3, meaning it is best-effort support.  That basically means users have to build it from source (no binary packages), and we won't hold up releases for bugs that only affect the architecture.

Separately, @gabrik is trying to install Foxy on 18.04, and is trying to install on arm64.  We only build binary packages for Foxy for Ubuntu 20.04, which is why there are no packages there.  As it stands right now, it is possible to build and run Foxy from source on 18.04, but we don't guarantee that capability.  So there is some chance it will break in the future.

The options to deal with this are:
1.  Upgrade to a supported architecture and distribution (Ubuntu 20.04 on arm64 in this case).  That is possible on a Raspberry Pi, but you do need a new image.
1.  Use an older version of ROS 2.  As mentioned previously, Eloquent works on Ubuntu 18.04.
1.  Build from source on your chosen platform.  This should work for both use-cases, but will be slow.

In any case, I don't think there is an issue here, so I'll close this out.  In the future, please ask these kinds of questions on answers.ros.org so we can have all of the questions and answers in one searchable place.

closed 2020-07-16 by clalancette

Leonti, 2020-07-16:
Thanks for the clarification @clalancette!
I thought it was a genuine bug (maybe a build step for `armhf` was failing) hence submitted it here and not on answers.ros.org

janblumenkamp, 2020-11-10:
Dear @clalancette,

could you elaborate why exactly armhf for Foxy is described as Tier 3? Armhf includes Raspberry Pi Zero, which arguably is an important potential platform for ROS 2, especially given its decentralized nature. The Raspberry Pi Zero is perfect for equipping large fleets of small robots with ROS 2. We want to use Raspberry Pi Zeros, but currently have to cross compile, which can be a quite cumbersome process.

Thanks for clarifying,
Jan

clalancette, 2020-11-10:
> could you elaborate why exactly armhf for Foxy is described as Tier 3?

Essentially, we ran into serious infrastructure issues when trying to get it building on the buildfarm.  There is/was a very bad bug in Ubuntu 20.04 which meant we could not run our normal Docker builds.  We ran out of time to work on it, and nobody else stepped up to try to fix it, so we were forced to drop it.  You can see some of the details at https://github.com/osrf/multiarch-docker-image-generation/issues/38#issuecomment-644817880 (also, there is a bug in Qemu 4.2 that also needs to be worked around, which we did for Noetic: https://github.com/osrf/multiarch-docker-image-generation/issues/36)