ros2/ros2 #981: Unable to bulid

chindimaga, 2020-07-20:
## Bug report

- Operating System:Ubuntu 18.04

- Installation type:https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/
I was following the documentation to install
when I execute the command 
  colcon build --symlink-install               
I get this error
**********************************************************************************************************
(base)  ✘ rahul@rahul-Inspiron-5570  ~/ros2_dashing  colcon build --symlink-install               
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> osrf_testing_tools_cpp
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> cyclonedds
Starting >>> ament_cppcheck
Starting >>> tinydir_vendor
Finished <<< poco_vendor [3.10s]                              
Starting >>> gtest_vendor
Finished <<< ament_package [4.35s]                            
Starting >>> ament_cmake_core
Finished <<< gtest_vendor [1.33s]                          
Finished <<< ament_cppcheck [4.52s]
Starting >>> osrf_pycommon
Starting >>> gmock_vendor                                  
Finished <<< ament_lint [4.65s]                            
Starting >>> ament_flake8
--- stderr: ament_cmake_core           
Traceback (most recent call last):
  File "/home/rahul/ros2_dashing/src/ament/ament_cmake/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 21, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/home/rahul/miniconda3/bin/python3
  /home/rahul/ros2_dashing/src/ament/ament_cmake/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/rahul/ros2_dashing/src/ament/ament_cmake/ament_cmake_core/package.xml
  /home/rahul/ros2_dashing/build/ament_cmake_core/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  CMakeLists.txt:15 (ament_package_xml)

---
Failed   <<< ament_cmake_core [2.46s, exited with code 1]
Aborted  <<< fastcdr [6.85s]
Aborted  <<< gmock_vendor [2.25s]
Aborted  <<< ament_flake8 [2.18s]
Aborted  <<< osrf_pycommon [2.30s]
Aborted  <<< tinydir_vendor [9.71s]                 
Aborted  <<< cyclonedds [10.6s]                     
Aborted  <<< osrf_testing_tools_cpp [11.1s]                                    

Summary: 5 packages finished [12.5s]
  1 package failed: ament_cmake_core
  7 packages aborted: ament_flake8 cyclonedds fastcdr gmock_vendor osrf_pycommon osrf_testing_tools_cpp tinydir_vendor
  1 package had stderr output: ament_cmake_core
  258 packages not processed
*************************************************************************************************************
Can somebody please help me?

fujitatomoya, 2020-07-20:
it tells that `catkin_pkg` is not installed, could you try `sudo apt install python3-catkin-pkg`, it would resolve this problem. also there is a thread related to this issue, https://answers.ros.org/question/337135/catkin_make-no-module-named-catkin_pkg

clalancette, 2020-07-20:
> it tells that `catkin_pkg` is not installed, could you try `sudo apt install python3-catkin-pkg`, it would resolve this problem. also there is a thread related to this issue, https://answers.ros.org/question/337135/catkin_make-no-module-named-catkin_pkg

While this will fix it, that should have been done automatically.

@chindimaga Did you follow the instructions to install dependencies via `rosdep` [here](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/#id7)?

chindimaga, 2020-07-20:
> it tells that `catkin_pkg` is not installed, could you try `sudo apt install python3-catkin-pkg`, it would resolve this problem. also there is a thread related to this issue, https://answers.ros.org/question/337135/catkin_make-no-module-named-catkin_pkg

Thanks. That issue got resolved.I created a new virtual environment and redid the installation. But one more issue popped up.
***********************************************************************************************************
Finished <<< launch_testing_ament_cmake [1.19s]
Starting >>> rcutils                     
--- stderr: rosidl_generator_c
CMake Error at /home/rahul/ros2_dashing/install/rosidl_adapter/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
  execute_process(/home/rahul/ros/bin/python3 -m rosidl_adapter
  --package-name rosidl_generator_c --arguments-file
  /home/rahul/ros2_dashing/build/rosidl_generator_c/rosidl_adapter__arguments__rosidl_generator_c_interfaces.json
  --output-dir
  /home/rahul/ros2_dashing/build/rosidl_generator_c/rosidl_adapter/rosidl_generator_c
  --output-file
  /home/rahul/ros2_dashing/build/rosidl_generator_c/rosidl_adapter/rosidl_generator_c_interfaces.idls)
  returned error code 1:

  AttributeError processing template 'msg.idl.em'

  Traceback (most recent call last):

    File "/home/rahul/ros2_dashing/install/rosidl_adapter/lib/python3.7/site-packages/rosidl_adapter/resource/__init__.py", line 48, in evaluate_template
      _interpreter = em.Interpreter(

  AttributeError: module 'em' has no attribute 'Interpreter'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "/home/rahul/miniconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/home/rahul/miniconda3/lib/python3.7/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/home/rahul/ros2_dashing/install/rosidl_adapter/lib/python3.7/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/home/rahul/ros2_dashing/install/rosidl_adapter/lib/python3.7/site-packages/rosidl_adapter/main.py", line 55, in main
      pathlib.Path(relative_path), output_dir)
    File "/home/rahul/ros2_dashing/install/rosidl_adapter/lib/python3.7/site-packages/rosidl_adapter/__init__.py", line 20, in convert_to_idl
      package_dir, package_name, interface_file, output_dir / 'msg')
    File "/home/rahul/ros2_dashing/install/rosidl_adapter/lib/python3.7/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file)
    File "/home/rahul/ros2_dashing/install/rosidl_adapter/lib/python3.7/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
    File "/home/rahul/ros2_dashing/install/rosidl_adapter/lib/python3.7/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /home/rahul/ros2_dashing/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:106 (rosidl_generate_interfaces)

---
Failed   <<< rosidl_generator_c [0.94s, exited with code 1]
Aborted  <<< rcutils [0.02s]
Aborted  <<< rosidl_generator_dds_idl [0.52s]
Aborted  <<< tlsf [2.39s]                           
Aborted  <<< urdf [2.11s]                           
Aborted  <<< class_loader [2.85s]                   
Aborted  <<< ament_index_cpp [4.93s]                
Aborted  <<< rviz_ogre_vendor [8.37s]   
*************************************************************************************************
I tried searching in stack overflow. but I could not find a solution.

clalancette, 2020-07-20:
> Yes. I had followed and installed all the dependencies. Turns out Even after resolving it I was unable to continue the installation.

It looks like you are probably using a Conda workspace, or a Python venv.  I think this is the case because the above errors are trying to using Python 3.7, but Ubuntu 18.04 (which Dashing was released for) uses Python 3.6.  I'll suggest you exit the venv or Conda workspace before trying again.  Note that you'll likely have to go back through all of the setup instructions to make this work.

clalancette, 2020-08-06:
I think the problem here is in environment settings.  I'm going to go ahead and close this.  If you are still having problems, please consider opening a new question on https://answers.ros.org .

closed 2020-08-06 by clalancette

FrankJIE09, 2022-01-01:
> it tells that `catkin_pkg` is not installed, could you try `sudo apt install python3-catkin-pkg`, it would resolve this problem. also there is a thread related to this issue, https://answers.ros.org/question/337135/catkin_make-no-module-named-catkin_pkg

I try `sudo apt install python3-catkin-pkg`.But it does not solve the problem.I gedit my ~/.bashrc .Add  like this one :**export PYTHONPATH=$PYTHONPATH:/home/cherry/anaconda3/envs/Franka/lib/python3.7/site-packages**. Reopen the terminal and remake. It works.

124451, 2022-06-08:
This link solves my problem:https://answers.ros.org/question/340510/catkin_make-no-module-named-catkin_pkg/

Feedback02, 2022-11-19:
if sudo apt install python3-catkin-pkg doesn't work:
i solved this problem by unistalling mamba (or conda if you use conda), and than reinstalling ROS2. probably it will work if you type "conda deactivate" and then reinstalling ROS2 again.i also (before installing ROS2) deleted/comment in bashrc the conda settings/source.

brianlmerritt, 2023-03-21:
FYI I had issues with ROS 2 humble on Ubuntu 20.04 because I had a build of python3.11 and the default python was 3.9.

If you have a look at dusty-containers, he deletes Python 3.9 because the right packages are actually install for python3.8.

The "catkin-pkg" problem is best fixed (yes, in my opinion) by using pip to install the packages, not using the default linux builds.

So python3.11 -m pip install catkin-pkg installs the right package for the python version I am using.  Check your python version and use that instead of python3.11 of course.