ros2/ros2 #984: No module named 'launch.actions'

ItsMeTheBee, 2020-07-20:
## Bug report

**Required Info:**

- Operating System:
  - ros foxy docker container running on Ubuntu 18 
- Installation type:
  - from docker hub

#### Steps to reproduce issue
- Start ros:foxy docker container
- Clone & Build vision_opencv like [this](https://index.ros.org/p/cv_bridge/github-ros-perception-vision_opencv/)
- run colcon test

#### Expected behavior
Tests complete

#### Actual behavior
```
--- stderr: opencv_tests                    
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3/dist-packages/pytest.py", line 102, in <module>
    raise SystemExit(pytest.main())
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 63, in main
    config = _prepareconfig(args, plugins)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 206, in _prepareconfig
    return pluginmanager.hook.pytest_cmdline_parse(
  File "/usr/lib/python3/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/lib/python3/dist-packages/pluggy/manager.py", line 92, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/lib/python3/dist-packages/pluggy/manager.py", line 83, in <lambda>
    self._inner_hookexec = lambda hook, methods, kwargs: hook.multicall(
  File "/usr/lib/python3/dist-packages/pluggy/callers.py", line 203, in _multicall
    gen.send(outcome)
  File "/usr/lib/python3/dist-packages/_pytest/helpconfig.py", line 94, in pytest_cmdline_parse
    config = outcome.get_result()
  File "/usr/lib/python3/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/lib/python3/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 739, in pytest_cmdline_parse
    self.parse(args)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 947, in parse
    self._preparse(args, addopts=addopts)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 893, in _preparse
    self.pluginmanager.load_setuptools_entrypoints("pytest11")
  File "/usr/lib/python3/dist-packages/pluggy/manager.py", line 297, in load_setuptools_entrypoints
    plugin = ep.load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_testing/__init__.py", line 15, in <module>
    from . import tools
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_testing/tools/__init__.py", line 18, in <module>
    from .process import launch_process
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_testing/tools/process.py", line 18, in <module>
    import launch.actions
ModuleNotFoundError: No module named 'launch.actions'
```

#### Additional information

- works fine in ros eloquent container

dirk-thomas, 2020-08-06:
The invocation fails because the `opencv_tests` package has a `launch` module (https://github.com/ros-perception/vision_opencv/tree/105da5391ea00e261d7ea6ef90aac408877d475f/opencv_tests/launch) which prevent the same module from the `launch` package (which does contain `actions`) to be found.

Please see ros-perception/vision_opencv#352 for a proposed patch.

dirk-thomas, 2020-08-27:
Since there was no further response I assume the referenced patch solved the problem and will go ahead and close this ticket.

closed 2020-08-27 by dirk-thomas