ros2/ros2 #986: Dashing Patch Release 8

nuclearsandwich, 2020-07-22:
This issue will track the broad status of the next Dashing patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

We're going to use a GitHub Project board for this patch release again: [Dashing Patch Release 8](https://github.com/orgs/ros2/projects/24)

As with prior releases, if you want to contribute fixes please open pull requests targeting the latest development branch and mention to the reviewer that you'd like the pull request to be considered for backporting into Dashing.

nuclearsandwich, 2020-11-20:
@jwhitleywork are you interested in seeing https://github.com/ros2/rmw_cyclonedds/pull/96 backported to Dashing? If so would you be willing to open the backport PR and make the release before Tuesday 2020-11-24? It can always go in a future patch release if you don't have the time right now.

JWhitleyWork, 2020-11-20:
@nuclearsandwich I'm not familiar with the back-story on this change. @eboasson Can you give me a primer? Is this something that could be easily ported to the `dashing-eloquent` branch and do we even want to?

eboasson, 2020-11-23:
@nuclearsandwich, @JWhitleyWork: that's already in dashing and eloquent (the dashing-eloquent branch has the commit). I'm pretty sure the `dashing-eloquent` branch was created more recently than that particular fix, and indeed the commit id is the same as in master.

JWhitleyWork, 2020-11-23:
@eboasson That's kind of what I figured. @nuclearsandwich, you should be fine to go ahead.

nuclearsandwich, 2020-11-24:
Great. Sorry for the noise. That PR was still on the Dashing patch release board so I assumed it was outstanding. I've removed it since that change appears to have been released in 0.5.1.

nuclearsandwich, 2020-12-02:
Packaging jobs:

Ubuntu AMD64 [image]
Ubuntu ARM64 [image]
Ubuntu ARMHF [image]
CentOS AMD64 [image]
Windows AMD64 [image]
Windows Debug AMD64 [image]

nuclearsandwich, 2020-12-03:
The sync and patch release is [out](https://discourse.ros.org/t/new-packages-and-patch-release-for-ros-2-dashing-diademata-2020-12-02/17719)!

PR #1064 (2020-12-02): Dashing Patch Release 8
Version updates for dashing patch release 8.

Closes #986.
closed 2020-12-03 by nuclearsandwich