ros2/ros2 #988: QNX Support

asobhy-qnx, 2020-07-23:
This issue will track patches related to building ROS2 for QNX 7.1 real-time operating system.

The patches will allow users to cross-compile ROS2 on their host system, PC, according to one of the supported architectures below.

**QNX supported architectures as of version 7.1:**
- x86_64
- armv8 (aarch64)
- armv7

asobhy-qnx, 2020-08-25:
is there a way to control the value for COLCON_CURRENT_PREFIX that will be placed in the setup and local setup scripts?

Currently I am patching the scripts with the value of the new path on target after the build is done, but I was thinking that there might be a variable I could set before building that could influence the value placed in the env sourcing scripts.

I already set the value TARGET_INSTALL_DIR but it has no effect on the value being placed inside those scripts.

asobhy-qnx, 2020-08-28:
For bullet dependency are python bindings required?

iluetkeb, 2020-09-17:
I'm curious why you're not using CMAKE_SYSROOT? It should address some of things automatically, particularly the rpath modifications.

asobhy-qnx, 2020-09-17:
CMAKE_SYSROOT expects a sysroot filesystem tree for a single architecture. Since QNX supports multiple archs our sysroot does not contain the usr/include directory since it is common between the three architectures.

iluetkeb, 2020-09-17:
Ah, interesting to hear. Personally, it would seem easier to me to copy over the /usr/include than to fiddle with rpaths, but that's your choice, of course :-)

asobhy-qnx, 2020-09-17:
doing that will break a lot of other qnx stuff that is based on this filesystem tree and will take a lot more work to change. Adding an rpath-link option only provides the location of the libraries during link time. All of the cross-compiled libraries are built without an embedded rpath.

clalancette, 2020-10-08:
@asobhy-qnx What's the status of this ticket?  Is there additional work that needs to be done here, or can we close this out?

closed 2020-10-08 by asobhy-qnx

tongtongdi, 2022-02-07:
Does ros2 support qnx7.0?