ros2/ros2 #989: zero time stamp for static_transform_publisher

Tahir-Rasheed-437, 2020-07-27:
I have recently started using  foxy. I am not  sure if it is really an issue or i am doing something wrong. Its about static transform publisher through a launch file. In my launch file, this is how i am generating my static transform between base_link and laser_0:

use_sim_time = LaunchConfiguration('use_sim_time', default='true')
return LaunchDescription([
        Node( 
            package='tf2_ros', 
            node_executable='static_transform_publisher',
            node_name='laser_0',
            arguments=['0.5397', '0.4037', '0.0', '0.920504942522', '0.390730918652', '5.18419788842e-07', '1.22132126011e-06', 'base_link', 'laser_0'],
            parameters=[{'use_sim_time': use_sim_time}]),  
])

If i echo tf_static, its output is:

transforms:
- header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: base_link
  child_frame_id: laser_0
  transform:
    translation:
      x: 0.5397
      y: 0.4037
      z: 0.0
    rotation:
      x: 0.920504942522
      y: 0.390730918652
      z: 5.18419788842e-07
      w: 1.22132126011e-06

the problem is the time stamp sec and nanosec should correspond to the clock of my simulation time but its not.

The output  of ros2 run tf2_ros tf2_echo  base_link laser_0 is:
At time 0.0
- Translation: [0.540, 0.404, 0.000]
- Rotation: in Quaternion [0.921, 0.391, 0.000, 0.000]

Am i missing something from my launch file? 
Secondly, in foxy, ros2 topic echo /tf_static does not always print the above output. Sometimes its just waiting without any output. Or sometimes it just print once and then later nothing.  On the contrary, it was continuously printing tf_static for dashing. Is it normal in foxy now?

allenh1, 2020-07-27:
The static transform publisher is different in Foxy, but I don't think that is causing your lack of timestamps.

https://github.com/ros2/geometry2/blob/7b0187f57d1d4f845601a774b609660af1dc92ba/tf2_ros/src/static_transform_broadcaster_node.cpp#L74

What is the `use_sim_time` value?

Tahir-Rasheed-437, 2020-07-27:
Yes you are right, my issue is somewhere else. I have set it to true in the second line of the launch file:

`use_sim_time = LaunchConfiguration('use_sim_time', default='true')`

Do i have to declare somewhere else as well?

allenh1, 2020-07-27:
> UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
UserWarning: The parameter 'node_name' is deprecated, use 'name' instead

This is just a deprecation warning. `node_executable` is now `executable`, and `node_name` is now `name`.

> i have just used by source the static_transform_broadcaster_node (the one you shared). time stamps are still zero. 

Do you have a `/clock` publisher? This is pasted for convenience (found [here](https://wiki.ros.org/Clock))

> If the /use_sim_time parameter is set, the ROS Time API will return time=0 until it has received a value from the /clock topic. Then, the time will only be updated on receipt of a message from the /clock topic, and will stay constant between updates.

While I can't find any documentation that confirms this is the behavior for ROS 2, I can't imagine it would be otherwise.

Since the message is created in the constructor, `rclcpp::spin` has not yet been invoked, so I would expect it to be set to `0`.

@Tahir-Rasheed-437 Since this is a static transform, I don't think this should be breaking anything, but let me know if that's incorrect.

@clalancette If my diagnosis was accurate, should there be a check on the `use_sim_time` parameter in the constructor that will delay until `this->now()` is not `0` before publishing the transform? I would argue that it's probably not worth spending cycles on since static transforms shouldn't change, but this does seem like odd behavior.

Tahir-Rasheed-437, 2020-07-29:
@allenh1 yes i do have a clock publisher.  My actual launch file looks something like this:

```
def generate_launch_description():
    use_sim_time = LaunchConfiguration('use_sim_time', default='True')

    return LaunchDescription([
        
        Node( 
            package='tf_publishser',
            node_executable='tf_publishser',
            node_name='tf_publishser1',
            parameters=[{'use_sim_time': use_sim_time}]),

        Node( 
            package='tf2_ros', 
            node_executable='static_transform_publisher',
            node_name='laser',
            arguments=['0.5397', '0.4037', '0.0', '0.920504942522', '0.390730918652', '5.18419788842e-07', '1.22132126011e-06', 'base_link', 'laser_0'],
            parameters=[{'use_sim_time': use_sim_time}]),  
    ])
```

My tf publisher is working perfectly fine and generating a tf with correct time stamp. only static tf's having issue. Currently i am having issues in navigation2 (https://github.com/ros-planning/navigation2/issues/1891) and the error i believes have something to do with tf's.

I have another query. Compare to dashing where the transforms in tf_static were published continuously, its not the case in foxy? After launching the above launch file, the tf_static is only published once and that's all. Is it the right behavior?

clalancette, 2020-08-03:
> @clalancette If my diagnosis was accurate, should there be a check on the `use_sim_time` parameter in the constructor that will delay until `this->now()` is not `0` before publishing the transform? [...]

Yeah, I agree that it is odd, but it isn't actually incorrect, in my view.  Since it has been declared a static transform, it shouldn't change and thus the timestamp isn't really all that interesting.

> I have another query. Compare to dashing where the transforms in tf_static were published continuously, its not the case in foxy? After launching the above launch file, the tf_static is only published once and that's all. Is it the right behavior?

That's correct.  In Eloquent we changed it so that `tf_static` is now a "transient_local" publication.  If you are familiar with ROS 1, this is similar to the "latched" topic there.  To get the previous state of `tf_static`, you need to subscribe to `/tf_static` as "transient_local" as well.  I'll note that the APIs internal to `tf2` already do this automatically, but if you are subscribing to it "by hand" or with the command-line tools, you'll have to be careful to make sure they are transient_local as well.

Tahir-Rasheed-437, 2020-08-07:
OK thank you very much for your replies. I still have this issue of wero time stamps but yes i guess time stamps are really not an issue in static tf's. On the other hand, simple tf's have correct time stamps and working fine. Thanks for the explaination on the transient_local. We can close this issue now if its okay.

clalancette, 2020-08-09:
All right, closing this out.

closed 2020-08-09 by clalancette