ros2/ros2 #990: ros2 bag record (foxy version) not able to record the topics (dashing version)

pulkitgoyal1506, 2020-07-29:
## Description
In my setup, I have two PCs on a LAN. PC-1 has the ROS (dashing version) running on it with topics publishing some data. PC-2 has the ROS (foxy version) running on it with recorder recording PC-1 data. I am recording data published on the topics of PC-1, using rosbag recorder on PC-2.

### Expected Behavior
I expect the recorder running ros foxy version to record the data being published from ros dashing version. Foxy is only able to list the topics from dashing and is unable to record, echo or hz them.

### Actual Behavior
Foxy bag recorder is not recording any data coming from dashing topics. Executing 'ros2 topic list' on PC-2 is listing all the topics running on PC-1. But 'ros2 bag record', 'ros2 topic hz' or 'ros2 topic echo' is not subscribing to any of those topics.

Output of ros2 bag record command on PC-2:
[image]

## To Reproduce

- PC1: ROS2 (dashing) publishing data on some topics
- PC2: ROS2 (foxy) recording data coming from PC-1

The commands that are not working:
``` ros2 topic echo '/topic/name'
``` ros2 bag record -a .

## System
- PC1: ROS 2 Dashing, Ubuntu 18.04
- PC2: ROS 2 Foxy, Ubuntu 20.04

## Additional context

Karsten1987, 2020-07-29:
As mentioned in https://github.com/ros2/rosbag2/issues/474, this is not a rosbag related problem. You can verify this by running a `ros2 topic pub /hello_from_dashing std_msgs/msg/String "{data: Hello World}"` on Dashing side and try to subscribe with `ros2 topic echo /hello_from_dashing` on Foxy side.

I've changed your title appropriately. Feel free to correct me if you see this as inaccurate.

clalancette, 2020-08-02:
We don't support communications between major releases of ROS.  In many cases it happens to work, but if it doesn't (which it clearly doesn't in this case), we don't guarantee to spend time on it.  For that reason, I'm going to close this.  If you happen to find the reason, and want to open a pull request to fix it, we'll happily review it.

closed 2020-08-02 by clalancette

Sollimann, 2021-11-26:
@pulkitgoyal1506 @Karsten1987 @clalancette any updates on this? Did you get it to work, if so, how? we also have the same issue. The robot itself is running ros2 dashing, while our software stack is running ros2 foxy.

adamdbrw, 2021-11-26:
@Sollimann we did make it work previously. I don't remember the exact setup, but surprisingly, setting one to FastDDS and the other to CycloneDDS solved the problem.

Sollimann, 2021-11-26:
@adamdbrw Okey, I see. which is using which DDS? Did you use this guide to set it up? https://docs.ros.org/en/foxy/Installation/DDS-Implementations.html

And also, might be a bit difficult to configure the ros2 dashing hosted on their RPI (which we have no insight to) in a docker container to use a different DDS :thinking:

adamdbrw, 2021-11-26:
It was probably FastRTPS on Dashing side (so no need to configure) and CycloneDDS on Foxy side. But as I said, it was a while ago so you need to check it yourself. Hope this helps.

Sollimann, 2021-11-26:
Great, thanks for the help! :) Will let you know if it worked or not!

adamdbrw, 2021-11-26:
Thanks, I would be interested to know. As @clalancette said, cross-version between major releases is not supported (as opposed to cross-vendor for same ROS version), so it is possible that updated rmw packages (there were several patches for each since I ran my setup) no longer allow this kind of communication.

Sollimann, 2021-12-01:
@adamdbrw I tried the following the guide [Working-with-Eclipse-CycloneDDS](https://docs.ros.org/en/foxy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html) to configure ros2 foxy to use Cyclone DDS and update my docker image as such:

DDS specifications on line **20, 31, 32**

[image]

Then I built the docker image for `linux/arm64` (arm64 required on rpi and jetson) using the following command:

```sh
$ docker build --platform linux/arm64 -t my-ros2-foxy-arm64-image .
```
and then ran it on my Nvidia Jetson using:

```sh
$ docker run ros2-foxy my-ros2-foxy-arm64-image --network host --privileged true
```

However, I'm still not able to publish/subscribe between foxy and dashing. **Suggestions for things I can try to make it work?** :) . The robot raspberry PI (top left) running ros2 dashing and the Nvidia Jetson (top right) running ros2 foxy is connected to the same network via a network switch (blue box with ethernet cables in bottom):

[image]

adamdbrw, 2021-12-01:
My setup was on a different architecture. Did you test Dashing to Dashing?
There are bridging solutions you can try if this doesn't work for you, but it is best to narrow it down and understand the reason.
It is also possible that earlier versions of rmw packages worked in this setup and current versions no longer do. 
I am certain it did run for us (I confirmed it was Dashing to Foxy) and we used this setup successfully for quite a while (but again, different platform, different versions of rmw packages).