ros2/ros2 #1002: Foxy Patch 2: tango_icons_vendor dependency

Jopand, 2020-08-11:
## Bug report

**Required Info:**

- Operating System:
  - Debian 10
- Installation type:
  - Source
- Version or commit hash:
  - Foxy Patch 2 (pulling source using ros2.repos)

#### Steps to reproduce issue
Run rosdep install as the "Building ROS 2 on Linux" guide states:
rosdep install --from-paths src --ignore-src --rosdistro foxy -y --skip-keys "console_bridge fastcdr fastrtps rti-connext-dds-5.3.1 urdfdom_headers"

#### Expected behavior
No error. The command should output:
"#All required rosdeps installed successfully"

#### Actual behavior
Error. The command outputs:
"ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
qt_gui: No definition of [tango_icons_vendor] for OS [debian]"

jacobperron, 2020-08-27:
The missing `tango_icons_vendor` dependency was added to the Foxy ros2.repos file in https://github.com/ros2/ros2/pull/1005.

@Jopand Please comment if this hasn't resolved the issue for you.

closed 2020-08-27 by jacobperron

Jopand, 2020-08-31:
@jacobperron Sorry for the late answer. This did resolve my issue. 👍