ros2/ros2 #1013: ROS2 python example package error

123dbl, 2020-08-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - ardent
- DDS implementation:
  - 
- Client library (if applicable):
  -  rclpy

#### Steps to reproduce issue

```
git clone https://github.com/ros2/examples.git
cd examples/rclpy/topics/minimal_publisher
colcon build --symlink-install
```

#### Expected behavior
```
Starting >>> examples_rclpy_minimal_publisher
Finished <<< examples_rclpy_minimal_publisher [3.52s]                       

Summary: 1 package finished [3.63s]
```
#### Actual behavior
```
colcon build --symlink-install                           
Starting >>> examples_rclpy_minimal_publisher
Traceback (most recent call last):lpy_minimal_publisher - 0.2s]
  File "<string>", line 1, in <module>
ImportError: No module named 'setuptools.extern.packaging'
--- stderr: examples_rclpy_minimal_publisher                   
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
  File "/usr/lib/python3/dist-packages/colcon_ros/package_identification/ros.py", line 130, in getter
    desc.path / 'setup.py', env=env)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 262, in get_setup_information
    setup_py, env=env)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 303, in _get_setup_information
    cwd=os.path.abspath(str(setup_py.parent)), check=True, env=env)
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1
---
Failed   <<< examples_rclpy_minimal_publisher [0.25s, exited with code 1]

Summary: 0 packages finished [0.37s]
  1 package failed: examples_rclpy_minimal_publisher
  1 package had stderr output: examples_rclpy_minimal_publisher
Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1
```

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2020-08-18:
ardent is E.O.L, could you try dashing or later instead? for distribution you can refer to https://index.ros.org/doc/ros2/Releases/.

clalancette, 2020-08-18:
> ardent is E.O.L, could you try dashing or later instead? for distribution you can refer to https://index.ros.org/doc/ros2/Releases/.

Yes.  Also, for starting questions like this, please ask them over on https://answers.ros.org.  That way we have the questions and answers all in one, searchable place.  I'm going to close this out for now.

closed 2020-08-18 by clalancette