ros2/ros2 #1017: Exception in handling service response if client is not available

blacksoul000, 2020-08-25:
In [Fast-RTPS PR](https://github.com/ros2/rmw_fastrtps/pull/390/files) was added check for subscribers count before publish service response and if it's equal 0 RMW_RET_ERROR is returned.
Application throws the unhandled exception RCLError.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - Foxy latest release
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Remove client while service callback is running

#### Expected behavior
No terminate.

#### Actual behavior
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
   what():  failed to send response: client will not receive response, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.2.0/src/rmw_response.cpp:124, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/service.c:342

#### Additional information
With Cyclone DDS all works fine

hidmic, 2020-08-25:
Hmm, yeah, [`rmw_fastrtps*_cpp`](https://github.com/ros2/rmw_fastrtps/blob/d7a9559729fb59da21ae921376ba98f907320018/rmw_fastrtps_shared_cpp/src/rmw_response.cpp#L125) and [`rmw_cyclonedds_cpp`](https://github.com/ros2/rmw_cyclonedds/blob/1bc587e77d1fc61df01b952e7053eff66069b61b/rmw_cyclonedds_cpp/src/rmw_node.cpp#L3578) behavior differs. Missed it back then. Thanks for reporting @blacksoul000 !

`RMW_RET_ERROR` is too generic for upper layers to do anything about the situation. I'm not entirely sure `RMW_RET_TIMEOUT` is super appropriate either, but for Foxy we may not have a choice. CC @jacobperron.

blacksoul000, 2020-08-25:
@hidmic any non-fault behavior will be good.

jacobperron, 2020-08-25:
Changing the return value to `RMW_RET_TIMEOUT` to make behavior between the RWM implementations sounds reasonable to me.

clalancette, 2020-09-10:
As I understand this ticket, there are actually two different questions:

1.  What to do for master/Galactic
1.  What to do for Foxy

I'm going to leave this one open as the ticket for dealing with master, and assign to @hidmic.  I'll then open a separate ticket to track the "backport" to Foxy, since I think it will be a somewhat different solution.

russkel, 2020-10-16:
I just ran into this where an experimental python node caused another node to terminate due to this exception.

hidmic, 2020-10-21:
I absolutely missed this ticket. https://github.com/ros2/rmw_fastrtps/pull/467 should solve it.

russkel, 2020-10-22:
@hidmic thanks for your hard work.

EduPonz, 2020-11-18:
Since ros2/rmw_fastrtps#467 and ros2/rmw_fastrtps#479 (backport to Foxy) have been merged, I think this issue may be closed

jacobperron, 2020-11-18:
> Since ros2/rmw_fastrtps#467 and ros2/rmw_fastrtps#479 (backport to Foxy) have been merged, I think this issue may be closed

:+1: 

Please re-open and comment if we're mistaken.

closed 2020-11-18 by jacobperron

connoranderson, 2022-04-05:
I'm able to reproduce this intermittently on version 5.0.1 (galactic) both in CI as well as on real wireless devices. To replicate the behavior we see in real life, I modified the demo add_two_ints client to the following:
```
example_interfaces::srv::AddTwoInts::Response::SharedPtr send_request(
    rclcpp::Node::SharedPtr node,
    rclcpp::Client<example_interfaces::srv::AddTwoInts>::SharedPtr client,
    example_interfaces::srv::AddTwoInts::Request::SharedPtr request) {
  auto result = client->async_send_request(request);
  std::this_thread::sleep_for(1s);  //       < ------------------------ MODIFIED
  // Wait for the result.
  if (rclcpp::spin_until_future_complete(node, result) ==
      rclcpp::FutureReturnCode::SUCCESS) {
    return result.get();
  } else {
    return NULL;
  }
}
```

Then, I launch a server **on a separate machine** (it doesn't break when running on local network) and then issue a bunch of service calls

On machine 1
`ros2 run demo_nodes_cpp add_two_ints_server`

On machine 2, connected via WIFI to machine 1
`seq 1 25 | parallel -j 100 ros2 run demo_nodes_cpp add_two_ints_client &`

It's not 100% of the time, but using this setup over a wifi-based network, I'm able to reproduce the exact error we see in the wild (below).

Terminate on machine 1
`terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to send response: client will not receive response, at /tmp/binarydeb/ros-galactic-rmw-fastrtps-shared-cpp-5.0.1/src/rmw_response.cpp:129, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/service.c:287`

While these steps are a bit severe (esp the blocking 1s sleep), they are representing our environment where it is common for a wireless device to lose connectivity periodically.

connoranderson, 2022-04-05:
Yeah, I'll create a new issue. Is ros2 repo the best one or should I put this in rmw_fastrtps? For now, I don't know if it will reproduce on rolling, since I get a segfault when running the same steps on rolling
```
docker run -it --rm --network=host -e RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros:rolling bash -c "apt update && apt install -y ros-rolling-demo-nodes-cpp ros-rolling-rmw-fastrtps-cpp && ros2 run demo_nodes_cpp add_two_ints_client"
...
[INFO] [1649178505.946512259] [add_two_ints_client]: service not available, waiting again...
[INFO] [1649178506.946785490] [add_two_ints_client]: service not available, waiting again...
[INFO] [1649178507.946862328] [add_two_ints_client]: service not available, waiting again...
[ros2run]: Segmentation fault
```

clalancette, 2022-04-05:
> Is ros2 repo the best one or should I put this in rmw_fastrtps?

Start with https://github.com/ros2/ros2, since we don't know where the problem is yet.

> For now, I don't know if it will reproduce on rolling, since I get a segfault when running the same steps on rolling

Hm, interesting.  I'm not able to reproduce this right now, but I'm not running over wifi.  We'll have to see if that is still a problem.