ros2/ros2 #1019: ROS2 Foxy: setup.sh enviroment variables not being set?

jamoralp, 2020-08-31:
## Bug report

**Required Info:**

- Operating System: Ubuntu Focal 20.04, running via `ros:foxy` Docker image
  - 
- Installation type: binaries
  - 
- Version or commit hash: https://dockr.ly/2ELWlUL
  - 
- DDS implementation: Fast-RTPS
  - 
- Client library (if applicable): rclcpp
  - 

Good morning,
I'm having some issues regarding the compilation package depending on `rclcpp`.
More specifically, the issue arises while compiling `soss-ros2` package, from [SOSS](https://github.com/eProsima/soss_v2/blob/feature/xtypes-dds/packages/ros2/CMakeLists.txt) repository. These issue is not present in older ROS 2 distros.

The issue is only reproducible when sourcing an installation of ROS 2 Foxy from binaries. If I compile ROS 2 Foxy from sources, then source my local environment `setup.bash`, the process finishes without any complaints.

#### Steps to reproduce issue

Run a docker container instance of `ros:foxy` Docker image. I'm following this approach since my laptop has not been updated yet to Ubuntu 20.04, but the output should be identical to the one described here.
```
docker run -ti --rm --net=host ros:foxy
```

Source your ROS 2 Foxy installation.
```
. /opt/ros/foxy/setup.bash
```

Install necessary packages.

```
apt update && apt install -y libyaml-cpp-dev libboost-dev libboost-program-options-dev
```

Create a workspace and download the SOSS repository.
```
mkdir -p ~/soss_ws/src
cd ~/soss_ws
git clone --recursive --branch feature/xtypes-dds https://github.com/eProsima/soss_v2.git src/soss
```
Compile the `soss-ros2` package.
```
colcon build --packages-up-to soss-ros2
```

#### Expected behavior

`soss-ros2` package should be compiled successfully.

#### Actual behavior

Following output is produced:
```
Starting >>> soss-ros2
--- stderr: soss-ros2
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find FastRTPS (missing: FastRTPS_INCLUDE_DIR FastRTPS_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /opt/ros/foxy/share/fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake:95 (find_package_handle_standard_args)
  /opt/ros/foxy/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_c-extras.cmake:7 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_runtime/cmake/rosidl_default_runtime-extras.cmake:12 (find_package)
  /opt/ros/foxy/share/rosidl_default_runtime/cmake/rosidl_default_runtimeConfig.cmake:41 (include)
  /opt/ros/foxy/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  /opt/ros/foxy/share/rcl_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rcl_interfaces/cmake/rcl_interfacesConfig.cmake:41 (include)
  /opt/ros/foxy/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rcl/cmake/rclConfig.cmake:41 (include)
  /opt/ros/foxy/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /opt/ros/foxy/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  CMakeLists.txt:8 (find_package)

---
Failed   <<< soss-ros2 [0.80s, exited with code 1]

Summary: 3 packages finished [42.7s]
  1 package failed: soss-ros2
  1 package had stderr output: soss-ros2
```

#### Additional information

The issue occurs during the `find_package(rclcpp)` CMake instruction. Although this project is not a ROS package, this is the first time this compilation issues have come to light; also, as I stated before, this is not happenning when compiling ROS 2 Foxy from sources, then sourcing my build environment and executing the same steps as described above, starting from step 3. Any thoughts?

clalancette, 2020-09-24:
I'm able to reproduce the problem.  My current suspicion is that the SOSS stuff is a non-ament package, using old-style CMake targets.  In Foxy, we pushed a lot to use new-style CMake targets.  There's probably some mismatch here where the old-style stuff isn't being exported as it should be, but I'm honestly not sure at the moment.

I'm fairly confident that one way to fix this would be to ament-ify these packages.  That may or may not fit in with the goals of the SOSS project right now.

jamoralp, 2020-09-28:
> I'm able to reproduce the problem. My current suspicion is that the SOSS stuff is a non-ament package, using old-style CMake targets. In Foxy, we pushed a lot to use new-style CMake targets. There's [...]

Hello @clalancette, thank you for digging into this issue.
Yes, that would make sense. I don't think there's a problem from our side to ament-ify SOSS packages, but I just could not understand why this happened only when migrating to Foxy. I'll try and report ASAP. Thank you!

dirk-thomas, 2020-09-28:
I don't think this has anything to do with the CMake target work since the failure happens when trying to find CMake packages. I also wouldn't expect this to be any different with pre-Foxy distros.

The reason is that the `CMAKE_PREFIX_PATH` isn't set for this plain CMake package. If you set it manually before the build works fine: `CMAKE_PREFIX_PATH=/opt/ros/foxy colcon build --packages-up-to soss-ros2`.

Only for ament packages `AMENT_PREFIX_PATH` is considered and if applicable added to the `CMAKE_PREFIX_PATH`.

Converting the package to be of type `ros.ament_cmake` works around the problem. You can emulate it by just dropping a `colcon.pkg` file with the following content into the package directory:

```
{
    "name": "soss-ros2",
    "type": "ros.ament_cmake",
    "dependencies": ["xtypes", "soss-core", "soss-rosidl"]
}
```

But I think there should be a proper way to support this - I am just not sure yet how that would look like.

dirk-thomas, 2020-09-28:
The package type doesn't need to be `ament_cmake` - a ROS CMake package should be sufficient:

```
"type": "ros.cmake"
```

Or an equivalent `package.xml` file:

```
<package ...>
  <depend>xtypes</depend>
  <depend>soss-core</depend>
  <depend>soss-rosidl</depend>
  <export>
    <build_type>cmake</build_type>
  </export>
</package>
```

jamoralp, 2021-02-15:
Sorry, I forgot to close this issue. This made the trick for me, thanks

closed 2021-02-15 by jamoralp