ros2/ros2 #1020: Foxy build can't receive messages published by Dashing if FastRTPS used

ozgudum, 2020-08-31:
## Bug report

**Required Info:**

- Operating System:
  - MacOs 10.15
- Installation type:
  - ROS2 Foxy Source (Subscriber), ROS2 Dashing (Publisher)
- DDS implementation:
  - FastRTPS

#### Steps to reproduce issue

Run talker node in demo_nodes_cpp package from Dashing build (either on the same machine or another machine on the same network). Listener node on Foxy will not receive/display messages if FastRTPS is used. If I switch to CycloneDDS on Foxy, it will receive/display the messages. Interestingly, Dashing can receive/display messages published on Foxy but not vice a versa.

I tested talker/listener example as well as publishing simple std_msgs::msg:String message. Neither of the message, are displayed on the foxy build. 

On Dashing 
```
ros2 run demo_nodes_cpp talker
```

On Foxy (FastRTPS)
```
ros2 run demo_nodes_cpp listener
```

Error seems to be on Foxy side and possible on FastRTPS. Anyone experienced the same issue or have thoughts on how to solve it? I need to use FastRTPS.

clalancette, 2020-08-31:
We do extensive testing among DDS/RTPS vendors within a ROS distribution, but we don't do any testing or support communication between different ROS versions.  That leaves us free to make changes to the protocol or how we use it between ROS distributions.

That is all to say that this isn't a combination we support, and we're unlikely to spend time tracking down the issue.  If you discover a fix for it, though, please feel free to open a pull request and we can discuss it there.

rafaelhuang-intel, 2020-09-23:
Hi,

I also meet same issue on Ubuntu (dashing talk & Foxy listen) & Yocto,

MiguelCompany, 2020-09-23:
@ozgudum @rafaelhuang-intel I think this was solved by eProsima/Fast-DDS#809.

Could you check using Fast DDS v1.8.4 on the Dashing side?

blacksoul000, 2020-09-24:
https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/#rmw-connext-cpp