ros2/ros2 #1021: Using the cmake package registry creates crosstalk between vendor packages on different workspaces

ivanpauno, 2020-09-03:
From the discussion [here](https://github.com/ros2/yaml_cpp_vendor/pull/16#issuecomment-686753998), this is the list of packages affected:
- [libcurl_vendor](https://github.com/ros/resource_retriever/blob/8053dd51632fae012a65f6f704d29d55ae5d856f/libcurl_vendor/CMakeLists.txt#L68)
- [sqlite3_vendor](https://github.com/ros2/rosbag2/blob/f8f5f669270081f9f5969c301b8b2ca39b75abbe/sqlite3_vendor/CMakeLists.txt#L12)
- [zstd_vendor](https://github.com/ros2/rosbag2/blob/f8f5f669270081f9f5969c301b8b2ca39b75abbe/zstd_vendor/CMakeLists.txt#L12)
- [tinyxml_vendor](https://github.com/ros2/tinyxml_vendor/blob/91ad3796eda73cf8cac09f7d0e7b2344d2e9acea/CMakeLists.txt#L12)
- [rviz_assimp_vendor](https://github.com/ros2/rviz/blob/40e3b9ae9712ce4df7ea368e714dbc509064e8b0/rviz_assimp_vendor/CMakeLists.txt#L61)
- [spdlog_vendor](https://github.com/ros2/spdlog_vendor/blob/0223d4586fff5863e6d380c38bde5fee46802af7/CMakeLists.txt#L12)
- [console_bridge_vendor](https://github.com/ros2/console_bridge_vendor/blob/6cffe405a707a4cbf6ff7a1d13ac22552b3e22ae/CMakeLists.txt#L12)
- [google_benchmark_vendor](https://github.com/ament/google_benchmark_vendor/blob/2bfd73afe808ba1ee8be3b87e5eebf4e26d7e759/CMakeLists.txt#L9)
- [eProsima/foonathan_memory_vendor](https://github.com/eProsima/foonathan_memory_vendor/blob/7ca1e109c2d4d7f0b3b03f6880c770ef00ed79de/CMakeLists.txt#L4)

### How to reproduce

- Build any of those vendor packages in a workspace (wsA), without having the system package installed.
- Repeat in an unrelated workspace (wsB). Do not source wsA before building.
- Delete wsA. Dependencies of the vendor package in wsB will have runtime problems, because it was using the package downloaded and built in wsA (found through the cmake package registry).

### Suggested action

Use `NO_CMAKE_PACKAGE_REGISTRY` option in all `find_package` calls of vendor packages using the "try to find then build approach".
In that way, crosstalk between packages is avoided.
A system installed version or a version in an underlay ws can still be found when this option is passed (see https://github.com/ros2/yaml_cpp_vendor/pull/16#discussion_r477515425).

### Alternative

Colcon could by default deactivate the cmake package registry.
Though that default can be nice we should still ignore the package registry in vendor packages, as some people might want to use the package registry (and using it in vendor packages is **always** a problem).

### Revisit

https://github.com/ros2/yaml_cpp_vendor/pull/16 should be revisited after a decision on how to handle this problem is taken.

gavanderhoorn, 2020-09-04:
As @dirk-thomas mentioned in https://github.com/ros2/yaml_cpp_vendor/pull/16#discussion_r477588763, we've ran into this in https://github.com/ros-industrial-consortium/tesseract_ext/issues/18 (and some others as well).

@ivanpauno: you mention you don't believe disabling the registry by default is *always* a good idea. Could you more clearly describe an example where it wouldn't be?

My reasoning is that if you're using Colcon (or Catkin), you're using the workspace approach. A workspace by definition is something which exploits isolation (similar to Conda, Conan, etc). Even when using "vendor pkgs", the idea is those packages only influence the state of the workspace, never anything outside of it.

At least, that has been my understanding of the workspace concept.

If you don't need or want the isolation, that to me means you should perhaps not be building something in a workspace.

With this in mind, it seems strange for a package in something which should be isolated to be registered in a global index where entities *outside* of this isolated environment can discover artefacts *inside* it -- this seems an undesirable situation to me. For overlayed workspaces, it could work, but for siblings fi you typically don't want anything from any sibling to show up in another. Otherwise you would not have made them siblings I believe.

The vendor pkgs were created specifically for the cases where a system-provided dependency is/was not the right version, or it wasn't built with the specific options required. If those then register themselves in the CMake registry, they become findable, even for non-workspace projects, which potentially will make them use these custom-built, more up-to-date versions. This could lead to incompatible transitive linking problems, where vendored versions bring in newer and incompatible versions of dependencies.

All-in-all, to me it seems like the registry is breaking the concept of workspace isolation, and going for a default which disables the registry makes sense.

ivanpauno, 2020-09-04:
> @ivanpauno: you mention you don't believe disabling the registry by default is always a good idea. Could you more clearly describe an example where it wouldn't be?

I think that disabling the cmake package registry by default globally may be a good idea.
Some people might still want to enable the package registry and don't use that default, so we cannot rely on it being disabled.

Thus, I think `NO_CMAKE_PACKAGE_REGISTRY` should be used were appropriate, regardless of what the global setting is.
In vendor packages following the "try find, then download and build" approach, using that option seems appropriate (even if you want to have the cmake package registry globally enabled, using the registry in vendor packages is undesired).

ivanpauno, 2020-09-04:
> With this in mind, it seems strange for a package in something which should be isolated to be registered in a global index where entities outside of this isolated environment can discover artefacts inside it 

I'm not sure if there's a way to avoid packages being registered in the cmake user registry.
All the mentioned options (the global and the local one), skip the package registry when doing `find_package`, but there's no way to avoid a package from being registered AFAIK.

ivanpauno, 2020-09-04:
There's a way of disabling packages from being registered (see https://cmake.org/cmake/help/v3.2/manual/cmake-packages.7.html#disabling-the-package-registry and https://cmake.org/cmake/help/v3.2/variable/CMAKE_EXPORT_NO_PACKAGE_REGISTRY.html#variable:CMAKE_EXPORT_NO_PACKAGE_REGISTRY).

If we want to have a global setting for avoiding the package registry, I think it's better to disable "registering", that to disable it in "find_package".
In that way, all packages from a workspace won't be accessible from outside.

ivanpauno, 2020-09-17:
Another detail to take in account is why the vendored packages are being registered in the package registry at all.
That should only happen when you do:
```cmake
export(PACKAGE)
```
(see [export](https://cmake.org/cmake/help/v3.2/command/export.html#command:export), [user pacakge registry](https://cmake.org/cmake/help/v3.2/variable/CMAKE_EXPORT_NO_PACKAGE_REGISTRY.html#variable:CMAKE_EXPORT_NO_PACKAGE_REGISTRY))
e.g.: when `yaml_cpp_vendor` downloads and builds `yaml_cpp`, it's being registered in the package registry though `yaml_cpp` never uses `export(PACKAGE)` directive.

I guess that [ExternalProject_Add](https://cmake.org/cmake/help/latest/module/ExternalProject.html) is automatically registering the downloaded package (if that's the case, that's an undocumented cmake feature/bug).

---

Besides why vendor package are getting registered in the cmake user registry, globally setting [CMAKE_EXPORT_PACKAGE_REGISTRY](https://cmake.org/cmake/help/v3.2/variable/CMAKE_EXPORT_NO_PACKAGE_REGISTRY.html#variable:CMAKE_EXPORT_NO_PACKAGE_REGISTRY) makes sense to me.