ros2/ros2 #1022: rosidl_generate_interfaces.cmake:240 ERROR

zc804351220, 2020-09-08:
When I custom ROS 2 msg  [https://index.ros.org/doc/ros2/Tutorials/Custom-ROS2-Interfaces/](url) 
Error :
CMake Error at /home/hagan/ros2_dashing/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:240 (list):
  list index: 1 out of range (-1, 0)
Call Stack (most recent call first):
  CMakeLists.txt:19 (rosidl_generate_interfaces)

How can I fix it ?

clalancette, 2020-09-08:
When you file a bug report, please include a lot more information, including the custom messages you are using, the CMakeLists.txt you are using, etc.  Otherwise we have no way of trying to debug or figure out what is going wrong.

zc804351220, 2020-09-09:
> When you file a bug report, please include a lot more information, including the custom messages you are using, the CMakeLists.txt you are using, etc. Otherwise we have no way of trying to debug or figure out what is going wrong.

Thank you ~
I fix it by changing  the path of ros-workspace which only contains english!

closed 2020-09-09 by zc804351220