ros2/ros2 #1024: Foxy: Exception in handling service response if client is not available

clalancette, 2020-09-10:
In [Fast-RTPS PR](https://github.com/ros2/rmw_fastrtps/pull/390/files) was added check for subscribers count before publish service response and if it's equal 0 RMW_RET_ERROR is returned.
Application throws the unhandled exception RCLError.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - Foxy latest release
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Remove client while service callback is running

#### Expected behavior
No terminate.

#### Actual behavior
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
   what():  failed to send response: client will not receive response, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.2.0/src/rmw_response.cpp:124, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/service.c:342

#### Additional information
With Cyclone DDS all works fine

---------------------------
Follow-ups

hidmic:

Hmm, yeah, rmw_fastrtps*_cpp and rmw_cyclonedds_cpp behavior differs. Missed it back then. Thanks for reporting @blacksoul000 !

RMW_RET_ERROR is too generic for upper layers to do anything about the situation. I'm not entirely sure RMW_RET_TIMEOUT is super appropriate either, but for Foxy we may not have a choice. CC @jacobperron.

--------------------------------------
blacksoul000

@hidmic any non-fault behavior will be good.

-------------------------------------
jacobperron:

Changing the return value to RMW_RET_TIMEOUT to make behavior between the RWM implementations sounds reasonable to me.

hidmic, 2020-10-21:
Once https://github.com/ros2/rmw_fastrtps/pull/467 is backported, this should be solved.

EduPonz, 2020-11-18:
@hidmic ,
@jacobperron backported it in ros2/rmw_fastrtps#479, which was merged yesterday, so I think this may be closed.

closed 2020-11-18 by jacobperron