ros2/ros2 #1026: Timer intervals and long running callbacks

rrrand, 2020-09-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - Fast-RTPS, CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
```
#include <rclcpp/rclcpp.hpp>
#include <unistd.h>

class CheckTimerNode : public rclcpp::Node
{
public:
    CheckTimerNode(const rclcpp::NodeOptions& options) :
        Node("CheckTimerNode", options)
    {
        m_clock = std::make_shared< rclcpp::Clock >(RCL_SYSTEM_TIME);
        m_lastTime1 = m_clock->now().seconds();
        m_timer1 = this->create_wall_timer(
                       std::chrono::milliseconds(100),
                       std::bind(&CheckTimerNode::timer1Callback, this));

        m_lastTime2 = m_clock->now().seconds();
        m_timer2 =this->create_wall_timer(
                      std::chrono::milliseconds(20),
                      std::bind(&CheckTimerNode::timer2Callback, this));
    }
private:
    void timer1Callback()
    {
        const auto newTime = m_clock->now().seconds();
        RCLCPP_WARN(this->get_logger(), "Timer 1 interval %f s", newTime - m_lastTime1);
        m_lastTime1 = newTime;
        usleep(40 * 1000);
    }

    void timer2Callback()
    {
        const auto newTime = m_clock->now().seconds();
        RCLCPP_WARN(this->get_logger(), "Timer 2 interval %f s", newTime - m_lastTime2);
        m_lastTime2 = newTime;
    }

    rclcpp::Clock::SharedPtr m_clock;
    double m_lastTime1;
    rclcpp::TimerBase::SharedPtr m_timer1;
    double m_lastTime2;
    rclcpp::TimerBase::SharedPtr m_timer2;
};

int main(int argc, char** argv)
{
    setvbuf(stdout, NULL, _IONBF, BUFSIZ);

    rclcpp::init(argc, argv);
    rclcpp::executors::SingleThreadedExecutor executor;
    rclcpp::NodeOptions options;
    auto node = std::make_shared< CheckTimerNode >(options);
    executor.add_node(node);
    executor.spin();
    rclcpp::shutdown();
    return 0;
}
```

#### Expected behavior
Timer 2 interval equals 0.4 s (timer 1 callback duration)  after timer 1 callback finish.

#### Actual behavior
Timer 2 interval equals 0.6 s (timer 1 callback duration + timer 2 interval) after timer 1 callback finish.

```
[WARN] [1599811854.819983825] [CheckTimerNode]: Timer 1 interval 0.100000 s
[WARN] [1599811854.860210220] [CheckTimerNode]: Timer 2 interval 0.060223 s
[WARN] [1599811854.879985116] [CheckTimerNode]: Timer 2 interval 0.019775 s
[WARN] [1599811854.899982655] [CheckTimerNode]: Timer 2 interval 0.019998 s
[WARN] [1599811854.919976506] [CheckTimerNode]: Timer 1 interval 0.099993 s
[WARN] [1599811854.960219230] [CheckTimerNode]: Timer 2 interval 0.060237 s
[WARN] [1599811854.979976024] [CheckTimerNode]: Timer 2 interval 0.019757 s
[WARN] [1599811854.999977168] [CheckTimerNode]: Timer 2 interval 0.020001 s
```

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2020-09-23:
I think this is not a problem, cyclic is processed in order to `create_wall_timer`.

specifically,

> [WARN] [1599811854.860210220] [CheckTimerNode]: Timer 2 interval 0.060223 s

this includes 0.02msec. (0.04msec sleep + 0.02msec period)

if we change the order to `create_wall_timer` like https://github.com/fujitatomoya/ros2_test_prover/blob/a4f82815618b3fe2dbebdf665ce6f798f590db65/src/ros2_1026.cpp#L9-L18,

```
[WARN] [1600872223.209071251] [CheckTimerNode]: Timer 2 interval 0.020029 s
[WARN] [1600872223.209228759] [CheckTimerNode]: Timer 1 interval 0.100047 s
[WARN] [1600872223.249597384] [CheckTimerNode]: Timer 2 interval 0.040526 s
[WARN] [1600872223.269047561] [CheckTimerNode]: Timer 2 interval 0.019451 s
[WARN] [1600872223.289054378] [CheckTimerNode]: Timer 2 interval 0.020007 s
[WARN] [1600872223.309052191] [CheckTimerNode]: Timer 2 interval 0.019998 s
[WARN] [1600872223.309210530] [CheckTimerNode]: Timer 1 interval 0.099981 s
[WARN] [1600872223.349647515] [CheckTimerNode]: Timer 2 interval 0.040593 s
[WARN] [1600872223.369066927] [CheckTimerNode]: Timer 2 interval 0.019421 s
...
```

[periods_ahead](https://github.com/ros2/rcl/blob/8bcada7b884693644d7ba2e3bd2c22fd9541eba7/rcl/src/rcl/timer.c#L277) will be

- 1: if create_wall_timer m_timer2 -> m_timer1
- 2: if create_wall_timer m_timer1 -> m_timer2

rrrand, 2020-09-23:
First of all I'd like to thank you for investigation!
I need some time to check applied links because rcl is something new to me.
Anyway I think that evalution of next time when to call timer should be independent from all other events. My example is very synthetic. In common we often cannot assume how much time callback requires.

fujitatomoya, 2020-09-24:
> Anyway I think that evalution of next time when to call timer should be independent from all other events.

agree. but with current design and implementation, i do not think it can be supported even with [MultiThreadedExecutor](class MultiThreadedExecutor : public rclcpp::Executor).

rrrand, 2020-09-25:
I should say you are right. Thank you.
My understatding of what to wait from timers was not correct.

I'd like to share some other examples.
Qt timers
```
#include <QTime>
#include <QTimer>
#include <QThread>
#include <QDebug>
#include <chrono>
#include <thread>

class TestTimer : public QObject
{
    Q_OBJECT

public:
    TestTimer() : QObject()
    {
        m_time.start();
        
        connect(&m_timer1, &QTimer::timeout, this, &TestTimer::testTimer1);
        m_timer1.start(100);
        
        connect(&m_timer2, &QTimer::timeout, this, &TestTimer::testTimer2);
        m_timer2.start(20);
    };

public slots:
    void testTimer1(void)
    {
        const auto newTime = m_time.elapsed();
        qDebug() << "Timer 1 interval" << newTime - m_lastTime1 << "s";
        m_lastTime1 = newTime;
        std::this_thread::sleep_for(std::chrono::milliseconds(40));

    };
    void testTimer2(void)
    {
        const auto newTime = m_time.elapsed();
        qDebug() << "Timer 2 interval" << newTime - m_lastTime2 << "s";
        m_lastTime2 = newTime;
    };
private:
    QTime m_time;
    QTimer m_timer1;
    QTimer m_timer2;
    int m_lastTime1 {0};
    int m_lastTime2 {0};
};
```
produce next output
```
Timer 1 interval 99 s
Timer 2 interval 60 s
Timer 2 interval 20 s
Timer 2 interval 20 s
Timer 1 interval 100 s
Timer 2 interval 59 s
Timer 2 interval 20 s
Timer 2 interval 20 s
Timer 1 interval 101 s
Timer 2 interval 61 s
Timer 2 interval 0 s
Timer 2 interval 20 s
Timer 2 interval 20 s
Timer 1 interval 99 s
Timer 2 interval 59 s
Timer 2 interval 20 s
Timer 2 interval 21 s
Timer 1 interval 101 s
```
So Qt timers behave almost like ROS timers here.

rrrand, 2020-09-25:
And another example. With sleep in suscription:
```
#include <chrono>
#include <functional>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"

using namespace std::chrono_literals;

using std::placeholders::_1;

class CheckTimerNode : public rclcpp::Node
{
public:
    CheckTimerNode(const rclcpp::NodeOptions& options) :
        Node("CheckTimerNode", options)
    {
        m_clock = std::make_shared< rclcpp::Clock >(RCL_SYSTEM_TIME);

        rclcpp::QoS qos = rclcpp::QoS(rclcpp::KeepAll()).reliable();
        m_publisher = this->create_publisher<std_msgs::msg::String>("topic", qos);
        m_lastTime1 = m_clock->now().seconds();
        m_timer1 = this->create_wall_timer(
                       std::chrono::milliseconds(100),
                       std::bind(&CheckTimerNode::timer1Callback, this));

        m_lastTime2 = m_clock->now().seconds();
        m_timer2 =this->create_wall_timer(
                      std::chrono::milliseconds(20),
                      std::bind(&CheckTimerNode::timer2Callback, this));

        m_subscription = this->create_subscription<std_msgs::msg::String>(
            "topic", qos, std::bind(&CheckTimerNode::topic_callback, this, _1));
    }
private:
    void timer1Callback()
    {

        const auto newTime = m_clock->now().seconds();
        RCLCPP_WARN(this->get_logger(), "Timer 1 interval %f s", newTime - m_lastTime1);
        m_lastTime1 = newTime;

        m_publisher->publish(std_msgs::msg::String());
    }

    void timer2Callback()
    {
        const auto newTime = m_clock->now().seconds();
        RCLCPP_WARN(this->get_logger(), "Timer 2 interval %f s", newTime - m_lastTime2);
        m_lastTime2 = newTime;
    }

    void topic_callback(const std_msgs::msg::String::SharedPtr msg) const
    {
      RCLCPP_INFO(this->get_logger(), "I heard: '%s'", msg->data.c_str());
      std::this_thread::sleep_for(std::chrono::milliseconds(40));
    }

    rclcpp::Clock::SharedPtr m_clock;
    double m_lastTime1;
    rclcpp::TimerBase::SharedPtr m_timer1;
    double m_lastTime2;
    rclcpp::TimerBase::SharedPtr m_timer2;

    rclcpp::Subscription<std_msgs::msg::String>::SharedPtr m_subscription;
    rclcpp::Publisher<std_msgs::msg::String>::SharedPtr m_publisher;
};

int main(int argc, char** argv)
{
    setvbuf(stdout, NULL, _IONBF, BUFSIZ);

    rclcpp::init(argc, argv);
    rclcpp::executors::SingleThreadedExecutor executor;
    rclcpp::NodeOptions options;
    auto node = std::make_shared< CheckTimerNode >(options);
    executor.add_node(node);
    executor.spin();
    rclcpp::shutdown();
    return 0;
}
```

Output:
```
[WARN] [1601037213.948371543] [CheckTimerNode]: Timer 1 interval 0.100031 s
[WARN] [1601037213.948428660] [CheckTimerNode]: Timer 2 interval 0.020084 s
[WARN] [1601037213.948456175] [CheckTimerNode]: I heard: ''
[WARN] [1601037213.988567737] [CheckTimerNode]: Timer 2 interval 0.040138 s
[WARN] [1601037214.008342937] [CheckTimerNode]: Timer 2 interval 0.019775 s
[WARN] [1601037214.028336615] [CheckTimerNode]: Timer 2 interval 0.019994 s
[WARN] [1601037214.048331803] [CheckTimerNode]: Timer 1 interval 0.099961 s
[WARN] [1601037214.048389715] [CheckTimerNode]: Timer 2 interval 0.020054 s
[WARN] [1601037214.048419008] [CheckTimerNode]: I heard: ''
[WARN] [1601037214.088517578] [CheckTimerNode]: Timer 2 interval 0.040127 s
[WARN] [1601037214.108344042] [CheckTimerNode]: Timer 2 interval 0.019826 s
[WARN] [1601037214.128346788] [CheckTimerNode]: Timer 2 interval 0.020002 s
[WARN] [1601037214.148335971] [CheckTimerNode]: Timer 1 interval 0.100003 s
```
Here we can see that timer 2 interval is 0.02 sec period or 0.04 sec sleep. And it is ok.

Should I close the issue?

fujitatomoya, 2020-09-26:
> Should I close the issue?

I guess it is up to you, but I think your original problem statement is not exactly the problem.

rrrand, 2020-09-28:
It's not a problem..

closed 2020-09-28 by rrrand