ros2/ros2 #1031: Rolling Ridley - call(ing) local_setup on Windows fails to find other setup scripts

ryankshah, 2020-09-18:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - From Source
- Version or commit hash:
  - From here: https://index.ros.org/doc/ros2/Installation/Rolling/Windows-Development-Setup/#windows-latest (https://github.com/ros2/ros2/commit/b8c14aa08cc4fb0b4e29e3ac579c9f4b32e1f077)
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
1. Follow all the setup steps
2. Run `call C:\dev\ros2_rolling\install\local_setup.bat`
```

#### Expected behavior

Source files as normal.

#### Actual behavior

Prints the following:

```
C:\WINDOWS\system32>call C:\dev\ros2_rolling\install\local_setup.bat
not found: "C:\dev\ros2_rolling\install\share/rttest/local_setup.bat"
not found: "C:\dev\ros2_rolling\install\share/tlsf/local_setup.bat"
not found: "C:\dev\ros2_rolling\install\share/tlsf_cpp/local_setup.bat"
not found: "C:\dev\ros2_rolling\install\share/pendulum_control/local_setup.bat"
```

ryankshah, 2020-09-18:
Seems this has happened before with no solution https://github.com/ros2/ros2/issues/709

clalancette, 2020-09-18:
This is currently expected, though it is ugly.  The problem is basically that those packages that are listed are Linux-only, so we skip them on Windows.  I'm not quite sure how to solve it.

Out of curiousity, is this causing problems for you?

dirk-thomas, 2020-09-18:
Duplicate of #709 and tracked in #914.

closed 2020-09-18 by dirk-thomas

ryankshah, 2020-09-18:
> This is currently expected, though it is ugly. The problem is basically that those packages that are listed are Linux-only, so we skip them on Windows. I'm not quite sure how to solve it.
> 
> Out of curiousity, is this causing problems for you?

Hi @clalancette I have not progressed further in the guide for building from source past the "Environment setup¶" point. I had it in mind that I could not carry on without this being complete. If there should be no issue for Windows regarding this, it should be reflected in the documentation.