ros2/ros2 #1035: Timer hangs in MultiThreaded composition

rrrand, 2020-09-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - Fast-RTPS, CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Install ros-foxy-composition. Change executable from 'component_container' to 'component_container_mt' and launch.

#### Expected behavior
Talker publishes every second.

#### Actual behavior
Talker timer interval varies from 1 second to 3 seconds with RMW_IMPLEMENTATION=rmw_fastrtps_cpp. Talker timer almost freezes RMW_IMPLEMENTATION=rmw_cyclonedds_cpp (interval can exceed 1 minute).

#### Additional information

fujitatomoya, 2020-09-24:
problem confirmed with https://github.com/ros2/ros2/commit/ca09b31ce2851773cf722e1fe1e64b70ee651fdd, i think that this is clearly bug.

- rmw_fastrtps

```
# ros2 launch composition composition_demo.launch.py
...
[component_container_mt-1] [INFO] [1600909915.464264807] [listener]: I heard: [Hello World: 3]
[component_container_mt-1] [INFO] [1600909915.872504503] [talker]: Publishing: 'Hello World: 4'
[component_container_mt-1] [INFO] [1600909915.872798995] [listener]: I heard: [Hello World: 4]
[component_container_mt-1] [INFO] [1600909920.459420276] [talker]: Publishing: 'Hello World: 5'
[component_container_mt-1] [INFO] [1600909920.459674758] [listener]: I heard: [Hello World: 5]
[component_container_mt-1] [INFO] [1600909921.295524181] [talker]: Publishing: 'Hello World: 6'
[component_container_mt-1] [INFO] [1600909921.295784763] [listener]: I heard: [Hello World: 6]
[component_container_mt-1] [INFO] [1600909923.741572267] [talker]: Publishing: 'Hello World: 7'
[component_container_mt-1] [INFO] [1600909923.741826639] [listener]: I heard: [Hello World: 7]
[component_container_mt-1] [INFO] [1600909923.865117757] [talker]: Publishing: 'Hello World: 8'
[component_container_mt-1] [INFO] [1600909923.865230528] [listener]: I heard: [Hello World: 8]
```

- rmw_cyclonedds

```
# ros2 launch composition composition_demo.launch.py
...
[component_container_mt-1] [INFO] [1600910072.363149916] [talker]: Publishing: 'Hello World: 6'
[component_container_mt-1] [INFO] [1600910072.363387628] [listener]: I heard: [Hello World: 6]
[component_container_mt-1] [INFO] [1600910077.455941432] [talker]: Publishing: 'Hello World: 7'
[component_container_mt-1] [INFO] [1600910077.456099793] [listener]: I heard: [Hello World: 7]
[component_container_mt-1] [INFO] [1600910092.669045052] [talker]: Publishing: 'Hello World: 8'
[component_container_mt-1] [INFO] [1600910092.669221723] [listener]: I heard: [Hello World: 8]
[component_container_mt-1] [INFO] [1600910094.663010070] [talker]: Publishing: 'Hello World: 9'
[component_container_mt-1] [INFO] [1600910094.663178151] [listener]: I heard: [Hello World: 9]
```

in addition, when it launches with `component_container_mt`, it consumes a lot of CPU resource.

fujitatomoya, 2020-09-28:
I believe the more number of threads exist, the higher possibility there is. (I confirmed with AMD Ryzen 9 3900X 12-Core Processor 24 threads) I think that the reason is that `scheduled_timers_` https://github.com/ros2/rclcpp/blob/148d295416dba3814162465392e56116195c2304/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp#L88-L94. If any `scheduled_timers_` is available, it just makes it difficult to erase `any_exec.timer` from it since other threads are trying to get `wait_mutex_`  -> continue always. and then eventually [period_ahead](https://github.com/ros2/rcl/blob/8bcada7b884693644d7ba2e3bd2c22fd9541eba7/rcl/src/rcl/timer.c#L277) makes it miss the timer. (I might be wrong, i will be looking into it.)

Note, there is another problem around `scheduled_timers_` https://github.com/ros2/rclcpp/pull/1328 (CC: @ivanpauno )

<details>
<summary>strace & GDB trace information</summary>
<pre>
<code>
#  ps -ef | grep 30858
root     30858 30848 56 22:15 pts/3    00:08:59 /root/ros2_ws/colcon_ws/install/rclcpp_components/lib/rclcpp_components/component_container_mt --ros-args -r __node:=my_container -r __ns:=/
# strace -tt -f -p 30858
...<snip>
[pid 30861] 22:30:40.083006 futex(0x55f94048d568, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1601299840, tv_nsec=82990256}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
[pid 30861] 22:30:40.083077 futex(0x55f94048cd60, FUTEX_WAKE_PRIVATE, 1) = 0
[pid 30861] 22:30:40.083111 futex(0x55f94048d568, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1601299840, tv_nsec=83100827}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
[pid 30861] 22:30:40.083171 futex(0x55f94048cd60, FUTEX_WAKE_PRIVATE, 1) = 0
[pid 30861] 22:30:40.083198 futex(0x55f94048d568, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1601299840, tv_nsec=83191737}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
[pid 30861] 22:30:40.083261 futex(0x55f94048cd60, FUTEX_WAKE_PRIVATE, 1) = 0
[pid 30861] 22:30:40.083293 futex(0x55f94048d568, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1601299840, tv_nsec=83487858}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
[pid 30861] 22:30:40.083560 futex(0x55f94048cd60, FUTEX_WAKE_PRIVATE, 1) = 0
[pid 30861] 22:30:40.083590 futex(0x55f94048d568, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1601299840, tv_nsec=83582850}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
[pid 30861] 22:30:40.083652 futex(0x55f94048cd60, FUTEX_WAKE_PRIVATE, 1) = 0
[pid 30861] 22:30:40.083679 futex(0x55f94048d568, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1601299840, tv_nsec=83673011}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
[pid 30861] 22:30:40.083743 futex(0x55f94048cd60, FUTEX_WAKE_PRIVATE, 1) = 0
[pid 30861] 22:30:40.083771 futex(0x55f94048d568, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=160129984
(gdb) info threads
  Id   Target Id                         Frame
* 1    Thread 0x7fe9a9608600 (LWP 30858) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  2    Thread 0x7fe9a8cdc700 (LWP 30860) futex_abstimed_wait_cancelable (private=0, abstime=0x0, clockid=0, expected=0,
    futex_word=0x7fe9aa5f8d60 <rclcpp::SignalHandler::signal_handler_sem_>) at ../sysdeps/nptl/futex-internal.h:320
  3    Thread 0x7fe9a84db700 (LWP 30861) futex_abstimed_wait_cancelable (private=<optimized out>, abstime=0x7fe9a84da680, clockid=<optimized out>,
    expected=0, futex_word=0x55f94048d568) at ../sysdeps/nptl/futex-internal.h:320
  4    Thread 0x7fe9a7cda700 (LWP 30862) __libc_recvmsg (flags=0, msg=0x7fe9a7cd91c0, fd=8) at ../sysdeps/unix/sysv/linux/recvmsg.c:28
  5    Thread 0x7fe9a74d9700 (LWP 30863) __libc_recvmsg (flags=0, msg=0x7fe9a74d81c0, fd=9) at ../sysdeps/unix/sysv/linux/recvmsg.c:28
  6    Thread 0x7fe9a6cd8700 (LWP 30864) __libc_recvmsg (flags=0, msg=0x7fe9a6cd71c0, fd=10) at ../sysdeps/unix/sysv/linux/recvmsg.c:28
  7    Thread 0x7fe9a6417700 (LWP 30865) futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x55f94048d728)
    at ../sysdeps/nptl/futex-internal.h:183
  8    Thread 0x7fe9a5c16700 (LWP 30866) futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7fe990000ba8)
    at ../sysdeps/nptl/futex-internal.h:183
  9    Thread 0x7fe9a5380700 (LWP 30867) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  10   Thread 0x7fe9a4b7f700 (LWP 30868) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  11   Thread 0x7fe98ffff700 (LWP 30869) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  12   Thread 0x7fe98f7fe700 (LWP 30870) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  13   Thread 0x7fe98effd700 (LWP 30871) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  14   Thread 0x7fe98e7fc700 (LWP 30872) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  15   Thread 0x7fe98dffb700 (LWP 30873) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  16   Thread 0x7fe98d7fa700 (LWP 30874) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
  17   Thread 0x7fe98cff9700 (LWP 30875) __lll_lock_wait (futex=futex@entry=0x55f94048a718, private=0) at lowlevellock.c:52
[... cut ...]
#2  0x00007fe9aa33f3b4 in __gthread_mutex_lock(pthread_mutex_t*) () from /root/ros2_ws/colcon_ws/install/rclcpp/lib/librclcpp.so
#3  0x00007fe9aa33f484 in std::mutex::lock() () from /root/ros2_ws/colcon_ws/install/rclcpp/lib/librclcpp.so
#4  0x00007fe9aa33f8a4 in std::lock_guard<std::mutex>::lock_guard(std::mutex&) () from /root/ros2_ws/colcon_ws/install/rclcpp/lib/librclcpp.so
#5  0x00007fe9aa37a3a4 in rclcpp::executors::MultiThreadedExecutor::run(unsigned long) ()
<snip...>
</code>
</pre>
</details>

ivanpauno, 2020-09-28:
I don't quite understand this issue, but I have opened a PR that removes `sceheduled_timers_` usage in the mt executor: https://github.com/ros2/rclcpp/pull/1328.

It would be great to test if that also solves this problem.

fujitatomoya, 2020-09-29:
@ivanpauno 

i will do that.

fujitatomoya, 2020-09-29:
confirmed https://github.com/ros2/rclcpp/pull/1328/ fixes the issue.

- rmw_fastdds

```
[component_container_mt-1] [INFO] [1601372025.872317566] [talker]: Publishing: 'Hello World: 1'
[component_container_mt-1] [INFO] [1601372025.872808480] [listener]: I heard: [Hello World: 1]
[component_container_mt-1] [INFO] [1601372026.872449828] [talker]: Publishing: 'Hello World: 2'
[component_container_mt-1] [INFO] [1601372026.872734021] [listener]: I heard: [Hello World: 2]
[component_container_mt-1] [INFO] [1601372027.872235906] [talker]: Publishing: 'Hello World: 3'
[component_container_mt-1] [INFO] [1601372027.872781040] [listener]: I heard: [Hello World: 3]
[component_container_mt-1] [INFO] [1601372028.872236644] [talker]: Publishing: 'Hello World: 4'
[component_container_mt-1] [INFO] [1601372028.872529507] [listener]: I heard: [Hello World: 4]
[component_container_mt-1] [INFO] [1601372029.872199911] [talker]: Publishing: 'Hello World: 5'
[component_container_mt-1] [INFO] [1601372029.872624625] [listener]: I heard: [Hello World: 5]
[component_container_mt-1] [INFO] [1601372030.872278543] [talker]: Publishing: 'Hello World: 6'
[component_container_mt-1] [INFO] [1601372030.872534205] [listener]: I heard: [Hello World: 6]
[component_container_mt-1] [INFO] [1601372031.872720731] [talker]: Publishing: 'Hello World: 7'
[component_container_mt-1] [INFO] [1601372031.873041964] [listener]: I heard: [Hello World: 7]
[component_container_mt-1] [INFO] [1601372032.872257135] [talker]: Publishing: 'Hello World: 8'
[component_container_mt-1] [INFO] [1601372032.872714529] [listener]: I heard: [Hello World: 8]
```

- rmw_cyclonedds

```
[component_container_mt-1] [INFO] [1601372109.950295335] [talker]: Publishing: 'Hello World: 1'
[component_container_mt-1] [INFO] [1601372109.950743537] [listener]: I heard: [Hello World: 1]
[component_container_mt-1] [INFO] [1601372110.950040348] [talker]: Publishing: 'Hello World: 2'
[component_container_mt-1] [INFO] [1601372110.950357710] [listener]: I heard: [Hello World: 2]
[component_container_mt-1] [INFO] [1601372111.949958243] [talker]: Publishing: 'Hello World: 3'
[component_container_mt-1] [INFO] [1601372111.950310606] [listener]: I heard: [Hello World: 3]
[component_container_mt-1] [INFO] [1601372112.950317865] [talker]: Publishing: 'Hello World: 4'
[component_container_mt-1] [INFO] [1601372112.950659037] [listener]: I heard: [Hello World: 4]
[component_container_mt-1] [INFO] [1601372113.950012608] [talker]: Publishing: 'Hello World: 5'
[component_container_mt-1] [INFO] [1601372113.950361731] [listener]: I heard: [Hello World: 5]
[component_container_mt-1] [INFO] [1601372114.950237556] [talker]: Publishing: 'Hello World: 6'
[component_container_mt-1] [INFO] [1601372114.950622808] [listener]: I heard: [Hello World: 6]
[component_container_mt-1] [INFO] [1601372115.950248843] [talker]: Publishing: 'Hello World: 7'
[component_container_mt-1] [INFO] [1601372115.950621225] [listener]: I heard: [Hello World: 7]
[component_container_mt-1] [INFO] [1601372116.950088295] [talker]: Publishing: 'Hello World: 8'
[component_container_mt-1] [INFO] [1601372116.950384157] [listener]: I heard: [Hello World: 8]
```

fujitatomoya, 2020-10-21:
https://github.com/ros2/rclcpp/pull/1382 is experimental so that there is an area for discussion. but it can fix the problem and save the cpu consumption.

fujitatomoya, 2021-01-28:
This should be fixed with https://github.com/ros2/rclcpp/pull/1516.

fujitatomoya, 2021-01-28:
@ivanpauno 

i confirmed that https://github.com/ros2/rclcpp/pull/1516 can fix this problem as well.

- cyclonedds

```
# ros2 launch composition composition_demo.launch.py
...<snip>
[component_container_mt-1] [INFO] [1611817735.084614477] [talker]: Publishing: 'Hello World: 1'
[component_container_mt-1] [INFO] [1611817735.085075631] [listener]: I heard: [Hello World: 1]
[component_container_mt-1] [INFO] [1611817736.084593803] [talker]: Publishing: 'Hello World: 2'
[component_container_mt-1] [INFO] [1611817736.085019077] [listener]: I heard: [Hello World: 2]
[component_container_mt-1] [INFO] [1611817737.084578112] [talker]: Publishing: 'Hello World: 3'
[component_container_mt-1] [INFO] [1611817737.084953676] [listener]: I heard: [Hello World: 3]
[component_container_mt-1] [INFO] [1611817738.084578878] [talker]: Publishing: 'Hello World: 4'
[component_container_mt-1] [INFO] [1611817738.085018892] [listener]: I heard: [Hello World: 4]
[component_container_mt-1] [INFO] [1611817739.084575016] [talker]: Publishing: 'Hello World: 5'
[component_container_mt-1] [INFO] [1611817739.084944959] [listener]: I heard: [Hello World: 5]
[component_container_mt-1] [INFO] [1611817740.084591976] [talker]: Publishing: 'Hello World: 6'
[component_container_mt-1] [INFO] [1611817740.084957980] [listener]: I heard: [Hello World: 6]
[component_container_mt-1] [INFO] [1611817741.084588515] [talker]: Publishing: 'Hello World: 7'
[component_container_mt-1] [INFO] [1611817741.084895548] [listener]: I heard: [Hello World: 7]
[component_container_mt-1] [INFO] [1611817742.084603086] [talker]: Publishing: 'Hello World: 8'
[component_container_mt-1] [INFO] [1611817742.084907429] [listener]: I heard: [Hello World: 8]
[component_container_mt-1] [INFO] [1611817743.084568834] [talker]: Publishing: 'Hello World: 9'
[component_container_mt-1] [INFO] [1611817743.084878147] [listener]: I heard: [Hello World: 9]
```

- fastdds

```
# RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 launch composition composition_demo.launch.py
...<snip>
[component_container_mt-1] [INFO] [1611817774.218779010] [talker]: Publishing: 'Hello World: 1'
[component_container_mt-1] [INFO] [1611817774.219381238] [listener]: I heard: [Hello World: 1]
[component_container_mt-1] [INFO] [1611817775.218640326] [talker]: Publishing: 'Hello World: 2'
[component_container_mt-1] [INFO] [1611817775.219119783] [listener]: I heard: [Hello World: 2]
[component_container_mt-1] [INFO] [1611817776.218647583] [talker]: Publishing: 'Hello World: 3'
[component_container_mt-1] [INFO] [1611817776.218924426] [listener]: I heard: [Hello World: 3]
[component_container_mt-1] [INFO] [1611817777.218654191] [talker]: Publishing: 'Hello World: 4'
[component_container_mt-1] [INFO] [1611817777.219018266] [listener]: I heard: [Hello World: 4]
[component_container_mt-1] [INFO] [1611817778.218681025] [talker]: Publishing: 'Hello World: 5'
[component_container_mt-1] [INFO] [1611817778.219128822] [listener]: I heard: [Hello World: 5]
[component_container_mt-1] [INFO] [1611817779.218701685] [talker]: Publishing: 'Hello World: 6'
[component_container_mt-1] [INFO] [1611817779.219029850] [listener]: I heard: [Hello World: 6]
[component_container_mt-1] [INFO] [1611817780.218699101] [talker]: Publishing: 'Hello World: 7'
[component_container_mt-1] [INFO] [1611817780.219057166] [listener]: I heard: [Hello World: 7]
[component_container_mt-1] [INFO] [1611817781.219171478] [talker]: Publishing: 'Hello World: 8'
[component_container_mt-1] [INFO] [1611817781.219456132] [listener]: I heard: [Hello World: 8]
[component_container_mt-1] [INFO] [1611817782.218689579] [talker]: Publishing: 'Hello World: 9'
[component_container_mt-1] [INFO] [1611817782.219035864] [listener]: I heard: [Hello World: 9]
[component_container_mt-1] [INFO] [1611817783.218741336] [talker]: Publishing: 'Hello World: 10'
[component_container_mt-1] [INFO] [1611817783.219077900] [listener]: I heard: [Hello World: 10]
```

fujitatomoya, 2021-02-17:
closing this, https://github.com/ros2/rclcpp/pull/1516 is merged.

fujitatomoya, 2021-02-17:
> closing this, ros2/rclcpp#1516 is merged.

either @rrrand or @ivanpauno, can you do me a favor?

clalancette, 2021-02-18:
Closing as ros2/rclcpp#1516 is merged.

closed 2021-02-18 by clalancette