ros2/ros2 #1036: Tracking Issue for RPM patching

cottsay, 2020-09-22:
I'm staring the process of adding patches to the release repositories for RPM packages in preparation for merging ros/rosdistro#26262. This is a great opportunity to enumerate everywhere that we need to patch for packaging purposes. Here, I'll link to the reviews for those patches and attempt to categorize the justification for the patches.

Once all of the patches have been shown to work properly on the buildfarm, I'll close this issue.

- [ ] `ament_cmake` :boot:
- [ ] `ament_lint` :red_circle: 
- [ ] `ament_package` :red_circle: 
- [ ] `cyclonedds` :question: 
- [ ] `fastrtps` :question: 
- [ ] `qt_gui_core` :red_circle: 
- [ ] `rmw_connext` :red_circle: 
- [ ] `rmw_implementation` :link: :red_circle: 
- [ ] `ros1_bridge` :question:
- [ ] `rosbag2` :red_circle: 
- [ ] `rosidl_typesupport`:link: :red_circle: 
- [ ] `ros_workspace` :boot:
- [ ] `rqt_plot` :red_circle: 
- [ ] `rviz` :back:

Legend:
- :boot: Boostrap
- :back: Backport
- :red_circle: RedHat-specific workaround (RPM/RHEL/CentOS/Fedora)
- :link: Group resolution
- :question: Other

clad26, 2020-10-13:
Does this mean we're getting ROS2 on Fedora soon？

cottsay, 2020-10-13:
> Does this mean we're getting ROS2 on Fedora soon?

We don't *currently* have any plans to release official Fedora packages for ROS 2, but we haven't really discussed it. We're shooting for at least some CentOS packages soon, though, which will give us an idea of what the maintenance burden will be like. Stay tuned.

gavanderhoorn, 2020-10-14:
You must have internal/private users of all this infrastructure, otherwise you would not be spending all this effort, correct @cottsay?

clad26, 2020-10-14:
Not being able to use ROS2 or Ignition Software on Fedora (or most linux distros other than Ubuntu) sucks...

clad26, 2020-10-14:
https://github.com/JuliaPackaging/BinaryBuilder.jl
Might help in building binaries for every Linux distro in general.

clalancette, 2020-10-14:
> Not being able to use ROS2 or Ignition Software on Fedora (or most linux distros other than Ubuntu) sucks...

For what it is worth, it is possible to use ROS 2 on Fedora if you build from source.

Also, you can do your development inside of containers.  A lot of people (including myself) run whatever they want as the host distribution, but then use LXC or Docker to do their actual ROS development.

> https://binarybuilder.org/
> Might help in building binaries for every Linux distro in general.
> Really easy easy to use too！

Yeah, there are a lot of options out there for building binaries for various Linux distributions.  The problem is that we have a pretty customized build system currently, and switching to anything else would be a very large effort (and possibly be disruptive to the community).

That's not to say it isn't worth investigating, but someone would need to step up and do a lot of research and work to figure out  what it would take to switch.

cottsay, 2021-02-12:
All of these patches are merged and building successfully on the pre-production farm, so I'm closing this issue.

closed 2021-02-12 by cottsay

clad26, 2021-02-12:
@cottsay 
So... CentOS is kinda dead(ish)
Any update for native Fedora binaries?

cottsay, 2021-02-12:
> So... CentOS is kinda dead(ish)

The target has always been RHEL and RHEL clones - CentOS is just the specific clone we're using to build the packages (following in EPEL's footsteps). We're also targeting RHEL 7, which is largely unaffected by the recent missteps by the CentOS project.

> Any update for native Fedora binaries?

Nothing has changed. I think we need to see how the RHEL packages go, and gauge how much it costs to maintain an additional RPM distribution in both time and raw compute costs before we could even start that discussion.

If I have anything to share, I'll do it on Discourse.