ros2/ros2 #1039: ros2 Foxy: parameter in yaml file with value "off" of "OFF" is forcefully assumed to be of type boolean

olivier-haddad, 2020-09-30:
On ros2 Foxy 
- Operating System: Ubuntu 20.04
- Installation type: from binaries
- Version or commit hash:
  - 
- DDS implementation: FastRTPS
- Client library (if applicable):  rclcpp

#### Steps to reproduce issue
-declare a parameter of type string. There is no need to even call get_parameter
-in the yaml file add the parameter:

    param1: OFF

```
    declare_parameter("param1", std::string("auto"));
```

#### Expected behavior
all works, calling get_parameter will work. As this is declared as a string, it shouldn't infer type.

#### Actual behavior
An exception is thrown as soon as the node instance is created.
I suppose it comes from some default value inference because of the value "off" that is interpreted as boolean "false"/

terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
  what():  parameter 'param1' has invalid type: expected [string] got [bool]

#### Additional information
Modifying the "OFF" value in the yaml file to "auto" or "test" will run fine.

gavanderhoorn, 2020-09-30:
Isn't this the yaml parser?

At least yaml 1.1 recognised `off` (and host of alternative spellings) as `false` I believe.

clalancette, 2020-09-30:
> At least yaml 1.1 recognised `off` (and host of alternative spellings) as `false` I believe.

Yeah, that's exactly right.  The parameter is being declared as a string, but then the YAML parser interprets the incoming value as a boolean.

@olivier-haddad If you change your parameter to:

```
param1: "OFF"
```

Does it work for you?

olivier-haddad, 2020-09-30:
Indeed param1: "OFF" works.

clalancette, 2020-09-30:
All right.  Them I'm tempted to declare this as "not-a-bug"; this is the way that yaml parsing works.  Unless you have other thoughts?

olivier-haddad, 2020-09-30:
If this is standard YAML parsing, then it is definitely not a bug.

I would, however add a little warning in the "parameters" documentation page as it took me a while to figure it out as string seems like the default "take all" parameter type.

clalancette, 2020-09-30:
> I would, however add a little warning in the "parameters" documentation page as it took me a while to figure it out as string seems like the default "take all" parameter type.

That makes sense.  Would you mind opening a PR to https://github.com/ros2/ros2_documentation for this?

In the meantime, I'm going to close this out.  Thanks.

closed 2020-09-30 by clalancette

olivier-haddad, 2020-09-30:
> That makes sense. Would you mind opening a PR to https://github.com/ros2/ros2_documentation for this?

Do you mean opening an issue or modifying the documentation and creating a Pull Request with the proposed changes.

clalancette, 2020-09-30:
> Do you mean opening an issue or modifying the documentation and creating a Pull Request with the proposed changes.

A pull request would be best, but if you just want to open an issue, you can do that too.