ros2/ros2 #1047: Skip python dependencies?

clad26, 2020-10-13:
Any way I can skip python dependencies while downloading the ros2 source?
I only want to use cpp, any option available for that?

clalancette, 2020-10-29:
I don't believe we have a way to skip python while downloading the source.

There is some talk about doing a "minimal" version of ROS 2 that doesn't need python; you can follow the discussion at https://github.com/ros-infrastructure/rep/pull/231 .

emersonknapp, 2020-10-29:
Downloading the source and building python-based packages are different things, really. You could construct a `.repos` file that only contained dependencies up through `rclcpp` fairly easily. This one https://github.com/emersonknapp/ros2-toolchain-crosscompile/blob/master/minimal_cpp_ros2_master.repos is definitely out of date by now, but is probably a fairly close approximation to that dependency set.

I will note that the commandline utilities (`ros2 topic`, `ros2 launch` etc) are all python-based, so you will have limited ways to interact with a ros2 application without any Python present. 

Another hiccup that hasn't been figured out yet in the ros-infrastructure/rep#231 discussion is that even for pure-C packages like `rcl`, many of the tests use `launch`, bringing in a Python dependency for running/building tests (this may not be as much of a concern in your case)